espressif/esp-va-sdk #1: AVS-SDK missing source code

mjadams, 2018-04-28:
The AVS-SDK is currently missing all source code. Additionally, the hardware guide is missing the schematics for the microphone array board.

ameyinamdar, 2018-04-28:
Hi @mjadams,

We are in a process of releasing the sources for the AVS-SDK, please stay tuned.

For the hardware guide and schematic, please refer to https://www.espressif.com/sites/default/files/documentation/esp32-lyratd-msc_user_guide_en.pdf

Thanks,
Amey

mjadams, 2018-04-29:
I look forward to assisting with a custom implementation of this AWS Alexa AVS Device when the source-code is released. Alternatively I will get in touch directly with Espressif through your Angus and Teo.

closed 2018-04-29 by mjadams

ameyinamdar, 2018-09-19:
@mjadams The source is now available on GitHub for application developers to build AVS based apps.