espressif/esp-va-sdk #5: No reaction when you press the button "Rec"

chen-zifeng, 2018-09-27:
log:
I (0) cpu_start: App cpu up.
I (1325) spiram: SPI SRAM memory test OK
I (1325) heap_init: Initializing. RAM available for dynamic allocation:
I (1326) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1332) heap_init: At 3FFBF248 len 00020DB8 (131 KiB): DRAM
I (1338) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (1345) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1351) heap_init: At 400998E8 len 00006718 (25 KiB): IRAM
I (1357) cpu_start: Pro cpu start user code
I (1362) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (1370) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (49) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (49) alexa: ==== Alexa SDK version: v1.0b1-163199 ====
Registering command: mem-dump 
I (69) scli: Initialising UART on port 0
I (69) uart: queue free spaces: 8
>> Registering command: task-dump 
Registering command: nvs-get 
Registering command: nvs-set 
Registering command: heap-trace 
I (89) wifi: wifi driver task: 3ffc6fec, prio:23, stack:3584, core=0
I (89) wifi: wifi firmware version: 633012a
I (89) wifi: config NVS flash: disabled
I (99) wifi: config nano formating: disabled
I (99) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (109) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (119) wifi: Init dynamic tx buffer num: 32
I (129) wifi: Init data frame dynamic rx buffer num: 26
I (129) wifi: Init management frame dynamic rx buffer num: 26
I (139) wifi: Init static tx buffer num: 26
I (139) wifi: Init static rx buffer size: 1600
I (139) wifi: Init static rx buffer num: 10
I (149) wifi: Init dynamic rx buffer num: 26
I (149) conn_mgr: Initialization success
I (159) conn_mgr: Received event: 64
I (159) softap_sm: SoftAP SSID: ESP-Alexa-4d01
I (219) phy: phy_version: 3960, 5211945, Jul 18 2018, 10:40:07, 0, 0
I (219) wifi: mode : softAP (80:7d:3a:e1:4d:01)
I (229) conn_mgr: Received event: 128
I (229) alexa: SOFTAP_STARTED event
I (229) protocomm_httpd: Adding endpoint : prov-session
I (239) protocomm_httpd: Adding endpoint : avsconfig
I (239) protocomm_httpd: Adding endpoint : prov-config
I (4979) wifi: n:1 0, o:1 0, ap:1 1, sta:255 255, prof:1
I (4979) wifi: station: 9c:e8:2b:67:ea:b9 join, AID=1, bgn, 20
I (4979) conn_mgr: Received event: 128
I (5099) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.2
I (5099) conn_mgr: Received event: 128

avsheth, 2018-09-27:
Hi @chen-zifeng 
Device is still in provisioning mode. Rec button will be effective as soon as device is connected to an AP.
Please provision the device using Android app first.

TraDao, 2018-10-04:
Hi @avsheth I have the same issue as @chen-zifeng . (same message in console)

I used Espressif app on android. The app says: "To provision your new device, please make sure that your device is turned on and the Wifi symbol blinks green". But the Wifi symbol green led does not blink, it is off. There is no way to set it blink. 

I tried to go to next step: connect to ESP-Alexa-xxx (without green led blinking). The board stays in provisioning mode forever. (it show same message as @chen-zifeng 's)

avsheth, 2018-10-04:
Hi @TraDao 
After connecting to ESP-Alexa-xxx, please go back to the App, where you'll see "Provision Network" button enabled, clicking on which will give you couple of text boxes for entering SSID and passphrase of the network you wish to connect to (for now you have to enter SSID and passphrase manually). Once the device is connected to the AP you can start a conversation with Alexa

TraDao, 2018-10-04:
Hi @avsheth 
It works well now.
Notes: I go back to the App, but don't see "Provision Network". Still see "Connect". Then, I installed the app on another android phone, and login with another amazon account. Setting up successfully on this phone. I can run the demo now.
(First failed phone is Google Pixel, Second phone is J7 pro)
Thank you very much!

avsheth, 2018-10-04:
Hi Thanks for letting us know. We will check if there's any issue with Google Pixel.

closed 2018-10-05 by kedars

xiaoseyihe, 2018-10-17:
hi @avsheth 感谢你的回答，我参考“After connecting to ESP-Alexa-xxx, please go back to the App, where you'll see "Provision Network" button enabled, clicking on which will give you couple of text boxes for entering SSID and passphrase of the network you wish to connect to (for now you have to enter SSID and passphrase manually). Once the device is connected to the AP you can start a conversation with Alexa”进行设置，设备连接WiFi，退出ESP-Alexa-xxx，
rst:0x10 (RTCWDT_RTC_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6200
load:0x40078000,len:9940
load:0x40080000,len:6756
entry 0x40080350
I (29) boot: ESP-IDF v3.1-91-gf11ac03-dirty 2nd stage bootloader
I (29) boot: compile time 11:57:34
I (33) boot: Enabling RNG early entropy source...
I (35) qio_mode: Enabling QIO for flash chip GD
I (40) boot: SPI Speed      : 80MHz
I (44) boot: SPI Mode       : QIO
I (48) boot: SPI Flash Size : 4MB
I (52) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  2 avs              WiFi data        01 02 00010000 00006000
I (86) boot:  3 factory          factory app      00 00 00020000 00300000
I (93) boot: End of partition table
I (97) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x14258c (1320332) map
I (455) esp_image: segment 1: paddr=0x001625b4 vaddr=0x3ffb0000 size=0x05088 ( 20616) load
I (462) esp_image: segment 2: paddr=0x00167644 vaddr=0x3ffb5088 size=0x00000 (     0) load
I (462) esp_image: segment 3: paddr=0x0016764c vaddr=0x40080000 size=0x00400 (  1024) load
I (471) esp_image: segment 4: paddr=0x00167a54 vaddr=0x40080400 size=0x085bc ( 34236) load
I (491) esp_image: segment 5: paddr=0x00170018 vaddr=0x400d0018 size=0xfa1c0 (1024448) map
I (762) esp_image: segment 6: paddr=0x0026a1e0 vaddr=0x400889bc size=0x117b8 ( 71608) load
I (786) esp_image: segment 7: paddr=0x0027b9a0 vaddr=0x400c0000 size=0x00000 (     0) load
I (786) esp_image: segment 8: paddr=0x0027b9a8 vaddr=0x50000000 size=0x00000 (     0) load
I (808) boot: Loaded app from partition at offset 0x20000
I (808) boot: Disabling RNG early entropy source...
I (809) spiram: SPI RAM mode: flash 80m sram 80m
I (814) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (821) cpu_start: Pro cpu up.
I (825) cpu_start: Starting app cpu, entry point is 0x400814bc
I (0) cpu_start: App cpu up.
I (1325) spiram: SPI SRAM memory test OK
I (1325) heap_init: Initializing. RAM available for dynamic allocation:
I (1325) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1331) heap_init: At 3FFBF100 len 00020F00 (131 KiB): DRAM
I (1338) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (1344) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1351) heap_init: At 4009A174 len 00005E8C (23 KiB): IRAM
I (1357) cpu_start: Pro cpu start user code
I (1362) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (1370) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (48) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (49) alexa: ==== Alexa SDK version: v1.0b1-165973 ====
Registering command: mem-dump 
I (99) scli: Initialising UART on port 0
I (99) uart: queue free spaces: 8

>> Registering command: task-dump 
Registering command: nvs-get 
Registering command: nvs-set 
Registering command: heap-trace 
I (109) wifi: wifi driver task: 3ffcafec, prio:23, stack:3584, core=0
I (109) wifi: wifi firmware version: 75aab56
I (119) wifi: config NVS flash: disabled
I (119) wifi: config nano formating: disabled
I (129) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (139) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (149) wifi: Init dynamic tx buffer num: 32
I (149) wifi: Init data frame dynamic rx buffer num: 32
I (149) wifi: Init management frame dynamic rx buffer num: 32
[... 168 lines cut ...]
I (475269) audio_stream: Stream i2s_reader Event Started
I (475279) dsp: Reader stream event 2
speech_recognizer: Ignoring data in idle state
speech_recognizer: Ignoring data in idle state
speech_recognizer: Ignoring data in idle state
Stopping audio stream i2s_reader
speech_recognizer: Ignoring data in idle state
I (475299) audio_stream: Stream i2s_reader Event Stopped
I (475309) dsp: Reader stream event 3
Generated stopped event for stream i2s_reader
I (475319) I2S: PLL_D2: Req RATE: 48000, real rate: 48076.000, BITS: 16, CLKM: 13, BCK: 8, MCLK: 12292917.167, SCLK: 1538432.000000, diva: 64, divb: 1
I (475329) audio_stream: Starting playback_writer stream
Starting audio stream playback_writer
I (475339) audio_stream: Stream playback_writer Event Started
E (475349) avs-nvs-util: Error opening avs namespace!
I (475349) [speaker]: Using default volume
I (475359) LYRAT_CB: Volume changed to 40
I (475359) auth-delegate-config: Returning auth delegate companion app configuration
I (475369) [auth-delegate]: Auth delegate with comp app ANMjgBHkgctVDuEpQkZQ amzn1.application-oa2-client.a5b0d0c6a5f546499d1e5043f3515e6b amzn://com.espressif.provision abcd1234
[... cut ...] nVjUQVpKfKv4ZSnN-Js72KG-sqwlHm939d08a8VkMO0wyq3VUaCG9Uw83YBz4Pnd20ctmBV0g1CQen47uR5MrZzwbMYkKpHsngymL5hUxDrLe17bElm_CZX94i6C_eIrngf-h-sdjKGrMioUyrvOgWnIyZCSqzebPF4Y2gv4J5KHwBVhfGaGLHz2OEa-tl3laCARmQ38nQyBcd5GQKoU5QmK6dQk85cHTpVk25SVBR-L8mqWJAwsr5DydWf3Xre7GgESW3fwksg2eplA7iw in NVS successfully
I (479759) [auth-delegate]: Authentication Done !
I (479759) avs-nvs-util: No value set for: cap_version
I (479769) capabilities: capabilities version not found in NVS or version is different

这是怎么回事？谢谢回答