espressif/esp-va-sdk #6: Flashing DSP firmware or custom codec on ESP32-LYRATD-MSC

theronic, 2018-10-09:
How do I get access to the low-level DSP routines (not just resampling), or flash the DSP chip? Specifically, I need direct access to each microphone stream and apply some biqaud filtering, correlation calculations.

kewalmshah, 2018-10-09:
Hi @theronic the current library has firmware for AEC on the DSP. We will provide the abstraction layer on how you can flash your own firmware and configuration on the DSP in the next release.

theronic, 2018-10-09:
Thanks for the quick reply, @kewalmshah. Just to clarify, when you say, "flash your own firmware and configuration on the DSP", you mean the DSP firmware can be flashed separately from the ESP32 firmware?

When is the next release scheduled for?

kewalmshah, 2018-10-10:
Hi @theronic. Yes you create DSP firmware separately. This firmware is flashed to ESP32 and eventually ESP32 communicates with DSP on SPI to load the firmware to the DSP. We are already working on it and the release should be ready in couple of weeks.

closed 2018-12-19 by kedars