espressif/esp-va-sdk #7: 通过手机APP配网后，没有指示灯显示吗？按照用例说明测试没有反应

xiaoseyihe, 2018-10-16:
显示已经配网成功，如下：

rst:0x1 (POWERON_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6200
load:0x40078000,len:9940
load:0x40080000,len:6756
entry 0x40080350
I (29) boot: ESP-IDF v3.1-91-gf11ac03-dirty 2nd stage bootloader
I (29) boot: compile time 11:57:34
I (33) boot: Enabling RNG early entropy source...
I (35) qio_mode: Enabling QIO for flash chip GD
I (40) boot: SPI Speed      : 80MHz
I (44) boot: SPI Mode       : QIO
I (48) boot: SPI Flash Size : 4MB
I (52) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  2 avs              WiFi data        01 02 00010000 00006000
I (85) boot:  3 factory          factory app      00 00 00020000 00300000
I (93) boot: End of partition table
I (97) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x14258c (1320332) map
I (455) esp_image: segment 1: paddr=0x001625b4 vaddr=0x3ffb0000 size=0x05088 ( 20616) load
I (462) esp_image: segment 2: paddr=0x00167644 vaddr=0x3ffb5088 size=0x00000 (     0) load
I (462) esp_image: segment 3: paddr=0x0016764c vaddr=0x40080000 size=0x00400 (  1024) load
I (471) esp_image: segment 4: paddr=0x00167a54 vaddr=0x40080400 size=0x085bc ( 34236) load
I (491) esp_image: segment 5: paddr=0x00170018 vaddr=0x400d0018 size=0xfa1c0 (1024448) map
I (762) esp_image: segment 6: paddr=0x0026a1e0 vaddr=0x400889bc size=0x117b8 ( 71608) load
I (786) esp_image: segment 7: paddr=0x0027b9a0 vaddr=0x400c0000 size=0x00000 (     0) load
I (786) esp_image: segment 8: paddr=0x0027b9a8 vaddr=0x50000000 size=0x00000 (     0) load
I (808) boot: Loaded app from partition at offset 0x20000
I (808) boot: Disabling RNG early entropy source...
I (809) spiram: SPI RAM mode: flash 80m sram 80m
I (814) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (821) cpu_start: Pro cpu up.
I (825) cpu_start: Starting app cpu, entry point is 0x400814bc
I (0) cpu_start: App cpu up.
I (1324) spiram: SPI SRAM memory test OK
I (1325) heap_init: Initializing. RAM available for dynamic allocation:
I (1325) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1331) heap_init: At 3FFBF100 len 00020F00 (131 KiB): DRAM
I (1337) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (1344) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1350) heap_init: At 4009A174 len 00005E8C (23 KiB): IRAM
I (1357) cpu_start: Pro cpu start user code
I (1361) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (1369) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (48) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (49) alexa: ==== Alexa SDK version: v1.0b1-165973 ====
Registering command: mem-dump 
I (99) scli: Initialising UART on port 0
I (99) uart: queue free spaces: 8

>> Registering command: task-dump 
Registering command: nvs-get 
Registering command: nvs-set 
Registering command: heap-trace 
I (109) wifi: wifi driver task: 3ffcafec, prio:23, stack:3584, core=0
I (119) wifi: wifi firmware version: 75aab56
I (119) wifi: config NVS flash: disabled
I (119) wifi: config nano formating: disabled
I (129) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (139) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (149) wifi: Init dynamic tx buffer num: 32
I (149) wifi: Init data frame dynamic rx buffer num: 32
I (149) wifi: Init management frame dynamic rx buffer num: 32
I (159) wifi: Init static tx buffer num: 32
I (169) wifi: Init static rx buffer size: 1600
I (169) wifi: Init static rx buffer num: 10
I (169) wifi: Init dynamic rx buffer num: 32
I (179) conn_mgr: Initialization success
I (179) conn_mgr: Received event: 64
I (189) softap_sm: SoftAP SSID: ESP-Alexa-4fa1
I (249) phy: phy_version: 4000, b6198fa, Sep  3 2018, 15:11:06, 0, 0
I (249) wifi: mode : softAP (80:7d:3a:e1:4f:a1)
I (249) wifi: Init max length of beacon: 752/752
I (249) wifi: Init max length of beacon: 752/752
I (259) conn_mgr: Received event: 128
I (259) alexa: SOFTAP_STARTED event
I (269) protocomm_httpd: Adding endpoint : prov-session
I (269) protocomm_httpd: Adding endpoint : avsconfig
I (279) protocomm_httpd: Adding endpoint : prov-config
I (3739) wifi: n:1 0, o:1 0, ap:1 1, sta:255 255, prof:1
I (3739) wifi: station: 8c:79:67:c8:70:ab join, AID=1, bgn, 20
I (3739) conn_mgr: Received event: 128
I (3959) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.2
I (3969) conn_mgr: Received event: 128

avsheth, 2018-10-16:
Hi @xiaoseyihe Please check issue #5 . Please go back to the app after connecting to softAP to resume provisioning.

kedars, 2018-12-19:
Closing this for now. Please re-open if the suggestion above doesn't work.

closed 2018-12-19 by kedars