espressif/esp-va-sdk #8: lyrat_alexa 与 lyrat_alexa_sr的区别

xiaoseyihe, 2018-10-17:
你好，从描述中看到，lyrat_alexa不支持“Alexa”唤醒，只支持按键，lyrat_alexa_sr可以支持“Alexa”唤醒，但是只支持 LyraT v4.1, LyraT v4.2，不支持 LyraT v4.3,是这样的吗？

我目前用的设备是 LyraT v4.3，怎么实现“Alexa”唤醒词唤醒功能？多谢指导

avsheth, 2018-10-17:
Hi @xiaoseyihe 
Latest release supports wakeword on LyraT v4.3 also.
Thanks for pointing out, will update the README

xiaoseyihe, 2018-10-17:
@avsheth 感谢您的回答，我烧写lyrat_alexa_sr的版本后，串口打印连接WiFi，为什么指示灯不显示呢？

串口打印如下：
Registering command: task-dump
Registering command: nvs-get
Registering command: nvs-set
Registering command: heap-trace
I (109) wifi: wifi driver task: 3ffcafec, prio:23, stack:3584, core=0
I (109) wifi: wifi firmware version: 75aab56
I (119) wifi: config NVS flash: disabled
I (119) wifi: config nano formating: disabled
I (129) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (139) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (149) wifi: Init dynamic tx buffer num: 32
I (149) wifi: Init data frame dynamic rx buffer num: 32
I (149) wifi: Init management frame dynamic rx buffer num: 32
I (159) wifi: Init static tx buffer num: 32
I (159) wifi: Init static rx buffer size: 1600
I (169) wifi: Init static rx buffer num: 10
I (169) wifi: Init dynamic rx buffer num: 32
I (179) conn_mgr: Initialization success
I (179) conn_mgr: Received event: 64
I (179) softap_sm: SoftAP SSID: ESP-Alexa-4fa1
I (249) phy: phy_version: 4000, b6198fa, Sep 3 2018, 15:11:06, 0, 0
I (249) wifi: mode : softAP (80:7d:3a:e1:4f:a1)
I (249) wifi: Init max length of beacon: 752/752
I (259) wifi: Init max length of beacon: 752/752
I (259) conn_mgr: Received event: 128
I (269) alexa: SOFTAP_STARTED event
I (269) protocomm_httpd: Adding endpoint : prov-session
I (279) protocomm_httpd: Adding endpoint : avsconfig
I (279) protocomm_httpd: Adding endpoint : prov-config
I (64189) wifi: n:1 0, o:1 0, ap:1 1, sta:255 255, prof:1
I (64189) wifi: station: 8c:79:67:c8:70:ab join, AID=1, bgn, 20
I (64189) conn_mgr: Received event: 128
I (64479) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.2
I (64479) conn_mgr: Received event: 128
I (102609) wifi: station: 8c:79:67:c8:70:ab leave, AID = 1
I (102609) wifi: n:1 0, o:1 0, ap:1 1, sta:255 255, prof:1
I (102609) conn_mgr: Received event: 128
I (416199) wifi: n:1 0, o:1 0, ap:1 1, sta:255 255, prof:1
I (416199) wifi: station: 8c:79:67:c8:70:ab join, AID=1, bgn, 20
I (416199) conn_mgr: Received event: 128
I (416789) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.2
I (416789) conn_mgr: Received event: 128
W (419199) wifi: rx ampdu exist: tid=0 new <ssn=74 win=0> old <ssn=74 win=6>
W (420199) wifi: rx ampdu exist: tid=0 new <ssn=131 win=0> old <ssn=131 win=6>
W (421209) wifi: rx ampdu exist: tid=0 new <ssn=237 win=0> old <ssn=236 win=6>
W (422199) wifi: rx ampdu exist: tid=0 new <ssn=268 win=0> old <ssn=268 win=6>
W (423199) wifi: rx ampdu exist: tid=0 new <ssn=308 win=0> old <ssn=308 win=6>
W (424199) wifi: rx ampdu exist: tid=0 new <ssn=334 win=0> old <ssn=334 win=6>
W (425209) wifi: rx ampdu exist: tid=0 new <ssn=395 win=0> old <ssn=395 win=6>
W (426209) wifi: rx ampdu exist: tid=0 new <ssn=448 win=0> old <ssn=448 win=6>
W (427259) wifi: rx ampdu exist: tid=0 new <ssn=494 win=0> old <ssn=494 win=6>
W (428209) wifi: rx ampdu exist: tid=0 new <ssn=521 win=0> old <ssn=521 win=6>
W (430209) wifi: rx ampdu exist: tid=0 new <ssn=586 win=0> old <ssn=586 win=6>
W (431209) wifi: rx ampdu exist: tid=0 new <ssn=620 win=0> old <ssn=620 win=6>
W (432209) wifi: rx ampdu exist: tid=0 new <ssn=644 win=0> old <ssn=644 win=6>
W (433209) wifi: rx ampdu exist: tid=0 new <ssn=658 win=0> old <ssn=658 win=6>
W (434219) wifi: rx ampdu exist: tid=0 new <ssn=681 win=0> old <ssn=681 win=6>
W (435209) wifi: rx ampdu exist: tid=0 new <ssn=707 win=0> old <ssn=707 win=6>
W (436209) wifi: rx ampdu exist: tid=0 new <ssn=715 win=0> old <ssn=715 win=6>
W (437209) wifi: rx ampdu exist: tid=0 new <ssn=753 win=0> old <ssn=753 win=6>
W (438209) wifi: rx ampdu exist: tid=0 new <ssn=770 win=0> old <ssn=770 win=6>
W (439239) wifi: rx ampdu exist: tid=0 new <ssn=792 win=0> old <ssn=792 win=6>
W (440219) wifi: rx ampdu exist: tid=0 new <ssn=819 win=0> old <ssn=819 win=6>
W (441229) wifi: rx ampdu exist: tid=0 new <ssn=834 win=0> old <ssn=834 win=6>
W (442229) wifi: rx ampdu exist: tid=0 new <ssn=870 win=0> old <ssn=870 win=6>
W (443229) wifi: rx ampdu exist: tid=0 new <ssn=903 win=0> old <ssn=903 win=6>
W (445229) wifi: rx ampdu exist: tid=0 new <ssn=923 win=0> old <ssn=923 win=6>
[... 114 lines cut ...]
I (475269) audio_stream: Stream i2s_reader Event Started
I (475279) dsp: Reader stream event 2
speech_recognizer: Ignoring data in idle state
speech_recognizer: Ignoring data in idle state
speech_recognizer: Ignoring data in idle state
Stopping audio stream i2s_reader
speech_recognizer: Ignoring data in idle state
I (475299) audio_stream: Stream i2s_reader Event Stopped
I (475309) dsp: Reader stream event 3
Generated stopped event for stream i2s_reader
I (475319) I2S: PLL_D2: Req RATE: 48000, real rate: 48076.000, BITS: 16, CLKM: 13, BCK: 8, MCLK: 12292917.167, SCLK: 1538432.000000, diva: 64, divb: 1
I (475329) audio_stream: Starting playback_writer stream
Starting audio stream playback_writer
I (475339) audio_stream: Stream playback_writer Event Started
E (475349) avs-nvs-util: Error opening avs namespace!
I (475349) [speaker]: Using default volume
I (475359) LYRAT_CB: Volume changed to 40
I (475359) auth-delegate-config: Returning auth delegate companion app configuration
I (475369) [auth-delegate]: Auth delegate with comp app ANMjgBHkgctVDuEpQkZQ amzn1.application-oa2-client.a5b0d0c6a5f546499d1e5043f3515e6b amzn://com.espressif.provision abcd1234
[... cut ...] Js72KG-sqwlHm939d08a8VkMO0wyq3VUaCG9Uw83YBz4Pnd20ctmBV0g1CQen47uR5MrZzwbMYkKpHsngymL5hUxDrLe17bElm_CZX94i6C_eIrngf-h-sdjKGrMioUyrvOgWnIyZCSqzebPF4Y2gv4J5KHwBVhfGaGLHz2OEa-tl3laCARmQ38nQyBcd5GQKoU5QmK6dQk85cHTpVk25SVBR-L8mqWJAwsr5DydWf3Xre7GgESW3fwksg2eplA7iw in NVS successfully
I (479759) [auth-delegate]: Authentication Done !
I (479759) avs-nvs-util: No value set for: cap_version
I (479769) capabilities: capabilities version not found in NVS or version is different

设备连接WiFi的指示灯一直不亮，这是怎么回事？谢谢回答

avsheth, 2018-10-17:
Hi @xiaoseyihe Print on the app is incorrect. We already have removed it and will be updated in the next release.
From your logs, I can see that device was successfully connected to the AP and Alexa authentication was also successful.

xiaoseyihe, 2018-10-18:
@avsheth  问题已经解决了，应该和WIFI有关。再请教个问题，怎么重配WIFI，比如设备之前连接 一个WIFI，现在让它连接另一个WIFI环境，在出现“Now connect your phone to the wifi network with a name like: Esp-Alexa-xxxx”,但是我再WiFi列表中找不到Esp-Alexa-xxxx，。之前的配网需要重置吗？或者需要什么操作，谢谢

avsheth, 2018-10-18:
Hi @xiaoseyihe 
If I understood it correctly, you want to change your WiFi network on the device.
Once device is connected to one network, softAP on the device is turned off.
You can use CLI to set new credentials like below (press Enter to bring up CLI):
>> nvs-set cm ssid string <your ssid>
>> nvs-set cm pass string <your pass>

xiaoseyihe, 2018-10-19:
Thank you for your answers, I have achieved it.

closed 2018-10-19 by xiaoseyihe