espressif/esp-va-sdk #11: 用例lyrat_alexa_sr怎么保存音频数据到SD卡

xiaoseyihe, 2018-10-22:
用例lyrat_alexa_sr怎么保存音频数据到SD卡？通过fopen保存数据到SD卡

xiaoseyihe, 2018-10-23:
@avsheth 你好，我怎么才能在测试用例lyrat_alexa_sr中 采集到MIC 数据并保存到SD卡中呢？

avsheth, 2018-10-23:
@xiaoseyihe In app_dsp.c, raw mic data is given to speech_recognizer_record API. You can write this buffer (passed to speech_recognizer_record API as an argument) to SD card.

xiaoseyihe, 2018-10-23:
感谢你的回答，我如下保存数据，但是代码运行出错，请问系统支持fopen、fwrite等系统函数吗？
 while(1) {
        rb_read(dd.temp_rb, (uint8_t *)dd.data_buf, SAMPLE_SZ * 2, portMAX_DELAY);
        sent_len = audio_resample((short *)dd.data_buf, (short *)dd.data_buf, SAMP_RATE, DETECT_SAMP_RATE, SAMPLE_SZ, SAMPLE_SZ, 2, &dd.resample);
        sent_len = audio_resample_down_channel((short *)dd.data_buf, (short *)dd.data_buf, DETECT_SAMP_RATE, DETECT_SAMP_RATE, sent_len, SAMPLE_SZ, 0, &dd.resample);
        sent_len = sent_len * 2;  //convert 16bit lengtth to number of bytes
       
        if(dd.detect_wakeword) {
            xQueueSend(dd.recog_queue, dd.data_buf, 0);
        } else if(dd.speech_recog_en) {
            speech_recognizer_record(dd.data_buf, sent_len);
            printf("recorded speech %d\n", sent_len);
            
            file = fopen("/sdcard/rec.pcm", "ab+");
			fwrite((short *)dd.data_buf,2,sent_len/2,file);
			fclose(file);

avsheth, 2018-10-23:
@xiaoseyihe  can you post the entire code? Just want to check and verify mount configuration

xiaoseyihe, 2018-10-24:
@avsheth 以下是app_dsp.c的代码。使用fopen、fwrite函数需要配置？怎么配置呢？
/****************  app_dsp.c    ***********/
/*
 *      Copyright 2018, Espressif Systems (Shanghai) Pte Ltd.
 *  All rights regarding this code and its modifications reserved.
 *
 * This code contains confidential information of Espressif Systems
 * (Shanghai) Pte Ltd. No licenses or other rights express or implied,
 * by estoppel or otherwise are granted herein.
 *
 * THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS
 * "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT
 * LIMITED TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS
 * FOR A PARTICULAR PURPOSE ARE DISCLAIMED. IN NO EVENT SHALL THE
 * COPYRIGHT HOLDER OR CONTRIBUTORS BE LIABLE FOR ANY DIRECT,
 * INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES
 * (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR
 * SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION)
 * HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT,
 * STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE)
 * ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED
 * OF THE POSSIBILITY OF SUCH DAMAGE.
 */

#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <sys/time.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <freertos/queue.h>

#include <esp_log.h>
#include <speech_recognizer.h>
#include <mem_utils.h>
#include "app_dsp.h"
#include <ui_led.h>
#include <ui_button.h>
#include <i2s_stream.h>
#include <media_hal.h>
#include <audio_board.h>
#include <media_hal.h>
#include <ringbuf.h>
#include "resampling.h"
//Speech recognition headers
#include <esp_wwe.h>

#define WWE_TASK_STACK (8 * 1024)
#define DETECT_SAMP_RATE 16000UL
#define SAMP_RATE 48000UL
#define SAMP_BITS 16
#define SAMPLE_FRAME 320
#define SAMPLE_MS 20
#define PCM_SIZE (4 * 1024)
//Sample size for 20millisec data on 48KHz/16bit sampling. Division factor is (sectomillisec * bitsinbytes)
#define SAMPLE_SZ ((SAMP_RATE * I2S_BITS_PER_SAMPLE_16BIT * SAMPLE_MS) / (1000 * 8))
static const char *TAG = "dsp";

#ifdef CONFIG_AWS_IOT_SDK
extern xQueueHandle wake_word_sem;
#endif

static struct dsp_data {
    int item_chunk_size;
    bool detect_wakeword;
    bool speech_recog_en;
    QueueHandle_t recog_queue;
    ringbuf_t *temp_rb;
    audio_resample_config_t resample;
    i2s_stream_t *read_i2s_stream;
[... 173 lines cut ...]
        i2s_stream_destroy(dd.read_i2s_stream);
        dd.read_i2s_stream = NULL;
    }

    if (esp_wwe_init() != ESP_OK) {
        ESP_LOGE(TAG, "Failed to init ESP-WWE");
        return;
    }

	
    //Initialize sound source
    dd.item_chunk_size = esp_wwe_get_sample_chunksize() * sizeof(int16_t);
    dd.recog_queue = xQueueCreate(1, dd.item_chunk_size);
    xTaskCreate(&nn_task, "nn", WWE_TASK_STACK, NULL, (CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT - 1), &dd.nn_task_handle);
    xTaskCreate(&read_rb_task, "rb read task", WWE_TASK_STACK, NULL, (CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT - 1), NULL);
    
    audio_stream_start(&dd.read_i2s_stream->base);
    vTaskDelay(10/portTICK_RATE_MS);
    audio_stream_stop(&dd.read_i2s_stream->base);
    i2s_set_clk(I2S_NUM_0, SAMP_RATE, SAMP_BITS, I2S_CHANNEL_STEREO);
    vTaskDelay(10/portTICK_RATE_MS);
    audio_stream_start(&dd.read_i2s_stream->base);
    dd.detect_wakeword = true;
	
}

avsheth, 2018-10-24:
@xiaoseyihe I don't see any calls to mount the SD card filesystem. Please refer to SD card [example](https://github.com/espressif/esp-idf/blob/master/examples/storage/sd_card/main/sd_card_example_main.c) to get started with SD card

xiaoseyihe, 2018-10-30:
@avsheth  i refer to SD card example to get started with SD card，but, the following error occurred。
I (3162) conn_mgr: got ip:192.168.1.6
I (3162) conn_mgr: Received event: 128
I (3172) alexa: got IPv4:192.168.1.6
I (3172) I2S: DMA Malloc info, datalen=blocksize=600, dma_buf_count=3
I (3182) I2S: DMA Malloc info, datalen=blocksize=600, dma_buf_count=3
I (3182) I2S: PLL_D2: Req RATE: 16000, real rate: 16025.000, BITS: 16, CLKM: 39, BCK: 8, MCLK: 4096000.000, SCLK: 512800.000000, diva: 64, divb: 4
I (3202) gpio: GPIO[22]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 1| Intr:0 
I (3212) gpio: GPIO[19]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 1| Intr:0 
I (3222) UI_LED: LED initialized
I (3222) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 1| Intr:1 
I (3232) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 1| Intr:3 
I (3242) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (3272) dsp: Created I2S audio stream
I (3272) audio_stream: Starting i2s_reader stream
I (3322) dsp: Initializing SD card
I (3322) dsp: Using SDMMC peripheral
I (3322) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (3352) sdmmc_cmd: sdmmc_card_init: send_scr (2) returned 0xffffffff
**E (3352) dsp: Failed to mount filesystem. If you want the card to be formatted, set format_if_mount_failed = true.**
I (3362) audio_stream: Starting playback_writer stream
Starting audio stream playback_writer
I (3372) audio_stream: Stream playback_writer Event Started
I (3372) avs-nvs-util: No value set for: speaker_vol
I (3382) [speaker]: Using default volume
I (3392) LYRAT_CB: Volume changed to 40

code show as below：
void app_dsp_init(void)
{
    dd.temp_rb = rb_init("nn-recog", 4 * 1024);
    ui_led_init();
    ui_button_init();

    i2s_stream_config_t i2s_cfg;
    memset(&i2s_cfg, 0, sizeof(i2s_cfg));
    i2s_cfg.i2s_num = 0;
    audio_board_i2s_init_default(&i2s_cfg.i2s_config);
    i2s_cfg.media_hal_cfg = media_hal_init(&media_hal_conf);

    dd.read_i2s_stream = i2s_reader_stream_create(&i2s_cfg);
    if (dd.read_i2s_stream) {
        ESP_LOGI(TAG, "Created I2S audio stream");
    } else {
        ESP_LOGE(TAG, "Failed creating I2S audio stream");
    }
    i2s_stream_set_stack_size(dd.read_i2s_stream, 5000);

    audio_io_fn_arg_t stream_reader_fn = {
        .func = dsp_write_cb,
        .arg = NULL,
    };
    audio_event_fn_arg_t stream_event_fn = {
        .func = reader_stream_event_handler,
    };
    if (audio_stream_init(&dd.read_i2s_stream->base, "i2s_reader", &stream_reader_fn, &stream_event_fn) != 0) {
        ESP_LOGE(TAG, "Failed creating audio stream");
        i2s_stream_destroy(dd.read_i2s_stream);
        dd.read_i2s_stream = NULL;
    }

    if (esp_wwe_init() != ESP_OK) {
        ESP_LOGE(TAG, "Failed to init ESP-WWE");
        return;
    }

	/************挂载SD卡到系统文件 20181024 start****************/
	ESP_LOGI(TAG, "Initializing SD card");
	
	#ifndef USE_SPI_MODE
[... 54 lines cut ...]
                "Make sure SD card lines have pull-up resistors in place.", esp_err_to_name(ret));
        }
        return;
    }

    // Card has been initialized, print its properties
    sdmmc_card_print_info(stdout, card);

	/************挂载SD卡到系统文件 20181024 end****************/
	
    //Initialize sound source
    dd.item_chunk_size = esp_wwe_get_sample_chunksize() * sizeof(int16_t);
    dd.recog_queue = xQueueCreate(1, dd.item_chunk_size);
    xTaskCreate(&nn_task, "nn", WWE_TASK_STACK, NULL, (CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT - 1), &dd.nn_task_handle);
    xTaskCreate(&read_rb_task, "rb read task", WWE_TASK_STACK, NULL, (CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT - 1), NULL);
    
    audio_stream_start(&dd.read_i2s_stream->base);
    vTaskDelay(10/portTICK_RATE_MS);
    audio_stream_stop(&dd.read_i2s_stream->base);
    i2s_set_clk(I2S_NUM_0, SAMP_RATE, SAMP_BITS, I2S_CHANNEL_STEREO);
    vTaskDelay(10/portTICK_RATE_MS);
    audio_stream_start(&dd.read_i2s_stream->base);
    dd.detect_wakeword = true;
	
}

avsheth, 2018-11-01:
@xiaoseyihe  can you enable USE_SPI_MODE macro and try again?
for SDMMC mode you have to flip switches 1 and 2 ON (JP8)