espressif/esp-va-sdk #13: can not find Provision Network button on android apk

wydavid, 2018-10-28:
Hi avsheth

 i can find esp_alexa_xxxx on the phone and connect the soft ap fine
but can not find the "Provision Network " button on the app as you said on another issue
so, could you help to give some comment on this? thanks

avsheth, 2018-10-28:
@wydavid could you share a screenshot of the app(or describe what do you see on the app) after you connect to softAP from the WiFi settings and go back to app?

wydavid, 2018-10-29:
yes,after connet soft ap, and back to app, nothing find. only amazon link.button

wydavid, 2018-10-30:
please help to check phone log and module log

```
10-29 22:47:09.787 4954-4954/? I/art: Late-enabling -Xcheck:jni
10-29 22:47:09.814 4954-4954/com.espressif.provision D/Configuration: this.defaultFontName:null
    this.defaultFontName:null
10-29 22:47:09.820 4954-4954/com.espressif.provision D/ThemeResources: ThemeResources init: com.espressif.provision; pid=4954
10-29 22:47:09.827 4954-4954/com.espressif.provision W/System: ClassLoader referenced unknown path: /data/app/com.espressif.provision-1/lib/arm64
10-29 22:47:09.828 4954-4954/com.espressif.provision D/ActivityThread: BIND_APPLICATION handled : 0 / AppBindData{appInfo=ApplicationInfo{701038e com.espressif.provision}}
10-29 22:47:09.835 4954-4954/com.espressif.provision D/Configuration: this.defaultFontName:null
10-29 22:47:09.835 4954-4954/com.espressif.provision D/Configuration: this.defaultFontName:null
10-29 22:47:09.845 4954-4954/com.espressif.provision W/art: Before Android 4.1, method android.graphics.PorterDuffColorFilter android.support.graphics.drawable.VectorDrawableCompat.updateTintFilter(android.graphics.PorterDuffColorFilter, android.content.res.ColorStateList, android.graphics.PorterDuff$Mode) would have incorrectly overridden the package-private method in android.graphics.drawable.Drawable
10-29 22:47:09.898 4954-4954/com.espressif.provision D/Configuration: this.defaultFontName:null
10-29 22:47:09.898 4954-4954/com.espressif.provision D/ActivityThread: ACT-AM_ON_RESUME_CALLED ActivityRecord{857f7af token=android.os.BinderProxy@b45f7bc {com.espressif.provision/com.espressif.ui.MainActivity}}
10-29 22:47:09.901 4954-4954/com.espressif.provision D/WindowClient: Add to mViews: com.android.internal.policy.PhoneWindow$DecorView{9edf666 I.E...... R.....ID 0,0-0,0}, this = android.view.WindowManagerGlobal@6184112
10-29 22:47:09.902 4954-4954/com.espressif.provision D/OpenGLRenderer: Dumper init 5 threads <0x7f8af8f200>
10-29 22:47:09.903 4954-4954/com.espressif.provision D/OpenGLRenderer: <com.espressif.provision> is running.
10-29 22:47:09.905 4954-4969/com.espressif.provision D/OpenGLRenderer: Use EGL_SWAP_BEHAVIOR_PRESERVED: false
10-29 22:47:09.906 4954-4969/com.espressif.provision D/OpenGLRenderer: CanvasContext() 0x7f86632000
10-29 22:47:09.908 4954-4963/com.espressif.provision I/System: FinalizerDaemon: finalize objects = 1
10-29 22:47:09.913 4954-4954/com.espressif.provision I/[MALI][Gralloc]: [+]r_hnd(0x7f803604a0), client(30), share_fd(28)
10-29 22:47:09.913 4954-4954/com.espressif.provision D/GraphicBuffer: register, handle(0x7f803604a0) (w:1856 h:2048 s:1856 f:0x1 u:0x000100)
10-29 22:47:09.914 4954-4954/com.espressif.provision D/ViewRootImpl: hardware acceleration is enabled, this = ViewRoot{21360e3 com.espressif.provision/com.espressif.ui.MainActivity,ident = 0}
10-29 22:47:09.921 4954-4954/com.espressif.provision D/ActivityThread: ACT-LAUNCH_ACTIVITY handled : 0 / ActivityRecord{857f7af token=android.os.BinderProxy@b45f7bc {com.espressif.provision/com.espressif.ui.MainActivity}}
10-29 22:47:09.952 4954-4969/com.espressif.provision D/OpenGLRenderer: CanvasContext() 0x7f86632000 initialize window=0x7f8c865210, title=com.espressif.provision/com.espressif.ui.MainActivity
10-29 22:47:09.952 4954-4954/com.espressif.provision D/Surface: Surface::allocateBuffers(this=0x7f8c865200)
10-29 22:47:09.955 4954-4969/com.espressif.provision I/OpenGLRenderer: Initialized EGL, version 1.4
10-29 22:47:09.957 4954-4969/com.espressif.provision D/OpenGLRenderer: Created EGL context (0x7f86603580)
10-29 22:47:09.958 4954-4969/com.espressif.provision I/OpenGLRenderer: Get enable program binary service property (1)
    Initializing program atlas...
10-29 22:47:09.958 4954-4969/com.espressif.provision D/ProgramBinary/Service: BpProgramBinaryService.getFileDescriptor
10-29 22:47:09.959 4954-4969/com.espressif.provision D/ProgramBinary/Service: BpProgramBinaryService.getProgramMapLen
10-29 22:47:09.959 4954-4969/com.espressif.provision D/ProgramBinary/Service: BpProgramBinaryService.getProgramMapArray
10-29 22:47:09.959 4954-4969/com.espressif.provision D/ProgramBinary/Service: BpProgramBinaryService.getProgramBinaryLen
10-29 22:47:09.959 4954-4969/com.espressif.provision I/OpenGLRenderer: Program binary detail: Binary length is 173456, program map length is 152.
    Succeeded to mmap program binaries. File descriptor is 37, and path is /dev/ashmem�.
    No need to use file discriptor anymore, close fd(37).
10-29 22:47:09.960 4954-4969/com.espressif.provision D/OpenGLRenderer: Initializing program cache from 0x7f904d7bd8, size = 4
10-29 22:47:09.961 4954-4969/com.espressif.provision D/OpenGLRenderer: -- init (key = 0x0000000000000000)
    -- init (key = 0x0000000000500041)
10-29 22:47:09.962 4954-4969/com.espressif.provision D/OpenGLRenderer: -- init (key = 0x0000000800000003)
10-29 22:47:09.963 4954-4969/com.espressif.provision D/OpenGLRenderer: -- init (key = 0x0000003800000000)
10-29 22:47:09.963 4954-4969/com.espressif.provision D/MALI: eglCreateImageKHR:511: [Crop] 0 0 1856 2048  img[1856 2048] 
10-29 22:47:09.964 4954-4969/com.espressif.provision D/Surface: Surface::connect(this=0x7f8c865200,api=1)
10-29 22:47:09.965 4954-4969/com.espressif.provision W/libEGL: [ANDROID_RECORDABLE] format: 1
10-29 22:47:09.965 4954-4969/com.espressif.provision D/mali_winsys: new_window_surface returns 0x3000
10-29 22:47:09.977 4954-4969/com.espressif.provision I/PerfService: PerfServiceNative api init
[... cut ...]
I (13884) wifi: station: 74:ac:5f:8b:8c:1b join, AID=1, bgn, 20
I (13884) conn_mgr: Received event: 128
I (39924) wifi: station: 74:ac:5f:8b:8c:1b leave, AID = 1
I (39924) wifi: n:1 0, o:1 0, ap:1 1, sta:255 255, prof:1
I (39924) conn_mgr: Received event: 128
I (46494) wifi: n:1 0, o:1 0, ap:1 1, sta:255 255, prof:1
I (46494) wifi: station: 74:ac:5f:8b:8c:1b join, AID=1, bgn, 20
I (46494) conn_mgr: Received event: 128
I (46714) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.2
I (46714) conn_mgr: Received event: 128
```
`

wydavid, 2018-10-30:
[image]

avsheth, 2018-10-30:
@wydavid We suspect it might be that your app is getting killed somehow in between you connect to softAP and going back to the app. We have created a youtube video [here](https://www.youtube.com/watch?v=dTJGbwirTnU&feature=youtu.be) showing the entire provisioning flow. Can you just check once if something is missing at your end?
If it still doesn't work could you tell us your Android version and phone model?

wydavid, 2018-10-30:
got it, thanks very much

closed 2018-10-30 by wydavid