espressif/esp-va-sdk #15: build error at lyratd_msc_alexa_sr example

mkonnov, 2018-11-08:
Trying to build with
``` bash
IDF_PATH=`realpath ../../../esp-idf` make -j 8
```

Compiler vomits the following
``` bash
/home/f77e/projects/esp32-lyratd-msc_jetblack/libs/esp-avs-sdk/examples/lyratd_msc_alexa_sr/build/lwip/liblwip.a(mem.o): In function `mem_free':
/home/f77e/projects/esp32-lyratd-msc_jetblack/libs/esp-idf/components/lwip/lwip/src/core/mem.c:144: multiple definition of `mem_free'
/home/f77e/projects/esp32-lyratd-msc_jetblack/libs/esp-avs-sdk/examples/lyratd_msc_alexa_sr/build/misc/libmisc.a(mem_utils.o):/home/f77e/projects/esp32-lyratd-msc_jetblack/libs/esp-avs-sdk/components/misc/mem_utils.c:101: first defined here
/home/f77e/projects/esp32-lyratd-msc_jetblack/libs/esp-avs-sdk/examples/lyratd_msc_alexa_sr/components/codec_zl38063/lib/libcodec_zl38063.a(vproc_common.o):(.literal.VprocHALRead+0x8): undefined reference to `lwip_ntohs'
/home/f77e/projects/esp32-lyratd-msc_jetblack/libs/esp-avs-sdk/examples/lyratd_msc_alexa_sr/components/codec_zl38063/lib/libcodec_zl38063.a(vproc_common.o): In function `VprocHALRead':
/builds/app-frameworks/esp-alexa/esp-audio-sdk-hal/esp_codec/zl38063/components/codec_zl38063/dsp_zl38063/vproc_common.c:136: undefined reference to `lwip_ntohs'
collect2: error: ld returned 1 exit status
/home/f77e/projects/esp32-lyratd-msc_jetblack/libs/esp-idf/make/project.mk:435: recipe for target '/home/f77e/projects/esp32-lyratd-msc_jetblack/libs/esp-avs-sdk/examples/lyratd_msc_alexa_sr/build/alexa.elf' failed
make: *** [/home/f77e/projects/esp32-lyratd-msc_jetblack/libs/esp-avs-sdk/examples/lyratd_msc_alexa_sr/build/alexa.elf] Error 1

```

avsheth, 2018-11-09:
@mkonnov It appears you are using esp-idf's master branch. Can you please check with release/v3.1 branch of esp-idf?

mkonnov, 2018-11-16:
@avsheth Hi !
Sorry for long response.

I've tried to do so with https://github.com/espressif/esp-idf/commit/020ade652d9b6b0b87b0bebe176a150cc4407749  and it works.

closed 2018-12-19 by kedars