espressif/esp-va-sdk #16: Example with support for tunein?

arlaor, 2018-11-08:
Hi.
Hello, I am using the ESP32-Lyrat module, and I would like to know if there is any developed code or example that allows TUNEIN to be used?

avsheth, 2018-11-09:
@arlaor TuneIn is supported by the SDK, you do not need any specific changes in the app code for that. So you can use default application available and can start using Amazon music and/or TuneIn.
We haven't tested VTuner yet though.

arlaor, 2018-11-09:
Hi @avsheth 
In this moment i have mount [lyrat_alexa](https://github.com/espressif/esp-avs-sdk/tree/master/examples/lyrat_alexa) example, but I could not connect using the espressif application to esp32-lyrat. Which example can I mount to my module to use tunein? i´m used android lineage 16.0 in mi samsung GTI-9192. in the application I do not get provision screen.
Which example has the greatest emphasis on using tunein in esp32-lyrat?

avsheth, 2018-11-10:
@arlaor 
1. Please checkout this [link](https://www.youtube.com/watch?v=dTJGbwirTnU&feature=youtu.be) to help you get through provisioning. Let us know if you still face any issues. 
2. You can use lyrat_alexa or lyrat_alexa_sr to play TuneIn stations. As soon as your board is provisioned you can wake the device using 'Alexa' (if you are using lyart_alexa_sr application) or you can press Rec button on the board to start recording your query (For both apps. Alexa is not required to be spoken in this case). And ask 'Play <station name> on TuneIn".
3. Unfortunately SDK doesn't have any example application which can load/play personalised radio station links.

arlaor, 2018-11-10:
Hi @avsheth 

I will perform tests with the shown in the video and I will be commenting. For what is shown in the video I have omitted to apply the [esp-idf-patches](https://github.com/espressif/esp-avs-sdk/tree/master/esp-idf-patches).
In the changelog v1.0b1r4 (Beta) - 2018-09-28, "Support for PLS playlist parser" is mentioned. Can this be done from the Lyrat_alexa example or separately?

arlaor, 2018-11-27:
Hi @avsheth 
I tried to follow the instructions given in the video, but when applying the patch time 0:31 sec
patch -p1 esp-avs-sdk/esp-idf-patches/memset-i2s-dma-buffers-zero.patch (location for me), I get the message -bash: patch: the order was not found
and therefore I believe that it does not make any modification that includes the patch.
I'm using mingW32.exe in windows 7. Do you have any suggestions to apply the patch?
so far the result is the beginning of this topic

avsheth, 2018-11-27:
Hi @arlaor
1. In the command above you've missed redirection operator after -p1, it should be patch -p1 < esp-avs-sdk/esp-idf-patches/memset-i2s-dma-buffers-zero.patch. Probably that is why you are getting the error
2. Regardless on mingw you'll have to install "patch" package using command "mingw-get install msys-patch" to apply patches. 
2. OR alternatively you can use command "git apply esp-avs-sdk/esp-idf-patches/memset-i2s-dma-buffers-zero.patch" if you don't want to install the package.

arlaor, 2018-11-29:
Hi @avsheth avsheth 

This repetitive message is for internet connection slow velocity?
*
rst:0x10 (RTCWDT_RTC_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6320
ho 0 tail 12 room 4
load:0x40078000,len:10096
load:0x40080000,len:6700
entry 0x4008034c
[0;32mI (31) boot: ESP-IDF v3.1.1-54-g87f810dca-dirty 2nd stage bootloader[0m
[0;32mI (31) boot: compile time 23:16:29[0m
[0;32mI (33) boot: Enabling RNG early entropy source...[0m
[0;32mI (37) qio_mode: Enabling QIO for flash chip GD[0m
[0;32mI (42) boot: SPI Speed      : 80MHz[0m
[0;32mI (47) boot: SPI Mode       : QIO[0m
[0;32mI (51) boot: SPI Flash Size : 4MB[0m
[0;32mI (55) boot: Partition Table:[0m
[0;32mI (58) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (73) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (80) boot:  2 avs              WiFi data        01 02 00010000 00006000[0m
[0;32mI (88) boot:  3 factory          factory app      00 00 00020000 00300000[0m
[0;32mI (95) boot: End of partition table[0m
[0;32mI (100) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x146b6c (1338220) map[0m
[0;32mI (462) esp_image: segment 1: paddr=0x00166b94 vaddr=0x3ffb0000 size=0x050cc ( 20684) load[0m
[0;32mI (469) esp_image: segment 2: paddr=0x0016bc68 vaddr=0x3ffb50cc size=0x00000 (     0) load[0m
[0;32mI (469) esp_image: segment 3: paddr=0x0016bc70 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (478) esp_image: segment 4: paddr=0x0016c078 vaddr=0x40080400 size=0x03f98 ( 16280) load[0m
[0;32mI (492) esp_image: segment 5: paddr=0x00170018 vaddr=0x400d0018 size=0xff0a8 (1044648) map[0m
[0;32mI (772) esp_image: segment 6: paddr=0x0026f0c8 vaddr=0x40084398 size=0x15e30 ( 89648) load[0m
[0;32mI (802) esp_image: segment 7: paddr=0x00284f00 vaddr=0x400c0000 size=0x00000 (     0) load[0m
[0;32mI (802) esp_image: segment 8: paddr=0x00284f08 vaddr=0x50000000 size=0x00000 (     0) load[0m
[0;32mI (824) boot: Loaded app from partition at offset 0x20000[0m
[0;32mI (824) boot: Disabling RNG early entropy source...[0m
[0;32mI (825) spiram: SPI RAM mode: flash 80m sram 80m[0m
[0;32mI (830) spiram: PSRAM initialized, cache is in low/high (2-core) mode.[0m
[0;32mI (837) cpu_start: Pro cpu up.[0m
[0;32mI (841) cpu_start: Starting app cpu, entry point is 0x400814cc[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (1347) spiram: SPI SRAM memory test OK[0m
[0;32mI (1347) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (1348) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (1354) heap_init: At 3FFBF358 len 00020CA8 (131 KiB): DRAM[0m
[0;32mI (1360) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
[0;32mI (1366) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (1373) heap_init: At 4009A1C8 len 00005E38 (23 KiB): IRAM[0m
[0;32mI (1379) cpu_start: Pro cpu start user code[0m
[0;32mI (1384) spiram: Adding pool of 4096K of external SPI memory to heap allocator[0m
[0;32mI (1392) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations[0m
[0;32mI (71) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (71) alexa: ==== Alexa SDK version: v1.0b1-172316 ====[0m
Registering command: mem-dump 
[0;32mI (121) scli: Initialising UART on port 0[0m
[0;32mI (121) uart: queue free spaces: 8[0m

>>Registering command: task-dump 
Registering command: nvs-get 
Registering command: nvs-set 
Registering command: nvs-erase 
Registering command: reboot 
Registering command: heap-trace 
I (141) wifi: wifi driver task: 3ffc6df4, prio:23, stack:3584, core=0
I (141) wifi: wifi firmware version: a063b8c
I (151) wifi: config NVS flash: disabled
[... 106 lines cut ...]
[sid: 1] Response code: 200
[sid: 1] Response content type is multipart/related; boundary=------abcde123; type=application/json
Final boundary:------abcde123:
[0;32mI (7401) [http_transport]: AVS level connction has now been established[0m
[0;32mI (7411) [http_transport]: Enqueueing avs_tx_sent for post_connect_object[0m
[0;32mI (7411) [http_stream]: [stream-new] 105152 spiram 1281416[0m
[0;32mI (7421) [http_transport]: New stream event: {"context":[{"header":{"namespace":"AudioPlayer","name":"PlaybackState"},"payload":{"token":"","offsetInMilliseconds":0,"playerActivity":"IDLE"}},{"header":{"namespace":"SpeechSynthesizer","name":"SpeechState"},"payload":{"token":"","offsetInMilliseconds":0,"playerActivity":"FINISHED"}},{"header":{"namespace":"SpeechRecognizer","name":"RecognizerState"},"payload":{"wakeword":"ALEXA"}},{"header":{"namespace":"Speaker","name":"VolumeState"},"payload":{"volume":40,"muted":false}},{"header":{"namespace":"Alerts","name":"AlertsState"},"payload":{"allAlerts":[],"activeAlerts":[]}},{"header":{"namespace":"Notifications","name":"IndicatorState"},"payload":{"isEnabled":false,"isVisualIndicatorPersisted":false}},{"header":{"namespace":"AudioActivityTracker","name":"ActivityState"},"payload":{}}],"event":{"header":{"namespace":"System","name":"SynchronizeState","messageId":"4bae5328-17cd-48e6-8b89-30d19f2e463b"},"payload":{}}}[0m
[... cut ...]
[sid: 3] Response code: 204
[0;32mI (8111) [http_transport]: [sid 3] stream close[0m
[0;32mI (8111) [http_stream]: [stream-delete] 104628 spiram 1287328[0m
W (8211) wifi: discard, ba not setup
W (8311) wifi: discard, ba not setup
W (9541) wifi: discard, ba not setup
W (12101) wifi: discard, ba not setup
W (17121) wifi: discard, ba not setup
W (23161) wifi: discard, ba not setup
W (35141) wifi: discard, ba not setup
W (36061) wifi: discard, ba not setup
W (39651) wifi: discard, ba not setup
*

avsheth, 2018-11-30:
Access Point

arlaor, 2018-11-30:
My router is connected in bridge mode.

avsheth, 2018-11-30:
Okay. I believe setting router in bridge mode shouldn't be an issue. Can you help me with some other setup related queries below:
1. Are you using any repeater ?
2. Can you tell me name of router manufacturer, is it from D-Link, TP-Link or some other? And if possible it's model number too

Also if you have or could arrange a router from some other manufacturer, can you check if you are seeing the same issue with that too?

arlaor, 2018-11-30:
Hi @avsheth 
The main router is a huawei adsl HG8546M. To this I connect with a Xiaomi mini router in bridge mode. The Xiomi mini router has the PADAVAN firmware installed (modified and adapted version of openwrt).
Although my ESP32-LYRAT connects to alexa and I can interact with it, the message
W (xxxx) wifi: discard, ba not setup
is presented in most cases after restarting the LYRAT module or turning it off and on again
I'm sorry, currently I do not have another router to perform tests.

avsheth, 2018-12-04:
would it be possible to directly connect the board to main router and see if the problem persists?
currently we do not have this setup hence couldn't cross check the issue.

arlaor, 2019-01-07:
Hi @avsheth 
So far I have not been able to perform tests with the board connected directly to the router. I have to wait to do it.
Other question:
Esp-avs-sdk has support for the [aithinker module esp32-A1S](http://wiki.ai-thinker.com/esp32-audio-kit)  or are you planning to support this board?

avsheth, 2019-01-28:
Hi @arlaor Unfortunately we do not support aithinker module in our SDK yet. However, you can make changes to app_lyrat_cb.c in lyrat or lyrat_sr app to make it work.
You might need to change i2s write calls in va_app_playback_data function to write to DAC of ESP32.

avsheth, 2019-01-28:
Hi @arlaor 
There isn't any documentation for this as such. But we are in a process of restructuring SDK to make is easy for users add support for a custom board (maybe with custom codec and DSP). This should be available by mid next week. I'll let you know once it is in place.

avsheth, 2020-01-19:
Hi @arlaor 
Support to add a custom board have been available for quite some time now. You may refer to it to add your own board. Closing this issue. If you face any issues, kindly open another one.

closed 2020-01-19 by avsheth