espressif/esp-va-sdk #18: default aws_iot mqtt example doesnt publish to topic (lyrat_alexa_sr)

kvbulusu, 2018-11-22:
Hi, 
The MQTT example is not publishing to the topic...It does subscribe and receive when I publish to the topic from AWS IOT Console client. I compared with AWS IOT Embedded C sdk example and if I comment the following , it publishes. I'm not an expert, so want to know what the below code is trying to do here in the code?

 if (xSemaphoreTake(wake_word_sem, 1000/portTICK_RATE_MS) != pdTRUE) {
                continue;
        }

Any suggestions or pointers?

Also if I want to set the clientid and topic name in sdkconfig, how should I do it ?

avsheth, 2018-11-23:
@kvbulusu 
The example in  the code publishes wake word count to AWS console.
Code is waiting for "Alexa" wakeword to be detected and when it does get the wakeword, it would put that semaphore and will go on to publish to AWS IoT.
If you need to modify the example to publish data either on some other events or periodically, you can remove above if statement.
As for setting clientID and topic name in sdkconfig, you'll have to those parameter in Kconfig.projbuild and modify parameters to aws_iot_mqtt_subscribe API accordingly.

kvbulusu, 2018-11-26:
@avsheth Thanks for the explanation. Was able to modify the code and adapt to my needs.

closed 2018-11-26 by kvbulusu