espressif/esp-va-sdk #19: Can't find sources for avs-sdk releases

mkonnov, 2018-12-03:
Hey guys !
I may not understand smth, but as topic says, I can't find any source code in https://github.com/espressif/esp-avs-sdk/releases.

Is there any way I can build that application images by myself ?
Thanks !

avsheth, 2018-12-04:
Hi @mkonnov 
You can download the source from https://github.com/espressif/esp-avs-sdk. 
You can either clone the git repo or download as zip and compile and run it.
Android apk for provisioning and Amazon account authentication can be found at https://github.com/espressif/esp-avs-sdk/releases/download/v1.0b1r3/esp-prov-v2.apk

mkonnov, 2018-12-04:
@avsheth, thanks for reply.

Not sure you understood me the correct way.
I'm talking about the  source code of release, specifically avs-ESP32LyraTDMSC-beta3.

I'm downloading the archive containing "the source code" of that release, but there's nothing except README and "doc" dir.

So how can I rebuild the same binaries ?

avsheth, 2018-12-04:
Oh. Thanks for bringing it to the notice. That appear to be broken, need to check.
As of now you can download/clone the git repo and build lyratd msc app from the same.

mkonnov, 2018-12-04:
Ok, now it seems a bit more clear :+1: 

I've tried the "lyratd_msc_alexa_sr" example, but it seems to be completely different from the binaries that's related to the release I mentioned. 
Though it's not a goal of current conversation, they has different user creds provisioning process and generally, the code that builds from the "examples" are not working, nowever I've tried to set it up using related Android app.
Can you please suggest me smth I can try ?
Thanks !

avsheth, 2018-12-04:
Can you try running `make erase_flash`, this will erase your entire flash and re-flash everything using `make flash`.
After which you can follow this [link](https://www.youtube.com/watch?v=dTJGbwirTnU&feature=youtu.be) to get started with newer provisioning. Hope this helps.

mkonnov, 2018-12-04:
Yep, every time I'm rebuilding another source, I'm erasing the flash.

Thanks for pointing to video ! 
Will try this out today and report the result.

mkonnov, 2018-12-04:
Ok, I've walked through a process described in video (applied patches, rebuild the sources using idf v3.1 branch, erased and flashed the binaries), but still can't provide the WiFi creds using a dedicated Android app.
@avsheth , can you please walk through that process and say if that error makes itself know on my side or it sits somewhere in app or code ?

avsheth, 2018-12-04:
can you tell how far have you been successfully able to proceed?
and if you can dump the console log as well that would be helpful

mkonnov, 2018-12-04:
Ooops sorry, forgot to did that.
In the Espressif app, I've passed "Login with Amazon screen", then connected to ESP-Alexa-xxx AP and tried to provide my WiFi creds.

Logs:
```
I (28204) wifi: n:1 0, o:1 1, ap:1 1, sta:255 255, prof:1
I (52494) wifi: n:1 1, o:1 0, ap:1 1, sta:255 255, prof:1
I (52494) wifi: station: 00:0a:f5:93:53:90 join, AID=1, bgn, 40U

I (52494) conn_mgr: Received event: 128
I (52914) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.2
I (52914) conn_mgr: Received event: 128
I (117434) protocomm_httpd: Session ID: 63
I (117664) security1: Device pubkey:
I (117664) security1: ac 7f 66 d2 75 48 43 bb ec 84 9f a3 cb ae 84 6b 
I (117664) security1: 73 2a 15 07 88 19 c7 1a 19 66 87 4a af 5b 4b 19 
I (117664) security1: Client pubkey:
I (117674) security1: 2b 43 5a 51 e9 d0 ea 34 16 6a 08 fd 71 5b ca b7 
I (117674) security1: 3b 70 87 ca f8 10 91 26 55 ac b0 45 8d a3 e8 5f 
I (117914) security1: Shared key:
I (117914) security1: 09 95 3b c5 ff 7f 7e 31 38 82 9b 98 05 95 43 f0 
I (117914) security1: f1 50 77 cc 14 d2 e0 25 33 82 d6 5d 09 cc 83 36 
I (117924) security1: Device random:
I (117924) security1: fa 99 b6 03 c5 48 73 f8 0b f4 1f 40 ab 57 16 05 
I (117934) protocomm_httpd: ADI Done
I (118024) protocomm_httpd: Session ID: 63
I (118024) security1: Client verifier:
I (118024) security1: ad 23 32 25 15 eb 09 05 db 3f 75 17 fc 0b 8c e5 
I (118024) security1: b6 b2 df 9b 29 56 df 57 3e 24 21 03 d9 25 63 4b 
I (118034) security1: Dec Client verifier:
I (118044) security1: 47 12 a1 36 b8 d8 e1 5d 48 c1 6d 19 31 4a 4d 43 
I (118044) security1: b0 6d 3f 03 13 42 d6 6b b5 d2 bd b9 68 e6 ab 1c 
E (118054) security1: Key mismatch. Close connection
E (118064) security1: Session setup error -1
E (118064) protocomm_httpd: data handler failed
I (118074) protocomm_httpd: ADI Done
W (118074) httpd_uri: httpd_uri: uri handler execution failed
I (118094) protocomm_httpd: Session ID: 63
I (118094) security1: Client verifier:
I (118104) security1: ad 23 32 25 15 eb 09 05 db 3f 75 17 fc 0b 8c e5 
I (118104) security1: b6 b2 df 9b 29 56 df 57 3e 24 21 03 d9 25 63 4b 
I (118114) security1: Dec Client verifier:
I (118114) security1: fe 38 75 75 10 5f 0f 51 6f 31 1d 8b 73 b2 5b 15 
I (118124) security1: 35 8b 22 00 d2 35 c8 17 3a 0a 1f 2f b3 28 d1 c1 
E (118134) security1: Key mismatch. Close connection
E (118134) security1: Session setup error -1
E (118144) protocomm_httpd: data handler failed
I (118144) protocomm_httpd: ADI Done
W (118154) httpd_uri: httpd_uri: uri handler execution failed
 ```

mkonnov, 2018-12-05:
Hi @avsheth !
Again, can you please try the provisioning process on your side ?
I'm stucking with that problem alone and just don't know the direction for investigation.

Would appreciate to you sooo much if you'll give me any result.
Thanks !

mkonnov, 2018-12-07:
@anujdeshpande yes, I have disabled it. Will try now with PoP set up by default.

anujdeshpande, 2018-12-07:
Awesome! If you've got everything working you can close this issue!

closed 2018-12-07 by mkonnov