espressif/esp-va-sdk #20: NVS stored data of wifi/AVS provisioning

robertoimai, 2018-12-15:
What are the keys/values I need to read/write to NVS to access provisioning of wifi and AVS settings?
I'm assuming they are stored in the avs partition, right?
I'm trying to create my own application to provision wifi/AVS and pass these settings through UART to my ESP firmware.
Provisioning every device I make using the current connection manager just takes too long and is not practical.
I need to create a script that will automate this whole provisioning process.

robertoimai, 2018-12-16:
I found the keys I need to use, but I also think there is a script already there for this purpose.
I don't think this is enabled on the example provided though.
It is my understanding that esp_prov.py allows us to provision it with cli, but the example code only enables softap provisioning. Am I correct?
I also found out that esp_prov.py on master branch of IDF is somewhat different than the one provided on AVS-SDK.
So, at this point, how do I enable the cli provisioning on AVS-SDK?
Does esp_prov.py allow for avs provisioning too? If so, how do I do it?
If not, how can I do avs provisioning after I have wifi provisioned?
When I run:
python esp_prov.py --ssid myssid --passphrase mypassword --sec_ver 1 --pop abcd1234 --prov_mode cli
I get this:
==== Esp_Prov Version: 0.1 ====
==== Starting Session ====
Client->Device msg 10015a25a201220a206beb223bd8d35689a4443c68f32c40148384dd269a5fb05374fe6936aee34276
Enter device->client msg 
But if I paste 10015a25a201220a206beb223bd8d35689a4443c68f32c40148384dd269a5fb05374fe6936aee34276 to the serial monitor of the example provided, nothing happens.
I am still at the console prompt with just >> waiting for a console command, which only accepts:
Registering command: task-dump
Registering command: nvs-get
Registering command: nvs-set
Registering command: nvs-erase
Registering command: reboot
Registering command: heap-trace

Can someone help out on getting this figured out?

avsheth, 2018-12-17:
Hi @robertoimai 
If you need to flash both wifi and avs tokens to multiple device, I recommend you take a look at nvs_partition_generator and utility of ESP-IDF. You can create a csv like below:

`key,type,encoding,value
cm,namespace,,
ssid,data,string,<your SSID>
pass,data,string,<your passphrase>
config,data,i8,1`

You can create multiple such CSVs and provide each to partition generator utility, which would generate corresponding binary. You can flash these binaries using esptool.py on various devices at address 0x9000. 
Unfortunately, there was a regression introduced in the latest release where setting nvs keyval pairs no longer sets/changes AP credentials. We are working on fixing it.

As for AVS credentials, you can keep refresh token and client ID in NVS. You can append below rows and columns in the above csv for Alexa credentials.

`avs,namespace,,
refreshToken,data,string,<your refresh token>
clientID,data,string,<your client ID>`

I hope this answers your question.

closed 2018-12-18 by robertoimai