espressif/esp-va-sdk #26: ESP32-LyraTD-MSC A v2.2 d block Failed to connect to ESP32: Timed out waiting for packet header

vijaypacketp, 2018-12-27:
Hi All, 
We got an ESP32-LyraTD-MSC A v2.2 dev board and  followed ESP32-LyraTD-MSC V2.2 Getting Started Guide to set up, after setting up as per document
It reported error when I ran the following steps.
1), Connected it to laptop through USB cable.
2), Turned the board on.
3), Ran "make flash".
4), The following messages were printed out on terminal:"make flash

Flashing binaries to serial port COM9 (app at offset 0x10000 )...
esptool.py v2.6-beta1
Serial port COM9
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
make: *** [/home/vkm/esp/esp-idf/components/esptool_py/Makefile.projbuild:63: flash] Error 2

Note: I connected  the console throgh COM9 using teraterm/minicom and am able to see the existing application logs, there I press Reset button and release the Boot button
and made sure that

rst:0x1 (POWERON_RESET),boot:0x7 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download

I also even tried with flash_download_tools_v3.6.5  and seen following message 

....._____....._____....._____....._____....._____....._____....._____....._____....._____....._____[2018-12-27 15:17:22
,753][ESP8266Loader_spi[1]][espDownloader.py][line:351][ERROR]: Chip sync error: Failed to connect to ESP32: Timed out w
aiting for packet header
[2018-12-27 15:17:22,755][ESP8266Loader_spi[1]][espDownloader.py][line:439][ERROR]: ESP32 Chip sync error esp_sync_block
ing.

Could you please tell us what is missing in terms of configuration to detect the flash 
Thank You
Kumar

avsheth, 2018-12-27:
Hi @vijaypacketp 
Just to confirm, you had pressed RST button while keeping BOOT button pressed when you start flashing, right?

vijaypacketp, 2018-12-27:
yes @avsheth  I have done it  and I can see below message 
rst:0x1 (POWERON_RESET),boot:0x7 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download

ameyinamdar, 2018-12-27:
Kumar, please note that you can't use the console while you're trying to program the device.

vijaypacketp, 2018-12-27:
Sorry  @ameyinamdar  I forgot to mention, I made sure to see the "wait for download message" and close the console then I  continued and given the  make flash

vijaypacketp, 2019-01-13:
Thanks All for valuable suggestions, looks like I received defective board, We order two more new boards they are working fine, able to flash the Firmware.

Thanks,
Kumar

closed 2019-01-13 by vijaypacketp