espressif/esp-va-sdk #27: 这个例程经常连接wifi后崩溃自动复位

ZKL11G, 2018-12-29:


ZKL11G, 2018-12-29:
rst:0x1 (POWERON_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6512
ho 0 tail 12 room 4
load:0x40078000,len:10096
load:0x40080000,len:6700
entry 0x4008034c
[0;32mI (31) boot: ESP-IDF v3.1.1-80-ga18f92b09-dirty 2nd stage bootloader[0m
[0;32mI (31) boot: compile time 15:38:39[0m
[0;32mI (35) boot: Enabling RNG early entropy source...[0m
[0;32mI (37) qio_mode: Enabling QIO for flash chip GD[0m
[0;32mI (42) boot: SPI Speed      : 40MHz[0m
[0;32mI (46) boot: SPI Mode       : QIO[0m
[0;32mI (50) boot: SPI Flash Size : 4MB[0m
[0;32mI (55) boot: Partition Table:[0m
[0;32mI (58) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (73) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (80) boot:  2 avs              WiFi data        01 02 00010000 00006000[0m
[0;32mI (88) boot:  3 factory          factory app      00 00 00020000 00300000[0m
[0;32mI (95) boot: End of partition table[0m
[0;32mI (99) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x167240 (1471040) map[0m
[0;32mI (545) esp_image: segment 1: paddr=0x00187268 vaddr=0x3ffb0000 size=0x04e4c ( 20044) load[0m
[0;32mI (552) esp_image: segment 2: paddr=0x0018c0bc vaddr=0x3ffb4e4c size=0x00000 (     0) load[0m
[0;32mI (553) esp_image: segment 3: paddr=0x0018c0c4 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (561) esp_image: segment 4: paddr=0x0018c4cc vaddr=0x40080400 size=0x03b44 ( 15172) load[0m
[0;32mI (576) esp_image: segment 5: paddr=0x00190018 vaddr=0x400d0018 size=0x10a2fc (1090300) map[0m
[0;32mI (903) esp_image: segment 6: paddr=0x0029a31c vaddr=0x40083f44 size=0x15f1c ( 89884) load[0m
[0;32mI (936) esp_image: segment 7: paddr=0x002b0240 vaddr=0x400c0000 size=0x00000 (     0) load[0m
[0;32mI (936) esp_image: segment 8: paddr=0x002b0248 vaddr=0x50000000 size=0x00000 (     0) load[0m
[0;32mI (958) boot: Loaded app from partition at offset 0x20000[0m
[0;32mI (958) boot: Disabling RNG early entropy source...[0m
[0;32mI (959) spiram: SPI RAM mode: flash 80m sram 80m[0m
[0;32mI (964) spiram: PSRAM initialized, cache is in low/high (2-core) mode.[0m
[0;32mI (971) cpu_start: Pro cpu up.[0m
[0;32mI (974) cpu_start: Starting app cpu, entry point is 0x400814c8[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (1481) spiram: SPI SRAM memory test OK[0m
[0;32mI (1481) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (1482) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (1488) heap_init: At 3FFC07F0 len 0001F810 (126 KiB): DRAM[0m
[0;32mI (1494) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
[0;32mI (1500) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (1507) heap_init: At 40099E60 len 000061A0 (24 KiB): IRAM[0m
[0;32mI (1513) cpu_start: Pro cpu start user code[0m
[0;32mI (1518) spiram: Adding pool of 4096K of external SPI memory to heap allocator[0m
[0;32mI (1526) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations[0m
[0;32mI (205) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (205) alexa: ==== Alexa SDK version: v1.0b1-172316 ====[0m
Registering command: mem-dump 
[0;32mI (255) scli: Initialising UART on port 0[0m
[0;32mI (255) uart: queue free spaces: 8[0m

>>Registering command: task-dump 
Registering command: nvs-get 
Registering command: nvs-set 
Registering command: nvs-erase 
Registering command: reboot 
Registering command: heap-trace 
I (275) wifi: wifi driver task: 3ffc8594, prio:23, stack:3584, core=1
I (275) wifi: wifi firmware version: 38990be
[... 61 lines cut ...]
[0;32mI (10335) [alexa]: Done getting current time: 1546045050[0m
[0;32mI (10355) [tls_certification]: Done setting global CA store[0m
[0;32mI (10355) audio_stream: Starting playback_writer stream[0m
Starting audio stream playback_writer
[0;32mI (10355) audio_stream: Stream playback_writer Event Started[0m
[0;32mI (10395) LYRAT_CB: Current mode is: 8[0m
[0;32mI (10395) [dialog]: Entering ALEXA_IDLE[0m
[0;31mE (10395) avs-nvs-util: Error opening avs namespace![0m
[0;32mI (10395) [speaker]: Using default volume[0m
[0;32mI (10405) LYRAT_CB: Volume changed to 40[0m
[0;32mI (10405) auth-delegate-config: No NVS partition initialised[0m
[0;31mE (10415) auth-delegate-config: No AVS static config partition found[0m
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40098d9d  PS      : 0x00060830  A0      : 0x800dbdaa  A1      : 0x3f821610  
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3f8218e0  
A10     : 0x00000003  A11     : 0x00060823  A12     : 0x00060820  A13     : 0x3f8222a0  
A14     : 0x00000000  A15     : 0x00000080  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x40098d9d  LEND    : 0x40098dad  LCOUNT  : 0xffffffff  

Backtrace: 0x40098d9d:0x3f821610

Rebooting...
ets Jun  8 201

ZKL11G, 2018-12-29:
板子是lyrat 4.3，例程是lyrat_alexa_sr

avsheth, 2018-12-31:
Hi @ZKL11G This occurs when the device is unable to find any Amazon tokens in NVS. To get rid of this you need to reset your device to factory mode and do provisioning using Android/iOS app.

P.S:  git pull latest changes, since it would allow you to use reset-to-factory push button when you hit this error. Otherwise you can also perform reset-to-factory using either esptool_py (erase_region 0x9000 0x6000) if you are using CLI,  or windows flasher tool.

ZKL11G, 2019-01-03:
> Hi @ZKL11G This occurs when the device is unable to find any Amazon tokens in NVS. To get rid of this you need to reset your device to factory mode and do provisioning using Android/iOS app.
>
> P.S: [...]

rst:0x1 (POWERON_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6512
ho 0 tail 12 room 4
load:0x40078000,len:10096
load:0x40080000,len:6700
entry 0x4008034c
[0;32mI (31) boot: ESP-IDF v3.1.1-80-ga18f92b09-dirty 2nd stage bootloader[0m
[0;32mI (31) boot: compile time 11:22:22[0m
[0;32mI (33) boot: Enabling RNG early entropy source...[0m
[0;32mI (37) qio_mode: Enabling QIO for flash chip GD[0m
[0;32mI (42) boot: SPI Speed      : 40MHz[0m
[0;32mI (47) boot: SPI Mode       : QIO[0m
[0;32mI (51) boot: SPI Flash Size : 4MB[0m
[0;32mI (55) boot: Partition Table:[0m
[0;32mI (58) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (66) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (73) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (80) boot:  2 factory          factory app      00 00 00010000 00300000[0m
[0;32mI (88) boot: End of partition table[0m
[0;31mE (92) esp_image: image at 0x10000 has invalid magic byte[0m
[0;33mW (98) esp_image: image at 0x10000 has invalid SPI mode 255[0m
[0;33mW (104) esp_image: image at 0x10000 has invalid SPI size 15[0m
[0;31mE (111) boot: Factory app partition is not bootable[0m
[0;31mE (116) boot: No bootable app partitions in the partition table[0m
user code done

修改之后log如下，还是没法配网

ZKL11G, 2019-01-03:
> Hi @ZKL11G This occurs when the device is unable to find any Amazon tokens in NVS. To get rid of this you need to reset your device to factory mode and do provisioning using Android/iOS app.
>
> P.S: [...]

配网成功后，连接aws失败，这是log

[0;31mE (67242) [auth-delegate]: Failed to connect to Auth URL "https://api.amazon.com/auth/O2/token"[0m
[0;32mI (67722) conn_mgr_prov: Stopping provisioning[0m
[0;31mE (67752) [auth-delegate]: Authentication attempt failed...retrying ![0m
[0;31mE (67752) [auth-delegate]: Please check connectivity and/or credentials[0m
[0;31mE (67752) [auth-delegate]: In case of authentication failure try resetting the device to factory mode and restart provisioning[0m

ZKL11G, 2019-01-03:
rst:0xc (SW_CPU_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6512
ho 0 tail 12 room 4
load:0x40078000,len:10096
load:0x40080000,len:6700
entry 0x4008034c
[0;32mI (31) boot: ESP-IDF v3.1.1-80-ga18f92b09-dirty 2nd stage bootloader[0m
[0;32mI (31) boot: compile time 11:22:22[0m
[0;32mI (33) boot: Enabling RNG early entropy source...[0m
[0;32mI (37) qio_mode: Enabling QIO for flash chip GD[0m
[0;32mI (42) boot: SPI Speed      : 40MHz[0m
[0;32mI (47) boot: SPI Mode       : QIO[0m
[0;32mI (51) boot: SPI Flash Size : 4MB[0m
[0;32mI (55) boot: Partition Table:[0m
[0;32mI (58) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (73) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (80) boot:  2 factory          factory app      00 00 00010000 00300000[0m
[0;32mI (88) boot: End of partition table[0m
[0;32mI (92) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1681ac (1474988) map[0m
[0;32mI (539) esp_image: segment 1: paddr=0x001781d4 vaddr=0x3ffb0000 size=0x04f00 ( 20224) load[0m
[0;32mI (546) esp_image: segment 2: paddr=0x0017d0dc vaddr=0x3ffb4f00 size=0x00000 (     0) load[0m
[0;32mI (547) esp_image: segment 3: paddr=0x0017d0e4 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (555) esp_image: segment 4: paddr=0x0017d4ec vaddr=0x40080400 size=0x02b24 ( 11044) load[0m
[0;32mI (568) esp_image: segment 5: paddr=0x00180018 vaddr=0x400d0018 size=0x10af94 (1093524) map[0m
[0;32mI (898) esp_image: segment 6: paddr=0x0028afb4 vaddr=0x40082f24 size=0x16f80 ( 94080) load[0m
[0;32mI (932) esp_image: segment 7: paddr=0x002a1f3c vaddr=0x400c0000 size=0x00000 (     0) load[0m
[0;32mI (932) esp_image: segment 8: paddr=0x002a1f44 vaddr=0x50000000 size=0x00000 (     0) load[0m
[0;32mI (954) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (954) boot: Disabling RNG early entropy source...[0m
[0;32mI (955) spiram: SPI RAM mode: flash 80m sram 80m[0m
[0;32mI (960) spiram: PSRAM initialized, cache is in low/high (2-core) mode.[0m
[0;32mI (967) cpu_start: Pro cpu up.[0m
[0;32mI (971) cpu_start: Starting app cpu, entry point is 0x400814cc[0m
[0;32mI (962) cpu_start: App cpu up.[0m
[0;32mI (1477) spiram: SPI SRAM memory test OK[0m
[0;32mI (1478) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (1478) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (1484) heap_init: At 3FFC0860 len 0001F7A0 (125 KiB): DRAM[0m
[0;32mI (1491) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
[0;32mI (1497) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (1503) heap_init: At 40099EA4 len 0000615C (24 KiB): IRAM[0m
[0;32mI (1510) cpu_start: Pro cpu start user code[0m
[0;32mI (1515) spiram: Adding pool of 4096K of external SPI memory to heap allocator[0m
[0;32mI (1523) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations[0m
[0;32mI (201) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (202) alexa: ==== Alexa SDK version: v1.0b1-177299 ====[0m
Registering command: mem-dump 
[0;32mI (262) scli: Initialising UART on port 0[0m
[0;32mI (262) uart: queue free spaces: 8[0m

>>Registering command: task-dump 
Registering command: nvs-get 
Registering command: nvs-set 
Registering command: nvs-erase 
Registering command: reboot 
Registering command: heap-trace 
Registering command: button 
Registering command: wifi-set
I (282) wifi: wifi driver task: 3ffc8604, prio:23, stack:3584, core=1
I (282) wifi: wifi firmware version: 38990be
I (292) wifi: config NVS flash: enabled
I (292) wifi: config nano formating: disabled
[0;32mI (302) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (312) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[... 77 lines cut ...]
[0;32mI (60482) dsp: Reader stream event 2[0m
[0;32mI (60492) [alexa]: Waiting for time to be updated[0m
[0;32mI (60502) [alexa]: Done getting current time: 1546487573[0m
[0;32mI (60552) [tls_certification]: Done setting global CA store[0m
[0;32mI (60552) audio_stream: Starting playback_writer stream[0m
Starting audio stream playback_writer
[0;32mI (60552) audio_stream: Stream playback_writer Event Started[0m
[0;32mI (60592) LYRAT_CB: Current mode is: 8[0m
[0;32mI (60592) [dialog]: Entering ALEXA_IDLE[0m
[0;31mE (60592) avs-nvs-util: Error opening avs namespace![0m
[0;32mI (60592) [speaker]: Using default volume[0m
[0;32mI (60602) LYRAT_CB: Volume changed to 40[0m
[0;32mI (60602) auth-delegate-config: Returning auth delegate companion app configuration[0m
[0;32mI (60612) [auth-delegate]: Auth delegate with comp app ANykCsgEokudUNvTyEap amzn1.application-oa2-client.a5b0d0c6a5f546499d1e5043f3515e6b amzn://com.espressif.provision abcd1234[0m
[0;31mE (60702) esp-tls: Failed to create socket (family 2 socktype 1 protocol 0)[0m
[0;31mE (60702) [auth-delegate]: Failed to connect to Auth URL "https://api.amazon.com/auth/O2/token"[0m
[0;31mE (61212) [auth-delegate]: Authentication attempt failed...retrying ![0m
[... cut ...]
[0;31mE (61242) esp-tls: Failed to create socket (family 2 socktype 1 protocol 0)[0m
[0;31mE (61242) [auth-delegate]: Failed to connect to Auth URL "https://api.amazon.com/auth/O2/token"[0m
[0;31mE (61752) [auth-delegate]: Authentication attempt failed...retrying ![0m
[0;31mE (61752) [auth-delegate]: Please check connectivity and/or credentials[0m
[0;31mE (61752) [auth-delegate]: In case of authentication failure try resetting the device to factory mode and restart provisioning[0m
[0;31mE (61782) esp-tls: Failed to create socket (family 2 socktype 1 protocol 0)[0m

ZKL11G, 2019-01-03:
[0;32mI (253372) subpub: Stack remaining for task 'aws_iot_task' is 1780 bytes[0m
[0;32mI (254472) subpub: Stack remaining for task 'aws_iot_task' is 1780 bytes[0m
[0;32mI (255572) subpub: Stack remaining for task 'aws_iot_task' is 1780 bytes[0m
[0;32mI (256672) subpub: Stack remaining for task 'aws_iot_task' is 1780 bytes[0m
[0;32mI (257772) subpub: Stack remaining for task 'aws_iot_task' is 1780 bytes[0m
[0;32mI (258872) subpub: Stack remaining for task 'aws_iot_task' is 1780 bytes[0m
[0;32mI (259972) subpub: Stack remaining for task 'aws_iot_task' is 1780 bytes[0m
[0;32mI (261072) subpub: Stack remaining for task 'aws_iot_task' is 1780 bytes[0m
[0;32mI (262172) subpub: Stack remaining for task 'aws_iot_task' is 1780 bytes[0m

closed 2019-01-03 by ZKL11G