espressif/esp-va-sdk #29: implicit declaration of function 'esp_bt_mem_release'

robertoimai, 2018-12-29:
I'm getting this error when building:
/home/owner/esp/esp-avs-sdk/components/conn_mgr_prov/src/conn_mgr_prov.c: In function 'conn_mgr_prov_mem_release':
/home/owner/esp/esp-avs-sdk/components/conn_mgr_prov/src/conn_mgr_prov.c:144:21: error: implicit declaration of function 'esp_bt_mem_release' [-Werror=implicit-function-declaration]
     esp_err_t err = esp_bt_mem_release(ESP_BT_MODE_BTDM);
                     ^

dhrishi, 2018-12-31:
@robertoimai There are some optimisations done for Bluetooth, which are not a part of ESP-IDF release/v3.1 branch but are recently added in 3.2 version and hence the compilation failure. Sorry for missing this in the documentation.
Can you please checkout release/v3.2 branch on Github esp-idf and give it a try? 
The official v3.2 release should be out in a couple of weeks and we will then move to that.

robertoimai, 2018-12-31:
Sorry, but this didn't go well :(
The patch failed.

owner@azena-dev:~/esp/esp-idf$ git apply ../esp-avs-sdk/esp-idf-patches/esp-tls-Add-support-for-global-CA-store.-All-mbedtls.patch
error: patch failed: components/esp-tls/esp_tls.c:21
error: components/esp-tls/esp_tls.c: patch does not apply
error: patch failed: components/esp-tls/esp_tls.h:54
error: components/esp-tls/esp_tls.h: patch does not apply
error: patch failed: components/mbedtls/port/include/mbedtls/esp_config.h:114
error: components/mbedtls/port/include/mbedtls/esp_config.h: patch does not apply

dhrishi, 2018-12-31:
@robertoimai This patch is already a part of release/v3.2 branch. So you don't need to apply that. As of now, apply the other patch: memset-i2s-dma-buffers-zero.patch. This patch is currently not in 3.2 but will also reflect soon.

robertoimai, 2018-12-31:
Ok, I'm on 3.2 and it doesn't complain anymore about that.
I am able to compile and load my code.
But, I am getting a different problem now.

I (34160) auth-delegate-config: Returning auth delegate companion app configuration
I (34170) [auth-delegate]: Auth delegate with comp app ANWGMUVqXgeavLhKlSYh amzn1.application-oa2-client.a5b0d0c6a5f546499d1e5043f3515e6b amzn://com.espressif.provision abcd1234
E (34220) esp-tls: Failed to create socket (family 2 socktype 1 protocol 0)
E (34220) esp-tls: Failed to open new connection
E (34220) [auth-delegate]: Failed to connect to Auth URL "https://api.amazon.com/auth/O2/token"
E (34740) [auth-delegate]: Authentication attempt failed...retrying !
E (34740) [auth-delegate]: Please check connectivity and/or credentials
E (34750) [auth-delegate]: In case of authentication failure try resetting the device to factory mode and restart provisioning
E (34760) esp-tls: Failed to create socket (family 2 socktype 1 protocol 0)
E (34770) esp-tls: Failed to open new connection
E (34800) [auth-delegate]: Failed to connect to Auth URL "https://api.amazon.com/auth/O2/token"
E (35300) [auth-delegate]: Authentication attempt failed...retrying !
E (35300) [auth-delegate]: Please check connectivity and/or credentials
E (35300) [auth-delegate]: In case of authentication failure try resetting the device to factory mode and restart provisioning
E (35310) esp-tls: Failed to create socket (family 2 socktype 1 protocol 0)
E (35340) esp-tls: Failed to open new connection
E (35340) [auth-delegate]: Failed to connect to Auth URL "https://api.amazon.com/auth/O2/token"
E (35850) [auth-delegate]: Authentication attempt failed...retrying !
E (35850) [auth-delegate]: Please check connectivity and/or credentials
E (35850) [auth-delegate]: In case of authentication failure try resetting the device to factory mode and restart provisioning
E (35880) esp-tls: Failed to create socket (family 2 socktype 1 protocol 0)
E (35890) esp-tls: Failed to open new connection
E (35890) [auth-delegate]: Failed to connect to Auth URL "https://api.amazon.com/auth/O2/token"
E (36400) [auth-delegate]: Authentication attempt failed...retrying !
E (36400) [auth-delegate]: Please check connectivity and/or credentials
E (36420) [auth-delegate]: In case of authentication failure try resetting the device to factory mode and restart provisioning
E (36440) esp-tls: Failed to create socket (family 2 socktype 1 protocol 0)
E (36440) esp-tls: Failed to open new connection
I (36440) conn_mgr_prov_handler: Connected state
E (36450) [auth-delegate]: Failed to connect to Auth URL "https://api.amazon.com/auth/O2/token"
E (36980) [auth-delegate]: Authentication attempt failed...retrying !
E (36980) [auth-delegate]: Please check connectivity and/or credentials
E (36980) [auth-delegate]: In case of authentication failure try resetting the device to factory mode and restart provisioning
E (36990) esp-tls: Failed to create socket (family 2 socktype 1 protocol 0)
E (37020) esp-tls: Failed to open new connection
E (37020) [auth-delegate]: Failed to connect to Auth URL "https://api.amazon.com/auth/O2/token"
I (37450) conn_mgr_prov: Stopping provisioning
I (37460) conn_mgr_prov: BT stack memory released
I (37460) wifi: station: c0:d3:c0:32:39:4f leave, AID = 1, bss_flags is 134243
I (37460) wifi: n:9 0, o:9 2, ap:9 2, sta:9 0, prof:9
I (37460) wifi: mode : sta (b4:e6:2d:b6:fa:31)
I (37470) conn_mgr_prov: Provisioning stopped
E (37530) [auth-delegate]: Authentication attempt failed...retrying !
E (37530) [auth-delegate]: Please check connectivity and/or credentials
E (37530) [auth-delegate]: In case of authentication failure try resetting the device to factory mode and restart provisioning
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4009a56b  PS      : 0x00060a30  A0      : 0x80151e70  A1      : 0x3ffd1200
0x4009a56b: strstr at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/string/../../../.././newlib/libc/string/strstr.c:55

A2      : 0x00000100  A3      : 0x3f53a470  A4      : 0x00000138  A5      : 0x3ffb8e04
A6      : 0x00006d6f  A7      : 0x00000013  A8      : 0x3ffca830  A9      : 0x00000001
A10     : 0x3f53e120  A11     : 0x00000000  A12     : 0x3ffb807a  A13     : 0x3ffb807a
A14     : 0x00006d6f  A15     : 0x7a616d61  SAR     : 0x00000018  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000100  LBEG    : 0x40099fb0  LEND    : 0x40099fbb  LCOUNT  : 0x00000000
0x40099fb0: memset at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memset.S:142

0x40099fbb: memset at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memset.S:152

Backtrace: 0x4009a56b:0x3ffd1200 0x40151e6d:0x3ffd1220 0x401018c7:0x3ffd1270 0x40101bc5:0x3ffd12a0 0x40101e00:0x3ffd12e0 0x401356ac:0x3ffd1300 0x400ff757:0x3ffd1340 0x400ffa86:0x3ffd1380 0x400ffce0:0x3ffd13d0 0x400f8e18:0x3ffd1400 0x400d7840:0x3ffd1440 0x400d2bb8:0x3ffd14f0 0x40090cc9:0x3ffd1510
0x4009a56b: strstr at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/string/../../../.././newlib/libc/string/strstr.c:55

0x40151e6d: mbedtls_x509_crt_parse at /home/owner/esp/esp-idf/components/mbedtls/mbedtls/library/x509_crt.c:1285

[... cut ...]
0x401356ac: http_connection_new at /home/owner/esp/esp-avs-sdk/components/httpc/httpc.c:277

0x400ff757: basic_player_play_url at ??:?

0x400ffa86: basic_player_play_url at ??:?

0x400ffce0: auth_delegate_init at ??:?

0x400f8e18: alexa_init at ??:?

0x400d7840: app_main at /home/owner/esp/esp-avs-sdk/examples/gemini_alexa_sr/main/app_main.c:135

0x400d2bb8: main_task at /home/owner/esp/esp-idf/components/esp32/cpu_start.c:492

0x40090cc9: vPortTaskWrapper at /home/owner/esp/esp-idf/components/freertos/port.c:435

robertoimai, 2019-01-02:
I tried this with or without BLE enabled with same results.
I also tried with the msc demo board as well as my custom hardware with same results.
I even tried the plain alexa msc example without any modification to the code and got same results.
I think the issue is within the pre-compiled alexa library not liking something from IDF 3.2

avsheth, 2019-01-21:
Hi @robertoimai Sorry for the late reply.
I tried to reproduce this issue but unfortunately not being able to reproduce it with release/v3.2 branch and lyratd_msc_sr application. I tried even by turning on BT from menuconfig but still it worked fine.
Can you tell if you are still facing this issue? If so, can you try commenting conn_mgr_prov_mem_release from app_main.c ?

robertoimai, 2019-01-28:
Due to the fact that we were having issues with IRAM, we decided to hold off on adding any voice interaction at this time. We will be using ADF for now because we ran out of time to develop a stable firmware for the launch of our product.

closed 2019-01-28 by robertoimai

myvobot, 2019-03-06:
I encountered with:
esp-tls: Failed to create socket (family 2 socktype 1 protocol 0)
and solved by adding a line before alexa_init(va_cfg):

`
vTaskDelay(2000 / portTICK_RATE_MS);
`

The problem was due to conn_mgr_prov.c will hold for 1000ms before stopping the provisioning task. Thus soft ap mode is still active and none of the http/https request is possible.

myvobot, 2019-03-07:
Also crash in mbedtls_x509_crt_parse, the cert's length & buf pointer are invalid.
The problem was related to ESP-IDF version. If I use release/v3.1, no problem at all.