espressif/esp-va-sdk #34: Provisioning errors ( mbedtls_ecdh_gen_public with error code : -0x4f80)

kvbulusu, 2019-01-15:
Havent seen this error before. Not sure why its failing? Any suggestions what to look for?

I (17508) wifi: new:<1,0>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
I (17508) wifi: station: 84:7a:88:09:b2:f1 join, AID=1, bgn, 20
W (17508) event: unexpected system event 63!
E (17518) event: mismatch or invalid event, id=63
E (17518) event: default event handler failed!
I (19238) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.2
E (49068) security1: Failed at mbedtls_ecdh_gen_public with error code : -0x4f80
E (49068) security1: Session setup error -1
E (49068) protocomm_httpd: Data handler failed
W (49078) httpd_uri: httpd_uri: uri handler execution failed

I'm using latest esp-idf and esp-va-sdk.

avsheth, 2019-01-15:
Are you on the esp-idf master branch by any chance?

kvbulusu, 2019-01-15:
Yes...
%git branch | grep \* | cut -d ' ' -f2
master

avsheth, 2019-01-15:
1. The SDK is tested with release/v3.1 branch of esp-idf actually. Can you try moving to release/v3.1 branch?
2. If you are still having issues, can you reverting any changes that you might have made on top of the SDK?

kvbulusu, 2019-01-16:
ok..It works after I moved...But some of the errors below were not present in earlier releases: Right now its connected to wifi/provisioning is complete, but not sure what these errors mean and if they have unintentional effects later on? Can we ignore these ?

W (34153) event: unexpected system event 63!
E (34153) event: mismatch or invalid event, id=63
E (34163) event: default event handler failed!

avsheth, 2019-01-21:
Great. Please ignore that error string for now. It won't affect overall functionality of the device.

closed 2019-01-28 by avsheth