espressif/esp-va-sdk #35: cant hear audio(alexa response) with (v1.0b1r7) release...

kvbulusu, 2019-01-19:
I'm using latest branch  and flashed lyrat_sr code ...Didnt add any of my code...After provisioning, I cant hear any audio when I ask a question to alexa... Didnt see the issues with earlier releases (v1.0b1r4) ..anything should I look for ?

I (1574951) [dialog]: Dialog new
I (1574951) [focus_manager]: Acquire channel: Dialog
I (1574951) LYRAT_CB: Current mode is: 4
I (1574951) [speech_recognizer]: On Focus: FOREGROUND
I (1574951) [dialog]: Entering VA_LISTENING
I (1574961) [sys_playback]: Acquire
I (1574971) [focus_manager]: Now active channel: Dialog
I (1574971) dsp: Flushed store data: 3840

I (1575021) subpub: Subscribe callback
I (1575021) subpub: test_topic/esp32	{ "wake_word_count" : 7 }
I (1575031) [http_stream]: [stream-new] 59420 spiram 1141188
I (1575031) [http_transport]: New stream event: {"context":[{"header":{"namespace":"Speaker","name":"VolumeState"},"payload":{"volume":40,"muted":false}},{"header":{"namespace":"SpeechRecognizer","name":"RecognizerState"},"payload":{"wakeword":"ALEXA"}},{"header":{"namespace":"SpeechSynthesizer","name":"SpeechState"},"payload":{"token":"amzn1.as-ct.v1.HomeAutomation#ACRI#DeviceTTSRendererV4_ac169946-10b0-4666-b399-9c8f1926a3e5","offsetInMilliseconds":3480,"playerActivity":"FINISHED"}},{"header":{"namespace":"AudioPlayer","name":"PlaybackState"},"payload":{"token":"","offsetInMilliseconds":0,"playerActivity":"IDLE"}},{"header":{"namespace":"AudioActivityTracker","name":"ActivityState"},"payload":{"dialog":{"interface":"SpeechSynthesizer","idleTimeInMilliseconds":394014}}},{"header":{"namespace":"Alerts","name":"AlertsState"},"payload":{"allAlerts":[],"activeAlerts":[]}},{"header":{"namespace":"Notifications","name":"IndicatorState"},"payload":{"isEnabled":false,"isVisualIndicatorPersisted":false}}],"event":{"header":{"namespace":"SpeechRecognizer","name":"Recognize","messageId":"5e42e56e-2abd-4abf-afa5-6f4732894598","dialogRequestId":"4ccc9981-afcc-45b7-9df6-5e467334c141"},"payload":{"profile":"NEAR_FIELD","format":"AUDIO_L16_RATE_16000_CHANNELS_1","initiator":{"type":"TAP","payload":{}}}}}
I (1575111) subpub: Stack remaining for task 'aws_iot_task' is 1780 bytes
I (1575141) [http_stream]: stream POST /v20160207/events sid:35
sys_playback: Got event stopped
I (1576241) subpub: Stack remaining for task 'aws_iot_task' is 1780 bytes
post_data_provider: Returning 48320 bytes on stream ID 35
post_data_provider: Returning 24320 bytes on stream ID 35
I (1577341) subpub: Stack remaining for task 'aws_iot_task' is 1780 bytes
post_data_provider: Returning 12160 bytes on stream ID 35
post_data_provider: Returning 6720 bytes on stream ID 35
post_data_provider: Returning 6720 bytes on stream ID 35
post_data_provider: Returning 6720 bytes on stream ID 35
post_data_provider: Returning 6720 bytes on stream ID 35
I (1578441) subpub: Stack remaining for task 'aws_iot_task' is 1780 bytes
post_data_provider: Returning 6720 bytes on stream ID 35
post_data_provider: Returning 6720 bytes on stream ID 35
post_data_provider: Returning 6720 bytes on stream ID 35
post_data_provider: Returning 6720 bytes on stream ID 35
post_data_provider: Returning 6720 bytes on stream ID 35
I (1579541) subpub: Stack remaining for task 'aws_iot_task' is 1780 bytes
post_data_provider: Returning 6720 bytes on stream ID 35
post_data_provider: Returning 6720 bytes on stream ID 35
post_data_provider: Returning 6720 bytes on stream ID 35
post_data_provider: Returning 6720 bytes on stream ID 35
post_data_provider: Returning 6720 bytes on stream ID 35
I (1580641) subpub: Stack remaining for task 'aws_iot_task' is 1780 bytes
post_data_provider: Returning 6720 bytes on stream ID 35
I (1580661) [http_stream]: Speculatively stopping capture
I (1580661) [http_stream]: Part end: JSON
JSON Data: {"directive":{"header":{"namespace":"SpeechRecognizer","name":"StopCapture","messageId":"0192d3f8-f267-4e84-b839-5c3370fe4585","dialogRequestId":"4ccc9981-afcc-45b7-9df6-5e467334c141"},"payload":{}}}
I (1580661) [http_stream]: Part begin
Namespace: SpeechRecognizer, Name: StopCapture
Got immediate stop of ring buffer. Sent 0 bytes
speech_recognizer: Sending stop command
Setting rb to stop immediately.
I (1580701) dsp: Sending stop command
I (1580721) dsp: Stopped I2S audio stream
I (1580721) [dialog]: Listen End
I (1580721) LYRAT_CB: Current mode is: 8
I (1580721) [dialog]: Entering VA_IDLE
[sid: 35] Response code: 204
I (1580941) [dialog]: Response 204
I (1580941) [focus_manager]: Release channel: Dialog
I (1580951) [sys_playback]: Release
I (1580951) [http_transport]: [sid 35] stream close
I (1580951) [http_stream]: [stream-delete - 35] sram 59084 spiram 1147008
sys_playback: Got event stopped

avsheth, 2019-01-19:
Can you check if you are getting response code 204 every time?
Response code 204 indicates that Alexa was not able to recognize the query.

kvbulusu, 2019-01-28:
I'm not seeing the issue anymore...closing it. Thanks for letting me know what 204 response means.

closed 2019-01-28 by kvbulusu