espressif/esp-va-sdk #36: Bluetooth microphone

Quanghoster, 2019-01-26:
Hi,
Is it possible to connect a bluetooth device in place of the Lyrat onboard microphones?

avsheth, 2019-01-28:
Hi @Quanghoster I didn't get you exactly. Do you mean you want to use any bluetooth headphone to talk to Alexa?

Dinesh-Ramakrishnan, 2019-02-05:
@avsheth 
Can we connect a bluetooth headset (with mic) to the Lyrat board and let the audio out and audio in drive through this bluetooth device?

Quanghoster, 2019-02-05:
I was thinking the same thing. Partly for a crude location service so that the system would know which lyrat board the user was closest to. Either using Bluetooth as above ir maybe an ibeacon to sense closest device and the stream the audio over WiFi. Depends which approach uses less power. However,  I've had trouble including Bluetooth support with this software became it's linked against an older version of the sdk

avsheth, 2020-01-19:
Sorry for quite a late reply :). Newer SDK version supports A2DP sink , using which you can listen to Alexa responses over a BT headphone/speaker. However getting commands over BT mic isn't yet supported.
For making Alexa work with a BT mic, there is a different protocol called AMA. However it relays the query to the phone, which in turn forwards it to the cloud on behalf of the user.
As for your particular need of replacing on-board mics with BT mics, that requires quite some work and may not be supported for a while. :(

Hope this helps.

Thanks

closed 2020-01-19 by avsheth