espressif/esp-va-sdk #39: any possibility to reduce libalexa.a file size?

kvbulusu, 2019-02-27:
We are trying to have ota partition and struggling with the size/space available...When we looked at the components/file sizes using make size-files, size-components , libalexa.a popped..Not sure how its compiled , but just want to explore the possibility if further optimizations are possible when creating that ...We need to reduce by 250KB and hoping to get more insight into how to compile that by ourselves or if anything else we can try...we already excluded few components that we really dont use ...  BTW, this is not a bug, but more of a qn.

avsheth, 2019-05-09:
Hi @kvbulusu 
Sorry about the delayed reply. We are in the process of updating github with latest release.
Newer library is compiled with gcc's size optimization flag. I will provide an update when latest release is uploaded.

kedars, 2019-05-17:
The release is now out. While the build flags do shrink the library, some additional completeness features and certification fixes may offset that gain. Please check if this works for you. Marking this as closed for now, please reopen in case of additional questions.

kvbulusu, 2019-05-19:
@kedars Thanks for getting back..  Main reason I asked, I was hoping to reduce/optimize the sound files ..When I did make size-files, I guess most of them from libalexa ...

Per-file contributions to ELF file:
             Object File DRAM .data & .bss   IRAM Flash code & rodata   Total
med_system_alerts_melodi          0      0      0          0   194474  194474
med_system_alerts_melodi          0      0      0          0   194474  194474
med_ui_endpointing.wav.t          0      0      0          0   134445  134445
med_system_alerts_melodi          0      0      0          0   128073  128073
med_system_alerts_melodi          0      0      0          0   128045  128045
med_ui_wakesound.wav.txt          0      0      0          0   124305  124305
  wakeNet3_model_float.o          0      0      0        158   107071  107229
med_alerts_notification_          0      0      0          0    93741   93741
med_ui_wakesound_touch.w          0      0      0          0    90657   90657
med_state_privacy_mode_o          0      0      0          0    86449   86449
med_state_privacy_mode_o          0      0      0          0    67245   67245
volume_sound_custom.wav.          0      0      0          0    57901   57901
med_alerts_notification_          0      0      0          0    57002   57002
med_alerts_notification_          0      0      0          0    46250   46250

Are there instructions how to compile this lib and  their sources? If you cant publish, its ok.

kedars, 2019-05-20:
Ah, right. You might have tried this already, if not, please try commenting out the call     alexa_tone_enable_larger_tones(); from the app_main.c. This should exclude some of the larger tones that get pulled in. I will take an action item to provide a way to remove all tones in a subsequent release