espressif/esp-va-sdk #40: Problem in Authentication token

kmakhil90, 2019-04-03:
Hi,

Why i am getting following error while booting lyratd-msr sample code from esp-va-sdk?

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6756
load:0x40078000,len:12116
ho 0 tail 12 room 4
load:0x40080400,len:7400
entry 0x40080780
[0;32mI (30) boot: ESP-IDF v4.0-dev-141-g106dc0590-dirty 2nd stage bootloader[0m
[0;32mI (30) boot: compile time 18:05:11[0m
[0;32mI (31) boot: Enabling RNG early entropy source...[0m
[0;32mI (37) qio_mode: Enabling default flash chip QIO[0m
[0;32mI (42) boot: SPI Speed      : 80MHz[0m
[0;32mI (46) boot: SPI Mode       : QIO[0m
[0;32mI (50) boot: SPI Flash Size : 4MB[0m
[0;32mI (54) boot: Partition Table:[0m
[0;32mI (58) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (73) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (80) boot:  2 factory          factory app      00 00 00010000 00300000[0m
[0;32mI (88) boot: End of partition table[0m
[0;32mI (92) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x18c750 (1623888) map[0m
[0;32mI (530) esp_image: segment 1: paddr=0x0019c778 vaddr=0x3ffb0000 size=0x03898 ( 14488) load[0m
[0;32mI (535) esp_image: segment 2: paddr=0x001a0018 vaddr=0x400d0018 size=0x115f88 (1138568) map[0m
[0;32mI (837) esp_image: segment 3: paddr=0x002b5fa8 vaddr=0x3ffb3898 size=0x01030 (  4144) load[0m
[0;32mI (839) esp_image: segment 4: paddr=0x002b6fe0 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (844) esp_image: segment 5: paddr=0x002b73e8 vaddr=0x40080400 size=0x1a568 (107880) load[0m
[0;32mI (904) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (904) boot: Disabling RNG early entropy source...[0m
[0;32mI (905) spiram: Found 64MBit SPI RAM device[0m
[0;32mI (909) spiram: SPI RAM mode: flash 80m sram 80m[0m
[0;32mI (915) spiram: PSRAM initialized, cache is in low/high (2-core) mode.[0m
[0;32mI (922) cpu_start: Pro cpu up.[0m
[0;32mI (926) cpu_start: Application information:[0m
[0;32mI (930) cpu_start: Project name:     alexa[0m
[0;32mI (935) cpu_start: App version:      ESP-VA-v1.0b1r7-2-g063d26f-dirt[0m
[0;32mI (942) cpu_start: Compile time:     Apr  1 2019 18:05:30[0m
[0;32mI (948) cpu_start: ELF file SHA256:  4c4f99fe9520278a...[0m
[0;32mI (954) cpu_start: ESP-IDF:          v4.0-dev-141-g106dc0590-dirty[0m
[0;32mI (961) cpu_start: Starting app cpu, entry point is 0x40081628[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (1456) spiram: SPI SRAM memory test OK[0m
[0;32mI (1457) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (1457) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (1463) heap_init: At 3FFBC258 len 00023DA8 (143 KiB): DRAM[0m
[0;32mI (1469) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (1476) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (1482) heap_init: At 4009A968 len 00005698 (21 KiB): IRAM[0m
[0;32mI (1488) cpu_start: Pro cpu start user code[0m
[0;32mI (1493) spiram: Adding pool of 4078K of external SPI memory to heap allocator[0m
[0;32mI (171) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (172) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations[0m
[0;32mI (172) app_main: ==== Voice Assistant SDK version: v1.0b1-178987 ====[0m
Registering command: mem-dump 
[0;32mI (212) scli: Initialising UART on port 0[0m
[0;32mI (212) uart: queue free spaces: 8[0m

>> Registering command: task-dump 
Registering command: cpu-dump 
Registering command: heap-trace 
Registering command: nvs-get 
Registering command: nvs-set 
Registering command: nvs-erase 
Registering command: reboot 
Registering command: wifi-set
ret: 0
I (232) wifi: wifi driver task: 3ffd0564, prio:23, stack:3584, core=1
I (242) wifi: wifi firmware version: 9c8361c
I (242) wifi: config NVS flash: enabled
I (242) wifi: config nano formating: disabled
[0;32mI (252) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (262) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
I (292) wifi: Init dynamic tx buffer num: 32
I (292) wifi: Init data frame dynamic rx buffer num: 32
I (292) wifi: Init management frame dynamic rx buffer num: 32
I (292) wifi: Init management short buffer num: 32
I (292) wifi: Init static tx buffer num: 32
I (302) wifi: Init static rx buffer size: 2212
I (302) wifi: Init static rx buffer num: 16
I (302) wifi: Init dynamic rx buffer num: 32
[0;32mI (312) conn_mgr_prov: Found ssid Uchi_Babu[0m
[0;32mI (312) conn_mgr_prov: Found password babu_uchi[0m
[0;32mI (322) app_main: Already provisioned, starting station[0m
[0;32mI (412) phy: phy_version: 4100, 6fa5e27, Jan 25 2019, 17:02:06, 0, 0[0m
I (412) wifi: mode : sta (30:ae:a4:c3:8a:5c)
I (532) wifi: new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:11
I (1292) wifi: state: init -> auth (b0)
I (1292) wifi: state: auth -> assoc (0)
I (1292) wifi: state: assoc -> run (10)
I (1312) wifi: connected with Uchi_Babu, channel 11, bssid = b0:4e:26:03:37:1f
I (1312) wifi: pm start, type: 1

[0;32mI (3732) event: sta ip: 192.168.0.149, mask: 255.255.255.0, gw: 192.168.0.1[0m
[0;32mI (3732) app_main: Connected with IP Address:192.168.0.149[0m
[0;32mI (3732) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3[0m
[0;32mI (3742) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3[0m
[0;32mI (3742) I2S: PLL_D2: Req RATE: 48000, real rate: 48076.000, BITS: 16, CLKM: 13, BCK: 8, MCLK: 12292917.167, SCLK: 1538432.000000, diva: 64, divb: 1[0m
[0;32mI (3762) SPI: b=1
[0m
[0;32mI (3762) SPI: 3- Loading the config file into the device RAM....
[0m
[0;32mI (4122) SPI: Device boot loading completed successfully...
[0m
[0;32mI (4122) gpio: GPIO[22]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (4132) dsp: Created I2S audio stream[0m
[0;32mI (4132) audio_stream: Starting i2s_reader stream[0m
Starting audio stream i2s_reader
[0;33mW (4182) I2S: I2S driver already installed[0m
.......................Init i2s stream...............
[0;32mI (4192) audio_stream: Stream i2s_reader Event Started[0m
[0;32mI (4192) dsp: Reader stream event 2[0m
Stopping audio stream i2s_reader
[0;32mI (4212) audio_stream: Stream i2s_reader Event Stopped[0m
[0;32mI (4212) dsp: Reader stream event 3[0m
Generated stopped event for stream i2s_reader
[0;32mI (4212) I2S: PLL_D2: Req RATE: 48000, real rate: 48076.000, BITS: 16, CLKM: 13, BCK: 8, MCLK: 12292917.167, SCLK: 1538432.000000, diva: 64, divb: 1[0m
Starting audio stream i2s_reader
[0;33mW (4242) I2S: I2S driver already installed[0m
.......................Init i2s stream...............
[0;32mI (4242) audio_stream: Stream i2s_reader Event Started[0m
[0;32mI (4242) dsp: Reader stream event 2[0m
[0;32mI (4252) [alexa]: Waiting for time to be updated[0m
[0;32mI (4582) [alexa]: Done getting current time: 1554268373[0m
[0;32mI (4602) [tls_certification]: Done setting global CA store[0m
[0;32mI (4602) auth-delegate-config: No Client ID or Refresh token in NVS[0m
[0;31mE (4602) [alexa]: Couldn't find valid authentication tokens. Please try resetting board to factory mode and do provisioning with companion app[0m

Any inputs help me to go further.

avsheth, 2019-05-09:
Please try erasing the NVS partition and retry provisioning with companion app.
This basically says that it was able to find wifi credentials but not the Amazon tokens, and as of now app cannot do it selectively, i.e provide either wifi or amazon tokens.

kedars, 2019-05-17:
Closing this now. Please re-open if this doesn't fix the issue for you.

closed 2019-05-17 by kedars

kmakhil90, 2019-05-17:
Latest idf update solved the problem it seems.