espressif/esp-va-sdk #46: Esp-va-SDK.... *** multiple target patterns. Stop. "AMAZON_ALEXA"

Yuvaraj4041, 2019-05-28:
Hi

**I was trying to compile "AMAZON_ALEXA" example under "esp-va-sdk".**

I am following the below steps.
Step 1 : Cloning ESP-IDF
Step 2 : Cloning ESP-VA-SDK by using "$ git clone https://github.com/espressif/esp-va-sdk.git"
Step 3 : Export COM PORT by "export ESPPORT= COM71"
Step 4 : Export IDF_PATH by "export IDF_PATH=C:/esp32/esp-idf"
Step 5 : Export Audio Board Path by "export AUDIO_BOARD_PATH=C:/esp32/esp-va-sdk/board_support_pkgs/lyratd_msc/audio_board/audio_board_lyratd_msc"

After this i was trying to compile using "make -j 8 flash monitor or make flash" command...

But it is not compiling and it is giving error for "make -j 8 flash monitor",
Yuvaraj@Yuvaraj-PC MINGW32 /c/esp32/esp-va-sdk/examples/amazon_alex a
$ make -j 8 flash monitor
Python requirements from C:/esp32/esp-idf/requirements.txt are satisfied.
DEFCONFIG
MENUCONFIG
configuration written to /c/esp32/esp-va-sdk/examples/amazon_alexa/sdkconfig
*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.
GENCONFIG
**App "alexa" version: 1.0-RC1
C:/esp32/esp-idf/make/project.mk:519: *** multiple target patterns. Stop.**

for make flash
Yuvaraj@Yuvaraj-PC MINGW32 /c/esp32/esp-va-sdk/examples/amazon_alexa
$ make flash
**C:/esp32/esp-idf/make/project.mk:519: *** multiple target patterns. Stop.**

Please anyone help me to solve this problem.

avsheth, 2019-05-28:
Hi @Yuvaraj4041 
On windows, you will have to set IDF_PATH enclosed in a double quote, if it has "C:" in the path. This is because in Makefile multiple targets could be separated via ':' and hence the error.
e.g. export IDF_PATH="C:/esp32/esp-idf". Please try setting it this way and clean all the example artifacts i.e ` rm -rf examples/amazon_alexa/build examples/amazon_alexa/sdkconfig examples/amazon_alexa/sdkconfig.old` and recompile.

Yuvaraj4041, 2019-05-28:
Hi @avsheth 

I am following below Steps: 
Yuvaraj@Yuvaraj-PC MINGW32 ~
$ cd c:/

Yuvaraj@Yuvaraj-PC MINGW32 /c
$ cd esp32/

Yuvaraj@Yuvaraj-PC MINGW32 /c/esp32
$ git clone --recursive https://github.com/espressif/esp-va-sdk.git
Cloning into 'esp-va-sdk'...

Yuvaraj@Yuvaraj-PC MINGW32 /c/esp32
$ cd esp-va-sdk/

Yuvaraj@Yuvaraj-PC MINGW32 /c/esp32/esp-va-sdk
$ cd examples/

Yuvaraj@Yuvaraj-PC MINGW32 /c/esp32/esp-va-sdk/examples
$ cd amazon_alexa/

Yuvaraj@Yuvaraj-PC MINGW32 /c/esp32/esp-va-sdk/examples/amazon_alexa
$ **export IDF_PATH="C:/esp32/esp-idf"**

Yuvaraj@Yuvaraj-PC MINGW32 /c/esp32/esp-va-sdk/examples/amazon_alexa
$ **export PATH="/mingw32/bin:$PATH"**

Yuvaraj@Yuvaraj-PC MINGW32 /c/esp32/esp-va-sdk/examples/amazon_alexa
$ **export AUDIO_BOARD_PATH="C:/esp32/esp-va-sdk/board_support_pkgs/lyratd_msc/audio_board/audio_board_lyratd_msc"**

Yuvaraj@Yuvaraj-PC MINGW32 /c/esp32/esp-va-sdk/examples/amazon_alexa
$ **make menuconfig**
MENUCONFIG
*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.
GENCONFIG
**App "alexa" version: 1.0-RC1
C:/esp32/esp-idf/make/project.mk:519: *** multiple target patterns.  Stop.**

I was did what you said, But Still i am having the same problem.

Yuvaraj4041, 2019-05-28:
Hi @avsheth Thanks.

It is working now.

I exported path like This.

export IDF_PATH="/C/esp32/esp-idf"
export AUDIO_BOARD_PATH="/C/esp32/esp-va-sdk/board_support_pkgs/lyratd_msc/audio_board/audio_board_lyratd_msc"

closed 2019-05-28 by Yuvaraj4041