espressif/esp-va-sdk #50: any dependency between alexa_init(va_cfg) & bluetooth initialization ?

kvbulusu, 2019-06-03:
Hi,
What is the right way to initialize BT and scan for BLE beacons with the current alexa code? If we initialize before or after this line   alexa_init(va_cfg) , it crashes.  If I just comment the following lines and add BT initialization/enable code, it works.  Bit confused as to what is happening inside alexa_init and why its effected with Bluetooth init/deinit, bluedroid enable/disable etc....

conn_mgr_prov_mem_release
alexa_init(va_cfg)
va_dsp_init()

BLE beacons along with amazon alexa are key for the product I'm working on and so if you have any suggestions, please let me know.

avsheth, 2019-06-03:
As explained in issue #49 , conn_mgr_prov_mem_release() releases BLE memory, effectively disabling BLE permanently. If you need to continue using BLE, you can remove this call from app_main.c.

kvbulusu, 2019-06-03:
I tried removing the call, but alexa_init crashes...

kvbulusu, 2019-06-03:
Ok...here is the confusion. Probably I'm not doing something right or my understanding of the BT is wrong. If I comment the  alexa init/va_dsp_init (the first if/endif), BT works fine. If I enable it , second if/endif doesnt work. I simply get BT controller not initialized.  So I was under the impression alexa_init() and BT has some dependency ... Please let me know if you have any suggestions.

 Per earlier comment, I already commented the  conn_mgr_prov_mem_release() from the code.

 #if 0
 ret = alexa_init(va_cfg);
  if (ret != ESP_OK) {
        while(1) vTaskDelay(2);
    }
    /* This is a blocking call */
    va_dsp_init();
    /* This is only supported with minimum flash size of 8MB. */
    alexa_tone_enable_larger_tones();
    #endif

     #if 1
    if(esp_bt_controller_get_status() == ESP_BT_CONTROLLER_STATUS_ENABLED){
        ESP_LOGW(TAG,"\nBT Controller is enabled\n");
    } else {
        ESP_LOGE(TAG,"\nBT Controller is NOT enabled.Attempting to initialize ...\n");
        esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
       ret  =   esp_bt_controller_init(&bt_cfg);
        if  (ret)   {
            ESP_LOGE(TAG,   "%s initialize  controller  failed\n",  __func__);
            return;
        }

        ret =   esp_bt_controller_enable(ESP_BT_MODE_BTDM);
        if  (ret)   {
            ESP_LOGE(TAG,   "%s enable  controller  failed\n",  __func__);
            return;
        }
    }

        ret =   esp_bluedroid_init();
        if  (ret)   {
            ESP_LOGE(TAG,   "%s init    bluetooth   failed\n",  __func__);
            return;
        }

        ret =   esp_bluedroid_enable();
        if  (ret)   {
            ESP_LOGE(TAG,   "%s enable  bluetooth   failed\n",  __func__);
            return;
        }

        //register the scan callback function to the gap module
        if ((ret = esp_ble_gap_register_callback(esp_gap_cb)) != ESP_OK) {
            ESP_LOGE(TAG, "gap register error: %s", esp_err_to_name(ret));
        }

        esp_ble_gap_set_scan_params(&ble_scan_params);
    #endif

Dolumus, 2019-06-24:
@kvbulusu 
Recently I had a little go at implementing Alexa and a2dp on the lyratd_msc board, which is currently on hold, and I was met with the same problem as you. I did a little digging around and what seems to be the problem is that there is not enough RAM in dram0_0_seg for both projects, ideally you'd want to allocate the ram differently to make full use of the second segment, but here's the quick and dirty way I found that "works".

I'm not familiar with the beacon requirements, but I'd suggest if it needs only BLE that you enable the controller in BLE mode since that'll save a bit of ram. As for BTDM:

In make menuconfig:
Component config -> Wi Fi ->
WiFi IRAM speed optimization disabled
Max number of WiFi static RX buffers (8)
Max number of WiFi dynamic RX buffers (32)
Max number of WiFi static TX buffers (8)

Component config -> Bluetooth -> Bluetooth Controller ->
Bluetooth Controller mode -> Bluetooth Dual mode

Component config -> Bluetooth -> BlueDroid Enable
Use dynamic memory allocation in BT/BLE stack
Classic Bluetooth enable (if needed)

Component config -> ESP32-specific -> SPI RAM config ->
Allow .bss segment place in external memory

Also in conn_mgr_prov.c I've commented out the release functions just in case and my code is called just above ret = alexa_init(va_cfg);.
Do note that if there is still not enough ram, functionality will be lost when the WiFi allocates memory where the xTaskCreate tasks are made.
This is just a quick fix I did to check some functionality and is by no means ideal, I'll be looking into allocating the ram correctly if I get the chance. 
If you haven't resolved it on your own by now hope this helps.