espressif/esp-va-sdk #51: dependency between alexa_init() & bluetooth initialization ? (IDFGH-1279)

kvbulusu, 2019-06-03:
Environment
Development Kit: ESP32-LyraT
Module or chip used: ESP32-WROOM-32
IDF version (run git describe --tags to find it): v4.0-dev-621-g81ca1c013
Build System: Make
Compiler version (run xtensa-esp32-elf-gcc --version to find it): 5.20
Operating System: Linux
Power Supply: USB

Problem Description
What is the right way to initialize BT and scan for BLE beacons with the current alexa code ( https://github.com/espressif/esp-va-sdk/tree/master/examples/amazon_alexa) ? If we initialize before or after this line alexa_init(va_cfg) , it crashes. If I just comment on the following lines and add BT initialization/enable code, it works. Bit confused as to what is happening inside alexa_init and why its effected with Bluetooth init/deinit, bluedroid enable/disable etc....

conn_mgr_prov_mem_release()
alexa_init(va_cfg)
va_dsp_init()

[app_main]: Connected with IP Address: 192.168.1.100

Debug Logs
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40097245  PS      : 0x00060e30  A0      : 0x80124e25  A1      : 0x3ffbbed0  
0x40097245: xQueueGenericSend at /home/Firmware/19May09/esp-idf/components/freertos/queue.c:2002

A2      : 0x00005450  A3      : 0x3ffbbf10  A4      : 0xffffffff  A5      : 0x00000000  
A6      : 0x03000000  A7      : 0x00000023  A8      : 0x80097238  A9      : 0x3ffbbeb0  
A10     : 0x00000002  A11     : 0x00000000  A12     : 0x000002cf  A13     : 0x3f40ae4b  
A14     : 0x3ffbbb80  A15     : 0x00000002  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00005450  LBEG    : 0x4009483c  LEND    : 0x4009486a  LCOUNT  : 0xffffffff  
0x4009483c: memcpy at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memcpy.S:168

0x4009486a: memcpy at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memcpy.S:201

ELF file SHA256: 27ba34a2fa9d5b952dfd42adcb6ce313f78d106506ac8052735eb90c92b9055d

Backtrace: 0x40097245:0x3ffbbed0 0x40124e22:0x3ffbbf10 0x401248b1:0x3ffbbf40 0x401089ed:0x3ffbbf70 0x400d0c53:0x3ffbc090
0x40097245: xQueueGenericSend at /home/Firmware/19May09/esp-idf/components/freertos/queue.c:2002

0x40124e22: btc_task_post at /home/Firmware/19May09/esp-idf/components/bt/bluedroid/btc/core/btc_task.c:129
 (inlined by) btc_transfer_context at /home/Firmware/19May09/esp-idf/components/bt/bluedroid/btc/core/btc_task.c:162

0x401248b1: esp_bluedroid_disable at /home/Firmware/19May09/esp-idf/components/bt/bluedroid/api/esp_bt_main.c:101

0x401089ed: app_main at /home/Firmware/19May09/esp-va-sdk/examples/amazon_alexa/main/app_main.c:366

0x400d0c53: main_task at /home/Firmware/19May09/esp-idf/components/esp32/cpu_start.c:529

mahavirj, 2019-06-03:
@kvbulusu I have transferred this issue from `esp-idf` to `esp-va-sdk`, I hope that you wanted to file issue against this repository itself.

kvbulusu, 2019-06-03:
@mahavirj Thanks. I was not sure whether it belongs to this repo or esp-idf .

avsheth, 2019-06-03:
Closing this since this is duplicate of issue #50

closed 2019-06-03 by avsheth