espressif/esp-va-sdk #53: AMAZON_ALEXA   Error

lin-zi-jun, 2019-06-25:
Hi

I was trying to compile "AMAZON_ALEXA" example under "esp-va-sdk".

I am following the below steps.
Step 1 : Cloning ESP-IDF
Step 2 : Cloning ESP-VA-SDK by using "$ git clone https://github.com/espressif/esp-va-sdk.git"
Step 3 : export IDF_PATH=~/esp/esp-idf
Step 4 : export AUDIO_BOARD_PATH=~/esp/esp-va-sdk/board_support_pkgs/lyratd_msc/audio_board/audio_board_lyratd_msc

After this i was trying to compile using "make -j 8 flash monitor or make flash" command...

But it is not compiling and it is giving error for "make -j 8 flash monitor",
CC build/bt/bluedroid/bta/dm/bta_dm_act.o
/home/lin/esp/esp-idf/components/bt/bluedroid/bta/dm/bta_dm_act.c: In function 'bta_dm_config_eir':
/home/lin/esp/esp-idf/components/bt/bluedroid/bta/dm/bta_dm_act.c:687:13: error: argument 1 null where non-null expected [-Werror=nonnull]
             memcpy(p_bta_dm_eir_cfg->bta_dm_eir_url, config_eir->eir_url, config_eir->eir_url_len);
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/lin/esp/esp-idf/components/bt/bluedroid/bta/dm/bta_dm_act.c:40:
/home/lin/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/string.h:31:9: note: in a call to function 'memcpy' declared here
 void *  memcpy (void *__restrict, const void *__restrict, size_t);
         ^~~~~~
cc1: some warnings being treated as errors
/home/lin/esp/esp-idf/make/component_wrapper.mk:289: recipe for target 'bluedroid/bta/dm/bta_dm_act.o' failed
make[1]: *** [bluedroid/bta/dm/bta_dm_act.o] Error 1
/home/lin/esp/esp-idf/make/project.mk:582: recipe for target 'component-bt-build' failed
make: *** [component-bt-bui

Please anyone help me to solve this problem.

kewalmshah, 2019-06-26:
What version of IDF are you using ?

lin-zi-jun, 2019-06-26:
The original version was V3.3

Now I'm checkout to V3.2 and  git submodule update --init 
use 5.2.0 tool chain To compile but It reported the following error:

Toolchain path: /home/lin/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-61-gab8375a
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Compiler version: 5.2.0
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-61-gab8375a
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Python requirements from /home/lin/esp/esp-idf/requirements.txt are satisfied.
CC build/app_trace/app_trace_util.o
xtensa-esp32-elf-gcc: error: unrecognized command line option '-mfix-esp32-psram-cache-issue'
/home/lin/esp/esp-idf/make/component_wrapper.mk:285: recipe for target 'app_trace_util.o' failed
make[1]: *** [app_trace_util.o] Error 1
/home/lin/esp/esp-idf/make/project.mk:530: recipe for target 'component-app_trace-build' failed
make: *** [component-app_trace-build] Error 2

lin-zi-jun, 2019-06-27:
I found the 5.2.0 tool chain, and now it works.
https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0.tar.gz

closed 2019-06-27 by lin-zi-jun