espressif/esp-va-sdk #55: Changed POP ..seeing security1: Key mismatch. Close connection

kvbulusu, 2019-07-14:
We changed the proof of possession (pop) key both in the android app and in app_main/app_auth files.  During provisioning, I see this issue below.  Is that ok or do we need to change anything else? Not sure why its saying key mismatch...Any suggestions?

E (15516) security1: Key mismatch. Close connection
E (15516) security1: Session setup error -1
D (15526) protocomm: SEC_EP Req handler returned -1
E (15526) protocomm_ble: Invalid content received, killing connection
D (15536) simple_ble: ESP_GATTS_DISCONNECT_EVT, reason = 22

avsheth, 2022-12-21:
Hi @Nickobellic15X 
Are you seeing the same issue? Can you share device logs?
Have you changed PoP in the firmware ? 
For now, can you please try restarting the entire process?

ReactRaylogic, 2023-07-25:
I am facing the same issue when i try to send customData to the custom-data endpoint.
i am able to do the wifi provisining just not able to send custom data via bluetooth before wifi connection.