espressif/esp-va-sdk #56: Cannot find CMakeLists.txt

tatsutaigu, 2019-07-18:
Hi,

I'm using below latest toolchain **esp-idf-tools-setup-2.0.exe**.
When type and execute `idf.py build`, toolchain displayed error.
How do I resolve it?

[https://docs.espressif.com/projects/esp-idf/en/latest/get-started/windows-setup.html](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/windows-setup.html)

> C:\ESP-IDF_toolchain\esp-va-sdk\examples\amazon_alexa>idf.py build
> Note: You are using Python 3.7.4. Python 3 support is new, please report any pro
> blems you encounter. Search for 'Setting the Python Interpreter' in the ESP-IDF
> docs if you want to use Python 2.7.
> Checking Python dependencies...
> Python requirements from c:\ESP-IDF_toolchain\esp-vas\requirements.txt are satis
> fied.
> CMakeLists.txt not found in project directory C:\ESP-IDF_toolchain\esp-va-sdk\ex
> amples\amazon_alexa
> 
> C:\ESP-IDF_toolchain\esp-va-sdk\examples\amazon_alexa>

My command sequence is here.
```
git clone --recursive https://github.com/espressif/esp-idf.git esp-vas
cd esp-vas
git checkout release/v3.2
cd ..

python -m pip install -r c:\ESP-IDF_toolchain\esp-vas\requirements.txt

git clone https://github.com/espressif/esp-va-sdk.git
cd C:\ESP-IDF_toolchain\esp-va-sdk\examples\amazon_alexa

set IDF_PATH=c:\ESP-IDF_toolchain\esp-vas
set ESPPORT=COM20
set AUDIO_BOARD_PATH=c:\ESP-IDF_toolchain\esp-va-sdk\board_support_pkgs\lyrat\audio_board\audio_board_lyrat

```

tatsutaigu, 2019-07-18:
I made **CMakeLists.txt** as below and save it to root directory on amazon_alexa sample project.

> # The following lines of boilerplate have to be in your project's
> # CMakeLists in this exact order for cmake to work correctly
> cmake_minimum_required(VERSION 3.5)
> 
> include($ENV{IDF_PATH}/tools/cmake/project.cmake)
> project(alexa)

However, I got below error.

> [1067/1068] Linking CXX executable alexa.elf
> FAILED: alexa.elf
> cmd.exe /C "cd . && C:\ESP-IDF_toolchain\.espressif\tools\xtensa-esp32-elf\esp32
> -2019r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-fr
> ame-address  -nostdlib @CMakeFiles\alexa.elf.rsp  -o alexa.elf  && cd ."
> c:/esp-idf_toolchain/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa
> -esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bi
> n/ld.exe: esp-idf/esp32/libesp32.a(cpu_start.c.obj):(.literal.main_task+0x18): u
> ndefined reference to `app_main'
> c:/esp-idf_toolchain/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa
> -esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bi
> n/ld.exe: esp-idf/esp32/libesp32.a(cpu_start.c.obj): in function `main_task':
> C:/ESP-IDF_toolchain/esp-idf/components/esp32/cpu_start.c:535: undefined referen
> ce to `app_main'
> collect2.exe: error: ld returned 1 exit status
> ninja: build stopped: subcommand failed.
> ninja failed with exit code 1
> 
> C:\ESP-IDF_toolchain\esp-va-sdk\examples\amazon_alexa>
>

kedars, 2019-08-13:
Hi @tatsutaigu CMake is not yet supported in Alexa. You can use the build instructions as documented in this https://github.com/espressif/esp-va-sdk/blob/master/README-Getting-Started.md file. 

We will update this issue once CMake support is added to this repository.

tcpipchip, 2020-05-31:
Hi, have you CMake ready ?
I would like to use with ECLIPSE!
Thanks!

PIYUSHGIRI2021, 2021-04-27:
hii,  i got a same issue,i downloaded this code for PPPOS from git https://github.com/loboris/ESP32-PPPOS-EXAMPLE.git
but first i get error cannot find CMakeList.txt...after that i add 
cmake file but after that also i get error ..i am using esp32...anyone know hot to resolve this issue ...than please let me know..
Thanks.