espressif/esp-va-sdk #60: Alexa Amazon

tcpipchip, 2019-08-13:
Thanks for help Kedar

I will compile the Amazon Alexa to ESP32. 

About my book, i have done the integration with WEB/APP + ALEXA + LAMBDA + AWS IOT + ESP32.

I can send today send **phrase** like, TURN ON THE LIGHT, TURN OFF THE LIGHT

Today i need to use the WEB and APP to talk with ALEXA, now i want to replace them by LyraT.

With LyraT SDK is possible to config it to send those **phrases** too ?

tcpipchip, 2019-08-13:
wich crosstool-ng to use ?
git checkout esp-2019r1-rc2 ?

tcpipchip, 2019-08-13:
Kedar, good morning!
Could you do a video or something else to splain step to step how to compile the Amazon example ?
I am getting these errors!
I already tried all the versions!
[image]

[image]

tcpipchip, 2019-08-13:
Looks that is not a problem with Toolchain
But asio lib problem
[image]

tcpipchip, 2019-08-13:
Only can save us in this first step :)

ChiragAtal, 2019-08-14:
Hi @tcpipchip , try doing `git submodule update --init --recursive` in esp-idf after you have changed your branch.

tcpipchip, 2019-08-14:
I did that but
[image]
[image]

tcpipchip, 2019-08-14:
[image]
This is a error that gave me! If i remove that line compiles 100%
Are you having that error too ?

tcpipchip, 2019-08-14:
is that line important ? I guess that the ALEXA dont use BLE services!

dhrishi, 2019-08-14:
Please follow https://github.com/espressif/esp-va-sdk/blob/master/README-Getting-Started.md to get started.

From the first screenshot you have posted: https://github.com/espressif/esp-va-sdk/issues/60#issuecomment-520942891

It says compiler version 8.2.0 is not supported, expected version is 5.2.0.

We support version 8.2.0 from version ESP-IDF 4.0 onwards (and the current master branch).  As `esp-va-sdk` uses IDF 3.2, you need to make sure that the correct version of toolchain is used. These compilation errors should be gone if you choose the right version. You may follow "Setup Toolchain" here: https://docs.espressif.com/projects/esp-idf/en/v3.2/

tcpipchip, 2019-08-14:
Yes, 
I was following those steps
But how i create a toolchain version 8.2.0 ? or if you give a link to download should be good ! :)

dhrishi, 2019-08-14:
You ned to use version 5.2.0 and not 8.2.0

https://docs.espressif.com/projects/esp-idf/en/v3.2/get-started/linux-setup.html?highlight=setup%20toolchain#toolchain-setup

Also, once that is done, make sure that the PATH variable is updated with the correct path of the toolchain. As I mentioned, please follow this: https://docs.espressif.com/projects/esp-idf/en/v3.2/

tcpipchip, 2019-08-14:
Nice, compiled 100% @dhrishi 
Thank you so much!
Next step is to install the app and config the AMAZON example, right ? Or do you have some other suggestion ? I want to connect to my amazon/SKILL account use the AVS services!
Give suggestions / tips for the next step after burn the software!
Thank you so much!!!!!
[image]

kedars, 2019-08-15:
Yes, please use the 'ESP Alexa' app from the PlayStore to configure your LyraT to your home Wi-Fi network and login with your Amazon account.

tcpipchip, 2019-08-15:
Thank you!!!!!!!!!!!!

I had problems with VM to recognize the USB SERIAL to burn the bin on Lyrat

I wll burn on Windows

Can you confirm me the address of 3 bin files on flashtools ?
bootloader, partitions and alexa 

Thanks again team!

tcpipchip, 2019-08-15:
[image]

tcpipchip, 2019-08-15:
is this ?
[image]

avsheth, 2019-08-15:
Yep. You can now try and provision the device using the app.

tcpipchip, 2019-08-15:
Thanks!!!! Can you post in your alexa bin files in some repository ? If mine not works and would like with yours compiled and compare too the bin size!

tcpipchip, 2019-08-16:
All is working friends :)

kedars, 2019-08-18:
Great

closed 2019-08-19 by kedars

tcpipchip, 2019-10-08:
hi 
Can you refresh my memory ?

> 
> osboxes@osboxes:~/esp/esp-va-sdk/examples/amazon_alexa$ ls
> additional_components  partitions_4mb_flash.csv  README-Alexa.md
> main                   partitions.csv            sdkconfig.bt.defaults
> Makefile               partitions_mkspiffs.csv   sdkconfig.defaults
> osboxes@osboxes:~/esp/esp-va-sdk/examples/amazon_alexa$ make
> Makefile:10: *** Please specify audio board path via AUDIO_BOARD_PATH.  Stop.
> osboxes@osboxes:~/esp/esp-va-sdk/examples/amazon_alexa$ 

thanks!

tcpipchip, 2019-10-08:
Sir, in the make menuconfig (ALEXA EXAMPLE) there is a ENDPOINT to a AWS IOT. Can you explain how works ? Is that the MQTT IOT of AMAZON (the THING), if yes, where i define the TOPIC ?

tcpipchip, 2020-06-21:
Hi, do you know how to use the alexa example with AI-THINKER ESP32-A1S ?