espressif/esp-va-sdk #61: Amazon Music Support

renovatio-systems, 2019-09-02:
Hi, We are considering using esp32-va-sdk in our commercial projects. But we would require Amazon Music (Prime) support. Could you please let us know the steps involved in achieving this? Regards Kumar

avsheth, 2019-09-03:
Hi @renovatio-systems In order to enable Amazon prime music and other media streaming services (such as Saavn for India) you need to get your product ID whitelisted from Amazon.
You can raise a request to whitelist your product ID from Amazon's developer website.

And you can then set this product_id in app_main.c as
va_cfg->product_id = "<your_product_id>"

Hope it helps.

Thanks,
Amit

renovatio-systems, 2019-09-03:
Hello @avsheth I just created an Amazon product under developer.amazon.com.

And then I set a parameter under app_main.c 

va_cfg->product_id = "<your_product_id>"

Recompiled, erased the flash completly, reprogrammed and reprovisioned with the app. It is still coming up with the same problem.

Is there any other setup that is needed? 

Regards
Kumar

renovatio-systems, 2019-09-03:
Also, there is another parameter device_serial_num? Is this needed at all?

avsheth, 2019-09-03:
Hi @renovatio-systems 
Just creating a new ID won't suffice. A request to an Amazon PoC needs to be sent explicitly to whitelist the ID. If you go to https://developer.amazon.com/alexa/console/avs/products -> select your product -> go to Entertainment, you'll see a list of MSPs and their respective statuses i.e. whether access has been requested/granted or not. In order to get the access you have to get in touch with Amazon's point of contact and get your product ID whitelisted.

For serial_num, it is mainly required at manufacturing time to identify individual devices. It is not required for Amazon music access. Access is specific to the product_id and all devices with the same product_id (even if they have same serial_num) will be able to access streaming services your ID has access to.

dhruvchikhaliya12, 2020-01-03:
But will they whitelist product id for a single unit.
Thanks : )

renovatio-systems, 2020-01-28:
Hi @avsheth,

We got out device white listed by Amazon. But when we try and provision the app with the new product ID and the Espressif Android App. It used to work for few weeks and suddenly stopped working and we cant seem to provision any more boards. We get the following error (not very helpful at all) [Error](https://ibb.co/30hKTTg)

avsheth, 2020-01-28:
Hi @renovatio-systems 
Basically the API key generated for your product ID should be embedded in the Android app. And the app needs to be rebuilt with this new product ID.
So product ID on the device should match with corresponding API key in the android app.
In order to generate/retrieve API key for your product, you can follow the guide [here](https://developer.amazon.com/en-US/docs/alexa/login-with-amazon/register-android.html#add-android-settings)

There are couple of things you need to change in the Android app source, you can refer to it's readme [here](https://github.com/espressif/esp-idf-provisioning-android/tree/versions/avs-ble).

Hope this helps.

renovatio-systems, 2020-02-12:
Thanks @avsheth 

Unfortunately, that link for the guide no longer exists. Could you please provide a newer link or an offline copy of the document.

Warm Regards
Renovatio Systems

omarfakhr, 2023-09-19:
Where exactly in app_main.c can i set:
va_cfg->product_id = "<your_product_id>"

And can I invent my own name? Or do i have to take the name already available on the board?