espressif/esp-va-sdk #62: CPU0 crash

DuHeLong, 2019-09-10:
Hello, when I was developing ALEXA products, I encountered a problem. On the basis of esp-avs-sdk, I added a UART1, which would cause CPU0 crash. Please see the attachment. I spent a week without finding the cause of the problem. Can you tell me where the cause is?
   **{}}}
[dialog]: Listen end
[0;31mE (1017817) esp-tls: read error :-76:[0m
[http_playback_stream]: [http_response_recv]: returning EAGAIN
[http_stream]: [sid: 341] Response code: 200
[dialog]: Response 200
[dialog]: Entering VA_THINKING
[directive_proc]: Json data: {"directive":{"header":{"namespace":"SpeechSynthesizer","name":"Speak","messageId":"6a5e07dd-12d1-4f20-bfdf-ee5fab14ebcd","dialogRequestId":"7b802a8d-d3eb-43ac-afaa-8fd9e1f5b5bb"},"payload":{"url":"cid:DeviceTTSRendererV4_520402b4-1816-4e58-ad9c-44d81935a45e_943857296","format":"AUDIO_MPEG","token":"amzn1.as-ct.v1.Domain:Global:Fallback#ACRI#DeviceTTSRendererV4_520402b4-1816-4e58-ad9c-44d81935a45e"}}}
[attachment]: New with content_id: DeviceTTSRendererV4_520402b4-1816-4e58-ad9c-44d81935a45e_943857296
[directive_proc]: Executing blocking directive: Speak
[dialog]: Starting speech
[focus_manager]: Acquire channel: Dialog
[dialog]: Entering VA_SPEAKING
[dialog]: Stream finished
[speech_recognizer]: On Focus: Background
[dialog]: Event STREAM_CLOSED unsupported in SPEAKING state
[speech_synthesizer]: On Focus: Foreground
[1018 seconds]: [http_stream]: [stream_delete: 341] Internal: 14440, External: 1180880
[attachment]: Get: DeviceTTSRendererV4_520402b4-1816-4e58-ad9c-44d81935a45e_943857296
[basic_player]: Play attachment or local
[audio_codec]: Starting audio codec mp3 codec
[focus_manager]: Now active channel: Dialog
[basic_player]: Codec event: Codec Started
[basic_player]: Codec event: Codec Set Frequency
[http_stream]: [stream_new]: Internal: 13460, External: 1179060
[http_transport]: New stream event: {"event": { "header": { "namespace": "SpeechSynthesizer", "name": "SpeechStarted", "messageId": "51f5dc34-7182-414c-87a7-1c2a487a5750" },"payload": {"token": "amzn1.as-ct.v1.Domain:Global:Fallback#ACRI#DeviceTTSRendererV4_520402b4-1816-4e58-ad9c-44d81935a45e"}}}
[basic_player]: Codec event: Codec Stopped
[http_stream]: [stream_post: 343]: /v20160207/events
[http_stream]: [sid: 343] Response code: 204
[1019 seconds]: [http_stream]: [stream_delete: 343] Internal: 14440, External: 1180880
[basic_player]: i2s writer stream event Stopped
[attachment]: Free DeviceTTSRendererV4_520402b4-1816-4e58-ad9c-44d81935a45e_943857296
[dialog]: Speech done
[directive_proc]: Directive completed
[http_stream]: [stream_new]: Internal: 13276, External: 1179128
[http_transport]: New stream event: {"event": { "header": { "namespace": "SpeechSynthesizer", "name": "SpeechFinished", "messageId": "e52a5891-4156-4b51-bce1-f8b0476b161f" },"payload": {"token": "amzn1.as-ct.v1.Domain:Global:Fallback#ACRI#DeviceTTSRendererV4_520402b4-1816-4e58-ad9c-44d81935a45e"}}}
[http_stream]: [stream_post: 345]: /v20160207/events
[dialog]: Speaking state timeout. Releasing focus
[focus_manager]: Release channel: Dialog
[focus_manager]: Found backgrounded focus contender: Content
[dialog]: Entering VA_IDLE
[audio_player]: On Focus: Foreground
[audio_player]: Processing request ===> Foreground
[alexa_player]: Notify event type: 5
[http_stream]: [sid: 345] Response code: 204
[1020 seconds]: [http_stream]: [stream_delete: 345] Internal: 14508, External: 1178912
[http_stream]: [stream_new]: Internal: 9524, External: 1179124
[http_transport]: New stream event: {"event": { "header": { "namespace": "AudioPlayer", "name": "PlaybackResumed", "messageId": "0d43252c-e1a4-415d-badb-5d62fb13368d" },"payload": {"token": "amzn1.as-ct.v1.Domain:Phatic:AlexaOriginals#ACRI#url#ACRI#AlexaOriginals|fb2b5755-3c06-42c5-ad08-68210b29afe2|https://dvi8md6049w6b.cloudfront.net/nina_v1_8794a2fd24b84c9fccc9__Stories_en_us_nina_the_gift.mp3?Expires=1568150168&Signature=bIDfJTdEZWqqIJINIErTsT~92PKJDMFuLjsCKzYjLXTeZ1fwmU0CfR2CEwqcSToIU4Qse~zyqOeL20-GJoslLBFuPHI-u8iXT2WrbdGGkao7u5XuFP6kezrVsWyJz1Xps82RdLm8u9c-2N8sC5I5f8iO14l1BZYa0lqjQoDlfAoQgpnL1~X15x3Rj48rR8djlYYdYKcjFNXBgcJFs~YrKPkiVHySrSvlkFNCvOTYhZH4g1yjUrWqL0M-EecvlYCUKsrx~qXsb3NYFN4PpXcqRKdAr4m8TGg7lbalrgDB2AdN2L~GbRdwsBjp91KK95nKQ6ZlT-GY3hKUshrkrOOkzQ__&Key-Pair-Id=APKAJMAIYIGWWT37OIHA|The Gift|1568121368068|ATVPDKIKX0DER|A11P2XGOGFOIQS|en_US|Alexa3P:1.0/2019/09/10/13/52faaa8f22444fa28890b633c3f24c6a/16:05::TNIH_2V.37025dcb-6702-48c4-9c8e-ba2c6688c610ZXV/0|1b8b9d9e-ba65-4723-8543-de79eec967b2|PhaticIntent", "offsetInMilliseconds":51152}}}
[http_stream]: [stream_post: 347]: /v20160207/events
[0;31mE (1021217) esp-tls: read error :-76:[0m
[http_playback_stream]: [http_response_recv]: returning EAGAIN
[http_stream]: [sid: 347] Response code: 204
[1021 seconds]: [http_stream]: [stream_delete: 347] Internal: 13836, External: 1161956
[0;31mE (1021947) esp-tls: read error :-76:[0m
[http_playback_stream]: [http_response_recv]: returning EAGAIN
[0;31mE (1022447) esp-tls: read error :-76:[0m
[http_playback_stream]: [http_response_recv]: returning EAGAIN
[0;31mE (1023317) esp-tls: read error :-76:[0m
[http_playback_stream]: [http_response_recv]: returning EAGAIN
[0;31mE (1023997) esp-tls: read error :-76:[0m
[http_playback_stream]: [http_response_recv]: returning EAGAIN
[0;31mE (1025167) esp-tls: read error :-76:[0m
[http_playback_stream]: [http_response_recv]: returning EAGAIN
[0;31mE (1025667) esp-tls: read error :-76:[0m
[http_playback_stream]: [http_response_recv]: returning EAGAIN
[0;31mE (1027217) esp-tls: read error :-76:[0m
[http_playback_stream]: [http_response_recv]: returning EAGAIN
[0;31mE (1028147) esp-tls: read error :-76:[0m
[http_playback_stream]: [http_response_recv]: returning EAGAIN
[... 58 lines cut ...]
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x4008885b  PS      : 0x00050f34  A0      : 0x40086c2c  A1      : 0x3ffbe2b0  
A2      : 0x00000001  A3      : 0x3ffd0360  A4      : 0x80127778  A5      : 0x40092274  
A6      : 0x3983126f  A7      : 0x00bb8000  A8      : 0x3ff5f000  A9      : 0x3ffbe280  
A10     : 0x00000009  A11     : 0x00060f21  A12     : 0x00000000  A13     : 0x00000008  
A14     : 0x3ffba3e8  A15     : 0x3ffba43c  SAR     : 0x00000012  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x4009bf34  LEND    : 0x4009bf62  LCOUNT  : 0x00000000  
Core 0 was running in ISR context:
EPC1    : 0x40088912  EPC2    : 0x4000bff0  EPC3    : 0x00000000  EPC4    : 0x4008885b

Backtrace: 0x4008885b:0x3ffbe2b0 0x40086c29:0x3ffbe2f0 0x4008890f:0x3ffc6890 0x401255d9:0x3ffc68d0 0x4011d741:0x3ffc6920 0x4018feca:0x3ffc6940

Core 1 register dump:
PC      : 0x4008fd60  PS      : 0x00060e34  A0      : 0x80091f2a  A1      : 0x3ffdf0a0  
A2      : 0x00000001  A3      : 0x00060e20  A4      : 0x00000001  A5      : 0x00000001  
A6      : 0x000000fe  A7      : 0x3ffde9c0  A8      : 0x00000000  A9      : 0x00000001  
A10     : 0x0000abab  A11     : 0x00000000  A12     : 0x3f40b3c4  A13     : 0x3f40b2f8  
A14     : 0x0003817a  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x4009bf34  LEND    : 0x4009bf62  LCOUNT  : 0x00000000  

Backtrace: 0x4008fd60:0x3ffdf0a0 0x40091f27:0x3ffdf0d0 0x40125154:0x3ffdf110

Entering gdb stub now.
$T06#ba**

avsheth, 2019-09-13:
Hi @DuHeLong can you please tell which release are you using?
- Going by the logs, I suspect that remaining internal memory is lower than on the vanilla SDK.
Do you have any threads with stacks in internal memory? If so, can you try move them to external memory?
- If it's easily reproducible, can you please post steps to reproduce them and get console logs via `make monitor` command? Since that would print name of the functions on stack which led to the crash.

DuHeLong, 2019-09-29:
Hello,
Thank you very much for your last email statement. I've found the reason for the problem: UART's priority setting is too high.
   But we have a new problem. If I keep talking to ALEXA and ask it to keep telling stories and playing songs, it will cause a crash. Our file versions: ESP-IDF v3.2-dirty 2nd stage bootloader, ALEXA: V1.0-RC1-4-g1a3661f. Print information is as follows ：
  [http_stream]: [stream_post: 125]: /v20160207/events
[http_stream]: [sid: 125] Response code: 204
[9522 seconds]: [http_stream]: [stream_delete: 125] Internal: 11916, External: 1160336, min ever internal: 1888, largest free block: 3316
***ERROR*** A stack overflow in task mp3 codec has been detected.
abort() was called at PC 0x4009740c on core 1

Backtrace: 0x400970ef:0x3f8c6750

Entering gdb stub now.
$T0b#e6

[Uploading MP3 overflow.txt…]()

closed 2019-09-29 by DuHeLong

avsheth, 2019-09-30:
Hi @DuHeLong 
Thanks for bringing this to the notice. We will take a look and push the fix as soon as we can.