espressif/esp-va-sdk #63: ssid reset

jhpark555, 2019-09-27:
Once I download the example, it looks like I can't change ssid name.  How can I reset the ssid and password?
Thanks.

avsheth, 2019-09-28:
1. In order to reset the WiFi credentials, you need to keep Action/Tap-to-talk button pressed for more than 5 seconds. Device will restart, and will stay in the unconfigured mode for 3 minutes, after which device will auto-reboot and connect to the last known network.

2. How are you deregistering from the Amazon account? Using the app? Could you please send the complete log starting from signing-out to signing back in?

jhpark555, 2019-09-30:
After I followed  the instruction #1 as you mentioned, all problem cleared.    Regrading the #2 , I deregistered at Alexa (https://alexa.amazon.com)->settings-devices->about.

avsheth, 2019-10-01:
Hi @jhpark555 
I am guessing you want to sign out of the existing Amazon account on the device and sign in with the new one. In which case, you have to use ESP-Alexa app. You need to go to Manage devices->Select your device->Select sign-out on top right corner. 
This will sign the device out of the current account and you can sign-in with a new account from ESP-Alexa app again. (go to Manage devices->Select your device->Sign-in with Amazon)

De-registering your device from Alexa app would deregister the device from the Alexa cloud and hence you are not able to communicate using that device. You can re-register by signing out and signing in again as explained above.

Hope this helps.