espressif/esp-va-sdk #65: Alexa Portuguese

tcpipchip, 2019-10-07:
Hi, 

Last week, Alexa was ported to speak in Portuguese!

do you know how can i enable this SDK to talk in Portuguese ?
I already implemented on AVS the portugueses intends and sentences!

Thanks!!!!

avsheth, 2019-10-18:
Hi @tcpipchip 
Sorry for the delayed response.
We are in the process of releasing this feature where you can set the Alexa language from the companion app.
We will update the repo by the end of next week.

Thanks

tcpipchip, 2019-10-18:
Do you mean the mobile APP, right ?
Must i modify or recompile too the SDK ?
this is the best news of the week! thank you so much!!!!!!!!!!!!!!!!!

avsheth, 2019-10-18:
1. Yes. You would be able to change it from Espressif's companion app. (which will be released too along with the new SDK).
2. Unfortunately you will need to recompile the SDK to bring in this new feature.

You can still try this feature in the existing SDK via CLI. On the device console you can enter command `locale pt-BR` to converse in portuguese. Only caveat being the setting will be resetted when device is rebooted.

tcpipchip, 2019-10-18:
Do you mean if i connect lyrat on uart i can type something on rxd and receive a answer ?

avsheth, 2019-10-18:
Yes.
You can either run `make monitor` to get the console or you can use any serial terminal emulator (teraterm, hyperterminal on windows or minicom,picocom on Linux/mac) to access the device console.

tcpipchip, 2019-10-18:
Wonderfull! When have it i will finish the book!!!!

tcpipchip, 2019-11-01:
Hi! Any update to Portuguese option in  SDK ?
I can see that the Mobile APP has update!

avsheth, 2019-11-02:
Hi @tcpipchip ,
I am afraid it might get delayed by a week more. In the meantime, did you get to try with the CLI option?

tcpipchip, 2019-11-07:
@avsheth 
Thanks! All going well now! CLI locale pt-BR works fine!!!!!
If someone in Brazil is interested, i did the Alexa BOOK using the Lyrat to tests :)

tcpipchip, 2019-11-14:
https://www.youtube.com/watch?v=rBbXBzWCDIg

avsheth, 2019-12-09:
Hi @tcpipchip 
Latest release includes support for A2DP sink and changing Alexa's language via the iOS and Android companion app.

Thanks

closed 2019-12-09 by avsheth

tcpipchip, 2019-12-12:
hi @avsheth 
Thank you so much!!!!
Book going well :)

tcpipchip, 2020-05-15:
Hi,
I am back with the esp-va-sdk

tcpipchip, 2020-05-15:
https://www.youtube.com/watch?v=nyVm7-5g9CQ