espressif/esp-va-sdk #67: E (23486) event_send: invalid event base with diaglow after entering wifi-set

kvbulusu, 2019-10-26:
I see the following error after entering the wifi-set <ssid> <pwd> and then the following message keeps printing ....Any suggestions as to what to look for...

E (23486) event_send: invalid event base 

Not sure if it helps, but the event_id is 26 
==========
I (1436) [diag_cli]: Registering command: mem-dump
I (1446) [diag_cli]: Registering command: task-dump
I (1446) [diag_cli]: Registering command: cpu-dump
I (1456) [diag_cli]: Registering command: heap-trace
I (1466) [va_diag_cli]: Registering command: nvs-get
I (1466) [va_diag_cli]: Registering command: nvs-set
I (1476) [va_diag_cli]: Registering command: nvs-erase
I (1476) [va_diag_cli]: Registering command: reboot
I (1486) [va_diag_cli]: Registering command: crash
I (1486) [wifi_cli]: Registering command: wifi-set
I (1516) wifi: wifi driver task: 3ffd1348, prio:23, stack:3584, core=0
I (1516) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1516) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1696) wifi: wifi firmware version: 27f2f06
I (1696) wifi: config NVS flash: enabled
I (1696) wifi: config nano formating: disabled
I (1696) wifi: Init dynamic tx buffer num: 32
I (1706) wifi: Init data frame dynamic rx buffer num: 32
I (1706) wifi: Init management frame dynamic rx buffer num: 32
I (1716) wifi: Init management short buffer num: 32
I (1716) wifi: Init static tx buffer num: 32
I (1726) wifi: Init static rx buffer size: 1600
I (1726) wifi: Init static rx buffer num: 10
I (1736) wifi: Init dynamic rx buffer num: 32
I (1736) [app_main]: ***************************
I (1746) [app_main]: ** Starting provisioning **
I (1746) [app_main]: ***************************
I (1756) [app_main]: Refer the README-Dialogflow.md and enter the CLI commands. Make sure to enter the nvs-set commands first and then the wifi-set command.
>> wifi-set Europe Europe
W (22586) event: handler already registered, overwriting
W (22586) event: handler already registered, overwriting
W (22596) event: handler already registered, overwriting
W (22596) event: handler already registered, overwriting
W (22606) event: handler already registered, overwriting
W (22616) event: handler already registered, overwriting
W (22616) event: handler already registered, overwriting
W (22696) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
[va_button]: button pressed: 36
I (22726) [va_button]: tap to talk pressed
I (22836) phy: phy_version: 4102, 2fa7a43, Jul 15 2019, 13:06:06, 0, 2
I (22896) wifi: mode : sta (24:0a:c4:9d:07:94)
>> I (23016) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (23436) wifi: state: init -> auth (b0)
I (23436) wifi: state: auth -> assoc (0)
I (23446) wifi: state: assoc -> run (10)
I (23456) wifi: connected with Europe, channel 1, BW20, bssid = 10:da:43:c5:83:c1
I (23486) wifi: pm start, type: 1

E (23486) event_send: invalid event base 
[va_button]: button pressed: 36
I (23656) [va_button]: tap to talk pressed
[va_button]: button pressed: 36
I (23986) [va_button]: tap to talk pressed
[va_button]: button pressed: 36
I (24316) [va_button]: tap to talk pressed
[va_button]: button pressed: 36
I (24646) [va_button]: tap to talk pressed
[va_button]: button pressed: 36
I (24976) [va_button]: tap to talk pressed
[va_button]: button pressed: 36

==================================

avsheth, 2019-10-31:
Are you, by any chance, re-registering esp_event_loop_init() ?

kvbulusu, 2019-11-04:
> Are you, by any chance, re-registering esp_event_loop_init() ?

I'm using the example (google_dialogflow) .Didn't customize it...I'm checking if its because esp-idf version....Do you see if we use esp-idf 4.0 ?

avsheth, 2019-11-11:
Hi @kvbulusu 
Alexa SDK is based on esp-idf v3.2 right now. Can you please try switching IDF branch to release/v3.2?

avsheth, 2019-12-09:
Hi @kvbulusu 
Do you see this issue with release/v3.2 based IDF as well?

kvbulusu, 2019-12-12:
The error is different with v3.2 .... This is default example from google_dialogflow

esp-va-sdk:
%git describe  --tags
1.0-RC1-4-g1a3661f

esp-idf:
%git describe  --tags
v3.2.3-105-g65db2bf87

V (877) esp_adapter: thread sem get: sem=0x3ffcfa84
V (877) esp_adapter: thread sem get: sem=0x3ffcfa84
D (887) RTC_MODULE: Wi-Fi takes adc2 lock.
D (887) phy_init: loading PHY init data from application binary
D (897) nvs: nvs_open_from_partition phy 0
D (897) nvs: nvs_get cal_version 4
V (907) phy_init: phy_get_rf_cal_version: 4009

D (907) nvs: nvs_get_str_or_blob cal_mac
D (917) nvs: nvs_get_str_or_blob cal_data
D (917) nvs: nvs_close 3
> I (967) phy: phy_version: 4009, e7844c0, Jul 15 2019, 13:20:06, 0, 0
I (977) wifi: mode : sta (24:0a:c4:9d:07:94)
E (977) event: Event loop not initialized via esp_event_loop_init, but esp_event_send called
V (977) esp_adapter: thread sem get: sem=0x3ffcfa84
I (987) wifi: Set ps type: 0

[va_button]: button pressed: 36
I (1047) [va_button]: tap to talk pressed

kvbulusu, 2019-12-12:
I just tried latest release with ver 3.2 and it seems working...not sure where the issue came from last release...will close.

closed 2019-12-12 by kvbulusu