espressif/esp-va-sdk #68: a2dp sink and alexa

jhpark555, 2019-10-29:
Can we use bluetooth a2dp and alexa/google service at the same time?
I tried to port a2dp sink demo code on alexa example, but there was an overflow error.

jhpark555, 2019-12-02:
Is A2DP with Alexa still ongoing?
Thanks.

avsheth, 2019-12-09:
Hi @jhpark555 
Alexa with A2DP sink support has been pushed in the latest release. You can try that out. May I know which evaluation board are you using? Since it would require the flash of size > 4MB.

Thanks

jhpark555, 2019-12-09:
Hello I'm using esp32_lyrat board.  By the way, I have a link error like below.
Would you please let me know what I am missing ?

LD build/alexa.elf
/home/ppark/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o):(.literal.bluetooth_internal_start+0x50): undefined reference to `esp_bt_gap_register_callback'
/home/ppark/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o):(.literal.bluetooth_internal_start+0x54): undefined reference to `esp_avrc_ct_init'
/home/ppark/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o):(.literal.bluetooth_internal_start+0x58): undefined reference to `esp_avrc_ct_register_callback'
/home/ppark/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o):(.literal.bluetooth_internal_start+0x5c): undefined reference to `esp_avrc_tg_init'
/home/ppark/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o):(.literal.bluetooth_internal_start+0x60): undefined reference to `esp_avrc_tg_register_callback'
/home/ppark/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o):(.literal.bluetooth_internal_start+0x64): undefined reference to `esp_avrc_psth_bit_mask_operation'
/home/ppark/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o):(.literal.bluetooth_internal_start+0x68): undefined reference to `esp_avrc_tg_set_psth_cmd_filter'
/home/ppark/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o):(.literal.bluetooth_internal_start+0x6c): undefined reference to `esp_avrc_rn_evt_bit_mask_operation'
/home/ppark/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o):(.literal.bluetooth_internal_start+0x70): undefined reference to `esp_avrc_tg_set_rn_evt_cap'
/home/ppark/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o):(.literal.bluetooth_internal_start+0x74): undefined reference to `esp_a2d_register_callback'
/home/ppark/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o):(.literal.bluetooth_internal_start+0x78): undefined reference to `esp_a2d_sink_register_data_callback'
/home/ppark/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o):(.literal.bluetooth_internal_start+0x7c): undefined reference to `esp_a2d_sink_init'
/home/ppark/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o):(.literal.bluetooth_internal_store_pairing$constprop$2+0x14): undefined reference to `esp_bt_gap_remove_bond_device'
/home/ppark/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o):(.literal.bluetooth_internal_enable_host_ct+0xc): undefined reference to `esp_bt_gap_set_security_param'
/home/ppark/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o):(.literal.bluetooth_internal_enable_discoverable+0x0): undefined reference to `esp_bt_gap_set_scan_mode'
/home/ppark/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o):(.literal.bluetooth_internal_mediacontrol_pause+0x8): undefined reference to `esp_avrc_ct_send_passthrough_cmd'
/home/ppark/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o):(.literal.bluetooth_internal_connect_device_by_id+0x8): undefined reference to `esp_a2d_sink_connect'
/home/ppark/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o):(.literal.bluetooth_internal_on_connect_success+0x14): undefined reference to `esp_bt_gap_get_remote_services'
/home/ppark/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o):(.literal.bluetooth_internal_disconnect_streaming_bt_sink+0x4): undefined reference to `esp_a2d_sink_disconnect'
/home/ppark/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o):(.literal.bluetooth_internal_scan_start+0x4): undefined reference to `esp_bt_gap_start_discovery'
/home/ppark/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o):(.literal.bluetooth_internal_scan_stop+0x4): undefined reference to `esp_bt_gap_cancel_discovery'
/home/ppark/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o):(.literal.bt_app_gap_cb+0x10): undefined reference to `esp_bt_gap_ssp_confirm_reply'
/home/ppark/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o):(.literal.bt_app_gap_cb+0x14): undefined reference to `esp_bt_gap_resolve_eir_data'
/home/ppark/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o): In function `bluetooth_internal_start':
(.text.bluetooth_internal_start+0x110): undefined reference to `esp_bt_gap_register_callback'
/home/ppark/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o): In function `bluetooth_internal_start':
(.text.bluetooth_internal_start+0x116): undefined reference to `esp_avrc_ct_init'
/home/ppark/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o): In function `bluetooth_internal_start':
(.text.bluetooth_internal_start+0x149): undefined reference to `esp_avrc_ct_register_callback'
/home/ppark/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bluetooth_internal.o): In function `bluetooth_internal_start':
(.text.bluetooth_internal_start+0x14f): undefined reference to `esp_avrc_tg_init'
[... cut ...]
/home/ppark/esp/esp-va-sdk/components/voice_assistant/lib/libalexa.a(bt_app_av.o): In function `bt_app_rc_tg_cb':
(.text.bt_app_rc_tg_cb+0x10a): undefined reference to `esp_a2d_media_ctrl'
collect2: error: ld returned 1 exit status
/home/ppark/esp/esp-idf/make/project.mk:460: recipe for target '/home/ppark/esp/esp-va-sdk/examples/amazon_alexa/build/alexa.elf' failed
make: *** [/home/ppark/esp/esp-va-sdk/examples/amazon_alexa/build/alexa.elf] Error 1
ppark@ubuntu:~/esp/esp-va-sdk/examples/amazon_alexa$

jhpark555, 2019-12-09:
Hello.  I'm using ESP32_LyraT board.  By the way I'm having a reboot problem.
I enabled BT classic and a2dp sink on the menuconfig and make ALEXA_BT=1, the
below reboot is happening. What am I missing?

I (62) boot: Chip Revision: 1
I (65) boot_comm: mismatch chip revision, expect 1, found 0
I (39) boot: ESP-IDF v3.2.3-35-g3bb4b4db4-dirty 2nd stage bootloader
I (39) boot: compile time 14:38:57
I (43) boot: Enabling RNG early entropy source...
I (45) qio_mode: Enabling QIO for flash chip GD
I (50) boot: SPI Speed      : 80MHz
I (54) boot: SPI Mode       : QIO
I (58) boot: SPI Flash Size : 4MB
I (62) boot: Partition Table:
I (66) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (81) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (88) boot:  2 factory          factory app      00 00 00010000 00390000
I (96) boot: End of partition table
I (100) boot_comm: mismatch chip revision, expect 1, found 0
I (106) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x22804c (2261068) map
I (684) esp_image: segment 1: paddr=0x00238074 vaddr=0x3ffbdb60 size=0x051ec ( 20972) load
I (691) esp_image: segment 2: paddr=0x0023d268 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/ppark/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (692) esp_image: segment 3: paddr=0x0023d670 vaddr=0x40080400 size=0x029a0 ( 10656) load
I (703) esp_image: segment 4: paddr=0x00240018 vaddr=0x400d0018 size=0x16a010 (1482768) map
0x400d0018: _stext at ??:?

I (1082) esp_image: segment 5: paddr=0x003aa030 vaddr=0x40082da0 size=0x1b068 (110696) load
0x40082da0: psram_cache_init at /home/ppark/esp/esp-idf/components/esp32/spiram_psram.c:850
 (inlined by) psram_enable at /home/ppark/esp/esp-idf/components/esp32/spiram_psram.c:765

E (1116) esp_image: Image length 3887296 doesn't fit in partition length 3735552
E (1116) boot: Factory app partition is not bootable
E (1119) boot: No bootable app partitions in the partition table
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6740
load:0x40078000,len:10016
load:0x40080400,len:7072
entry 0x40080750

avsheth, 2019-12-10:
Hi @jhpark555 
As mentioned in the changelog [here](https://github.com/espressif/esp-va-sdk/blob/master/CHANGELOG.md#changelog), A2DP sink requires flash size > 4MB. Since size of the Alexa binary increases substantially.

jhpark555, 2019-12-10:
Thanks. I understood.   Then, I need to buy esp32-lyratd-syna with 8MB flash board.

jhpark555, 2019-12-16:
Even if I replaced the flash to 8mb. Still it is crashing.  Do I need to change partition to support 8MB?
Please let me know where I have to change.

avsheth, 2019-12-17:
does the device crash with default partitions.csv as well?

jhpark555, 2019-12-17:
Yes, I think so.

avsheth, 2019-12-17:
can you post a complete failure log with that? Since from the above logs, I am not really able to identify the actual error. Booting up seems to be happening normally.

jhpark555, 2019-12-17:
Okay. I'm at home now. I'll post full log message on tomorrow morning.   Thanks.

jhpark555, 2019-12-17:
I changed to use default partition table by blocking partitions_4mb_flash_csv in audio_board.mk.  Below is the full log data.     Thanks.

MONITOR
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6740
load:0x40078000,len:10016
load:0x40080400,len:7072
entry 0x40080750
I (61) boot: Chip Revision: 1
I (64) boot_comm: mismatch chip revision, expect 1, found 0
I (38) boot: ESP-IDF v3.2.3-35-g3bb4b4db4-dirty 2nd stage bootloader
I (39) boot: compile time 10:18:03
I (52) boot: Enabling RNG early entropy source...
I (52) qio_mode: Enabling default flash chip QIO
I (52) boot: SPI Speed      : 80MHz
I (56) boot: SPI Mode       : QIO
I (60) boot: SPI Flash Size : 8MB
I (65) boot: Partition Table:
I (68) boot: ## Label            Usage          Type ST Offset   Length
I (75) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (83) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (90) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (98) boot:  3 factory          factory app      00 00 00010000 00100000
I (105) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (113) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (120) boot: End of partition table
I (125) boot: Defaulting to factory image
I (129) boot_comm: mismatch chip revision, expect 1, found 0
I (136) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x22804c (2261068) map
I (714) esp_image: segment 1: paddr=0x00238074 vaddr=0x3ffbdb60 size=0x051ec ( 20972) load
I (720) esp_image: segment 2: paddr=0x0023d268 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/ppark/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (721) esp_image: segment 3: paddr=0x0023d670 vaddr=0x40080400 size=0x029a0 ( 10656) load
I (733) esp_image: segment 4: paddr=0x00240018 vaddr=0x400d0018 size=0x16a010 (1482768) map
0x400d0018: _stext at ??:?

I (1112) esp_image: segment 5: paddr=0x003aa030 vaddr=0x40082da0 size=0x1b068 (110696) load
0x40082da0: psram_cache_init at /home/ppark/esp/esp-idf/components/esp32/spiram_psram.c:850
 (inlined by) psram_enable at /home/ppark/esp/esp-idf/components/esp32/spiram_psram.c:765

E (1146) esp_image: Image length 3887296 doesn't fit in partition length 1048576
E (1146) boot: Factory app partition is not bootable
E (1148) esp_image: image at 0x110000 has invalid magic byte
E (1155) boot_comm: mismatch chip ID, expect 0, found 65297
E (1161) boot_comm: can't run on lower chip revision, expect 1, found 46
W (1168) esp_image: image at 0x110000 has invalid SPI mode 47
E (1175) boot: OTA app partition slot 0 is not bootable
E (1180) esp_image: image at 0x210000 has invalid magic byte
E (1187) boot_comm: mismatch chip ID, expect 0, found 22049
E (1193) boot_comm: can't run on lower chip revision, expect 1, found 47
W (1200) esp_image: image at 0x210000 has invalid SPI mode 75
W (1207) esp_image: image at 0x210000 has invalid SPI size 11
E (1213) boot: OTA app partition slot 1 is not bootable
E (1219) boot: No bootable app partitions in the partition table
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
[... 123 lines cut ...]
I (77) boot: compile time 10:18:03
I (90) boot: Enabling RNG early entropy source...
I (90) qio_mode: Enabling default flash chip QIO
I (90) boot: SPI Speed      : 80MHz
I (95) boot: SPI Mode       : QIO
I (99) boot: SPI Flash Size : 8MB
I (103) boot: Partition Table:
I (107) boot: ## Label            Usage          Type ST Offset   Length
I (114) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (122) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (129) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (137) boot:  3 factory          factory app      00 00 00010000 00100000
I (145) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (152) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (160) boot: End of partition table
I (164) boot: Defaulting to factory image
I (169) boot_comm: mismatch chip revision, expect 1, found 0
I (175) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x22804c (2261068) map
I (753) esp_image: segment 1: paddr=0x00238074 vaddr=0x3ffbdb60 size=0x051ec ( 20972) load
I (759) esp_image: segment 2: paddr=0x0023d268 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/ppark/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (760) esp_image: segment 3: paddr=0x0023d670 vaddr=0x40080400 size=0x029a0 ( 10656) load
I (772) esp_image: segment 4: paddr=0x00240018 vaddr=0x400d0018 size=0x16a010 (1482768) map
0x400d0018: _stext at ??:?

avsheth, 2019-12-19:
Hi @jhpark555 
Partition size of ota0 and ota1 that you've set (0x100000) is insufficient for the image (as indicated by the error message "E (1146) esp_image: Image length 3887296 doesn't fit in partition length 1048576"). The default partition table, [partitions.csv](https://github.com/espressif/esp-va-sdk/blob/master/examples/amazon_alexa/partitions.csv), sets this to 0x420000, since binary size is much larger.

If it isn't fitting in 8MB flash, you may remove ota_1 partition for now (this would only affect OTA functionality).

jhpark555, 2019-12-20:
Hello,
When I used defalut partitions.csv and removed ota_1, the alexa worked fine but the bluetooth didn't work.     Can you tell me which board you used ? I need to prepare for demo for our customer in a month.   Thanks.

avsheth, 2019-12-23:
It shouldn't be board specific.
Did you compile with ALEXA_BT=1 ? It is mentioned [here](https://github.com/espressif/esp-va-sdk/blob/master/README-Getting-Started.md#build-and-flash-the-project)

jhpark555, 2020-01-07:
Even if I enabled ALEXA_BT=1 while compiling, I can't find bluetooth supported device from the PC or phone.  Only Amazon can works.

rst:0x1 (POWERON_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6740
load:0x40078000,len:10016
load:0x40080400,len:7072
entry 0x40080750
I (61) boot: Chip Revision: 1
I (64) boot_comm: mismatch chip revision, expect 1, found 0
I (38) boot: ESP-IDF v3.2.3-35-g3bb4b4db4-dirty 2nd stage bootloader
I (38) boot: compile time 10:46:40
I (52) boot: Enabling RNG early entropy source...
I (52) qio_mode: Enabling default flash chip QIO
I (52) boot: SPI Speed      : 80MHz
I (56) boot: SPI Mode       : QIO
I (60) boot: SPI Flash Size : 8MB
I (64) boot: Partition Table:
I (68) boot: ## Label            Usage          Type ST Offset   Length
I (75) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (83) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (90) boot:  2 fctry            WiFi data        01 02 00010000 00006000
I (98) boot:  3 otadata          OTA data         01 00 00016000 00002000
I (105) boot:  4 ota_0            OTA app          00 10 00020000 00420000
I (113) boot: End of partition table
I (117) boot: No factory image, trying OTA 0
I (122) boot_comm: mismatch chip revision, expect 1, found 0
I (128) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x1fc714 (2082580) map
I (662) esp_image: segment 1: paddr=0x0021c73c vaddr=0x3ffbdb60 size=0x038d4 ( 14548) load
I (666) esp_image: segment 2: paddr=0x00220018 vaddr=0x400d0018 size=0x166bb0 (1469360) map
0x400d0018: _stext at ??:?

I (1038) esp_image: segment 3: paddr=0x00386bd0 vaddr=0x3ffc1434 size=0x017c8 (  6088) load
I (1040) esp_image: segment 4: paddr=0x003883a0 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/ppark/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (1046) esp_image: segment 5: paddr=0x003887a8 vaddr=0x40080400 size=0x1da08 (121352) load
I (1110) boot: Loaded app from partition at offset 0x20000
I (1110) boot: Disabling RNG early entropy source...
[conn_mgr_prov]: Found ssid: ESS-Sound-2.4G
[conn_mgr_prov]: Found password: phoenixplus
[va_button]: button pressed: 36
[app_main]: Connected with IP Address: 192.168.101.169
[alexa]: Waiting for time to be updated
[alexa]: Done getting current time: 1578420829
[alexa]: Authentication done
[dialog]: Entering VA_IDLE
[speaker]: Volume changed to 40
[capabilities]: Capabilities unchanged
[endpoint_handler]: AVS endpoint: https://avs-alexa-na.amazon.com
W (1994) I2S: I2S driver already installed
W (2014) I2S: I2S driver already installed
[http_stream]: [stream_new]: Internal: 54100, External: 1278256
[auth-delegate]: Token will be refreshed after 3000 seconds.
[http_stream]: [stream_get: 1]: /v20160207/directives
[http_stream]: [sid: 1] Response code: 200
[http_transport]: AVS level connction has now been established: https://avs-alexa-na.amazon.com
[http_stream]: [stream_new]: Internal: 53052, External: 1270920
[http_transport]: New stream event: {"context":[{"header":{"namespace":"Speaker","name":"VolumeState"},"payload":{"volume":40,"muted":false}},{"header":{"namespace":"SpeechRecognizer","name":"RecognizerState"},"payload":{"wakeword":"ALEXA"}},{"header":{"namespace":"SpeechSynthesizer","name":"SpeechState"},"payload":{"token":"","offsetInMilliseconds":0,"playerActivity":"PLAYING"}},{"header":{"namespace":"AudioPlayer","name":"PlaybackState"},"payload":{"token":"","offsetInMilliseconds":0,"playerActivity":"IDLE"}},{"header":{"namespace":"AudioActivityTracker","name":"ActivityState"},"payload":{}},{"header":{"namespace":"Alerts","name":"AlertsState"},"payload":{"allAlerts":[],"activeAlerts":[]}},{"header":{"namespace":"Notifications","name":"IndicatorState"},"payload":{"isEnabled":false,"isVisualIndicatorPersisted":false}}],"event":{"header":{"namespace":"System","name":"SynchronizeState","messageId":"9bc366a8-091f-4ff9-8d36-60553287cb1d"},"payload":{}}}
[http_stream]: [stream_post: 3]: /v20160207/events
[http_stream]: [sid: 3] Response code: 204
############## Alexa is ready ##############
[3 seconds]: [http_stream]: [stream_delete: 3] Internal: 52564, External: 1278240, min ever internal: 45080, largest free block: 25012
[http_stream]: [stream_new]: Internal: 54480, External: 1278660
[http_stream]: [stream_get: 5]: /ping
[http_stream]: [sid: 5] Response code: 204
[184 seconds]: [http_stream]: [stream_delete: 5] Internal: 52444, External: 1278448, min ever internal: 45080, largest free block: 25012
[http_stream]: [stream_new]: Internal: 53220, External: 1278660
[http_stream]: [stream_get: 7]: /ping
[http_stream]: [sid: 7] Response code: 204
[364 seconds]: [http_stream]: [stream_delete: 7] Internal: 54076, External: 1278448, min ever internal: 45080, largest free block: 25012
[http_stream]: [stream_new]: Internal: 54244, External: 1278660
[http_stream]: [stream_get: 9]: /ping
[http_stream]: [sid: 9] Response code: 204
[544 seconds]: [http_stream]: [stream_delete: 9] Internal: 53960, External: 1278448, min ever internal: 45080, largest free block: 25012
[http_stream]: [stream_new]: Internal: 52372, External: 1278496
[http_stream]: [stream_get: 11]: /ping
[http_stream]: [sid: 11] Response code: 204
[... cut ...]
[http_stream]: [stream_new]: Internal: 54008, External: 1278496
[http_stream]: [stream_get: 13]: /ping
[http_stream]: [sid: 13] Response code: 204
[905 seconds]: [http_stream]: [stream_delete: 13] Internal: 53728, External: 1278284, min ever internal: 45080, largest free block: 25012
[http_stream]: [stream_get: 15]: /ping: 52140, External: 1278496
[http_stream]: [sid: 15] Response code: 204
[1085 seconds]: [http_stream]: [stream_delete: 15] Internal: 53612, External: 1278284, min ever internal: 45080, largest free block: 25012

jhpark555, 2020-01-07:
Hello

New issues happens to me. I erased flash in order to write again. At this time, I have a credential and token issues.  Please look at below log data.   Constantly repetitive message are appearing.

[conn_mgr_prov]: Scan results:
    S.N. SSID                             RSSI
    [ 0] ESS-Sound-2.4G                    -34
    [ 1] ChromecastAudio7999.a             -47
    [ 2] ESSNETGR88                        -64
    [ 3] SOUND ROOM                        -71
    [ 4] EERO                              -83
    [ 5] ESSNETGR87                        -86
    [ 6] USR5461                           -87
    [ 7] AP5                               -92
[avs_config]: APP Got: ANRjiVJkZIhlzUMekyvx, amzn1.application-oa2-client.e8c06545d9764a05b8b6d91eecfeb23a, amzn-com.espressif.provbleavs://?methodName=signin, abcd1234
[auth-delegate]: Auth delegate with comp app: ANRjiVJkZIhlzUMekyvx, amzn1.application-oa2-client.e8c06545d9764a05b8b6d91eecfeb23a, amzn-com.espressif.provbleavs://?methodName=signin, abcd1234E (38681) esp-tls: couldn't get hostname for :api.amazon.com:
E (38691) esp-tls: Failed to open new connection
E (38701) httpc: Failed to create a new TLS connection
E (38701) [auth-delegate]: Failed to connect to Auth URL "https://api.amazon.com/auth/O2/token"
E (38711) [auth-delegate]: Authentication attempt failed. Retrying. Please check connectivity and/or credentials. In case of authentication failure try resetting the device to factory mode and restart provisioning
[conn_mgr_prov_handler]: WiFi Credentials Received: 
	ssid: ESS-Sound-2.4G 
	password: phoenixplus
W (38921) wifi: alloc eb len=76 type=2 fail, heap:4143112

W (38921) wifi: m f probe req l=0

W (39041) wifi: alloc eb len=76 type=2 fail, heap:4143968

W (39041) wifi: m f probe req l=0

W (39161) wifi: alloc eb len=76 type=2 fail, heap:4143968

W (39161) wifi: m f probe req l=0

E (39231) esp-tls: couldn't get hostname for :api.amazon.com:
E (39231) esp-tls: Failed to open new connection
E (39231) httpc: Failed to create a new TLS connection
E (39231) [auth-delegate]: Failed to connect to Auth URL "https://api.amazon.com/auth/O2/token"
E (39241) [auth-delegate]: Authentication attempt failed. Retrying. Please check connectivity and/or credentials. In case of authentication failure try resetting the device to factory mode and restart provisioning
W (39281) wifi: alloc eb len=76 type=2 fail, heap:4143968

vikramdattu, 2020-01-07:
For BT not working issue...

If you had previously build without ALEXA_BT=1, you will need a clean build. i.e., you need to remove build/ sdkconfig and sdkconfig.old.
And then do a build with ALEXA_BT=1

jhpark555, 2020-01-07:
I followed your instruction but I have reboot error like below. And continuously it reboots.

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6388
load:0x40078000,len:9324
load:0x40080400,len:6456
entry 0x40080738
I (59) boot: Chip Revision: 1
I (64) boot_comm: mismatch chip revision, expect 1, found 0
I (39) boot: ESP-IDF v3.2.3-35-g3bb4b4db4-dirty 2nd stage bootloader
I (39) boot: compile time 15:24:06
I (52) boot: Enabling RNG early entropy source...
I (52) boot: SPI Speed      : 80MHz
I (52) boot: SPI Mode       : DIO
I (56) boot: SPI Flash Size : 8MB
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 fctry            WiFi data        01 02 00010000 00006000
I (93) boot:  3 otadata          OTA data         01 00 00016000 00002000
I (100) boot:  4 ota_0            OTA app          00 10 00020000 00420000
I (108) boot: End of partition table
I (112) boot: No factory image, trying OTA 0
I (117) boot_comm: mismatch chip revision, expect 1, found 0
I (123) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x214684 (2180740) map
I (739) esp_image: segment 1: paddr=0x002346ac vaddr=0x3ffbdb60 size=0x051ec ( 20972) load
I (746) esp_image: segment 2: paddr=0x002398a0 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/ppark/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (747) esp_image: segment 3: paddr=0x00239ca8 vaddr=0x40080400 size=0x06368 ( 25448) load
I (763) esp_image: segment 4: paddr=0x00240018 vaddr=0x400d0018 size=0x15ff34 (1441588) map
0x400d0018: _stext at ??:?

I (1165) esp_image: segment 5: paddr=0x0039ff54 vaddr=0x40086768 size=0x17700 ( 96000) load
0x40086768: ram_set_txcap_reg at /home/aiqin/git_tree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_cal.c:2458 (discriminator 1)

I (1216) boot: Loaded app from partition at offset 0x20000
I (1216) boot: Disabling RNG early entropy source...
[conn_mgr_prov]: Found ssid: ESS-Sound-2.4G
[conn_mgr_prov]: Found password: phoenixplus
[va_button]: button pressed: 36
[app_main]: Connected with IP Address: 192.168.101.169
[alexa]: Waiting for time to be updated
[alexa]: Done getting current time: 1578440038
[alexa]: Authentication done
[dialog]: Entering VA_IDLE
E (1952) [bluetooth-internal]: Error reading paired device list from NVS
[speaker]: Volume changed to 40
[capabilities]: Capabilities unchanged
[endpoint_handler]: Cannot find endpoint URL in NVS. Setting default: https://avs-alexa-eu.amazon.com
[endpoint_handler]: AVS endpoint: https://avs-alexa-eu.amazon.com
W (2832) I2S: I2S driver already installed
W (2862) I2S: I2S driver already installed
/home/ppark/esp/esp-idf/components/freertos/tasks.c:684 (xTaskCreateStaticPinnedToCore)- assert failed!
abort() was called at PC 0x40091733 on core 0
0x40091733: xTaskCreateStaticPinnedToCore at /home/ppark/esp/esp-idf/components/freertos/tasks.c:4691

Backtrace: 0x40093937:0x3ffbc750 0x40093c69:0x3ffbc770 0x40091733:0x3ffbc790 0x4011bafe:0x3ffbc7d0 0x4010396a:0x3ffbc800 0x400d0e43:0x3ffbc920
0x40093937: invoke_abort at /home/ppark/esp/esp-idf/components/esp32/panic.c:707

0x40093c69: abort at /home/ppark/esp/esp-idf/components/esp32/panic.c:707

0x40091733: xTaskCreateStaticPinnedToCore at /home/ppark/esp/esp-idf/components/freertos/tasks.c:4691

0x4011bafe: xTaskCreateStatic at /home/ppark/esp/esp-idf/components/freertos/include/freertos/task.h:608
 (inlined by) va_dsp_init at /home/ppark/esp/esp-va-sdk/board_support_pkgs/lyrat/dsp_driver/lyrat_driver/components/va_dsp/va_dsp.c:274

0x4010396a: app_main at /home/ppark/esp/esp-va-sdk/examples/amazon_alexa/main/app_main.c:212

0x400d0e43: main_task at /home/ppark/esp/esp-idf/components/esp32/cpu_start.c:506

avsheth, 2020-01-08:
Can you try pulling latest IDF v3.2 branch?
Just to confirm, you are trying with the default sdkconfig.bt.defaults right?

jhpark555, 2020-01-08:
I'm using sdkconfig. Below is this  configuration.  In the menuconfig, I selected bluetooth dual mode and enabled classic bluetooth. then I compiled with ALEXA_BT=1 option. Do I have wrong configuration?
Thanks.

#
# Automatically generated file; DO NOT EDIT.
# Espressif IoT Development Framework Configuration
#
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0000

#
# SDK tool configuration
#
CONFIG_TOOLPREFIX="xtensa-esp32-elf-"
CONFIG_PYTHON="python"
CONFIG_MAKE_WARN_UNDEFINED_VARIABLES=y

#
# Bootloader config
#
CONFIG_LOG_BOOTLOADER_LEVEL_NONE=
CONFIG_LOG_BOOTLOADER_LEVEL_ERROR=
CONFIG_LOG_BOOTLOADER_LEVEL_WARN=
CONFIG_LOG_BOOTLOADER_LEVEL_INFO=y
CONFIG_LOG_BOOTLOADER_LEVEL_DEBUG=
CONFIG_LOG_BOOTLOADER_LEVEL_VERBOSE=
CONFIG_LOG_BOOTLOADER_LEVEL=3
CONFIG_BOOTLOADER_SPI_WP_PIN=7
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
CONFIG_BOOTLOADER_FACTORY_RESET=
CONFIG_BOOTLOADER_APP_TEST=
CONFIG_BOOTLOADER_WDT_ENABLE=y
CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE=
CONFIG_BOOTLOADER_WDT_TIME_MS=9000

#
# Security features
#
CONFIG_SECURE_SIGNED_APPS_NO_SECURE_BOOT=
CONFIG_SECURE_BOOT_ENABLED=
CONFIG_FLASH_ENCRYPTION_ENABLED=

#
# Serial flasher config
#
CONFIG_ESPTOOLPY_PORT="/dev/ttyUSB0"
CONFIG_ESPTOOLPY_BAUD_115200B=
CONFIG_ESPTOOLPY_BAUD_230400B=
CONFIG_ESPTOOLPY_BAUD_921600B=y
CONFIG_ESPTOOLPY_BAUD_2MB=
CONFIG_ESPTOOLPY_BAUD_OTHER=
CONFIG_ESPTOOLPY_BAUD_OTHER_VAL=115200
CONFIG_ESPTOOLPY_BAUD=921600
CONFIG_ESPTOOLPY_COMPRESSED=y
CONFIG_FLASHMODE_QIO=y
CONFIG_FLASHMODE_QOUT=
CONFIG_FLASHMODE_DIO=
CONFIG_FLASHMODE_DOUT=
CONFIG_ESPTOOLPY_FLASHMODE="dio"
CONFIG_ESPTOOLPY_FLASHFREQ_80M=y
CONFIG_ESPTOOLPY_FLASHFREQ_40M=
CONFIG_ESPTOOLPY_FLASHFREQ_26M=
CONFIG_ESPTOOLPY_FLASHFREQ_20M=
CONFIG_ESPTOOLPY_FLASHFREQ="80m"
CONFIG_ESPTOOLPY_FLASHSIZE_1MB=
CONFIG_ESPTOOLPY_FLASHSIZE_2MB=
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=
CONFIG_ESPTOOLPY_FLASHSIZE_8MB=y
CONFIG_ESPTOOLPY_FLASHSIZE_16MB=
CONFIG_ESPTOOLPY_FLASHSIZE="8MB"
CONFIG_ESPTOOLPY_FLASHSIZE_DETECT=y
[... 955 lines cut ...]
CONFIG_SPIFFS_DBG=
CONFIG_SPIFFS_API_DBG=
CONFIG_SPIFFS_GC_DBG=
CONFIG_SPIFFS_CACHE_DBG=
CONFIG_SPIFFS_CHECK_DBG=
CONFIG_SPIFFS_TEST_VISUALISATION=

#
# TCP/IP Adapter
#
CONFIG_IP_LOST_TIMER_INTERVAL=120
CONFIG_TCPIP_LWIP=y

#
# Virtual file system
#
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y

#
# Wear Levelling
#
CONFIG_WL_SECTOR_SIZE_512=
CONFIG_WL_SECTOR_SIZE_4096=y
CONFIG_WL_SECTOR_SIZE=4096

avsheth, 2020-01-08:
I see quite a few differences when in terms of the configuration. (pinned core for BT task, Debug logs for BT enabled etc.). 
I suggest please try removing all older artifacts (build, sdkconfig, sdkconfig.old) and try once with `make ALEXA_BT=1 erase_flash flash`. See if it works and gradually work up with other changes.
Right now, it's really tight in terms of internal memory availability when BT and Alexa are ran simultaneously.

jhpark555, 2020-01-14:
I followed your recommendation and at this time I'm having a error and reboot again continuously.

E (2440) [bluetooth-internal]: Error reading paired device list from NVS

Is that something to do with flash size? I used 8mb.  Can you let me know what the error means?

vikramdattu, 2020-01-14:
The error just indicates that you don't have any paired devices yet.

Just go ahead and pair a device.

jhpark555, 2020-01-14:
Before I can try paring, the system already went crash. only one time , when the wifi has failure, the bluetooth seems to be started pairing.   Do I need 16mb memory instead of 8mb?

vikramdattu, 2020-01-14:
Hi @jhpark555 
1. Can you please share the log file to be analysed for crash case?
2. 8MB is sufficient enough.

jhpark555, 2020-01-15:
Yes.  here is.

--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6388
load:0x40078000,len:9324
load:0x40080400,len:6456
entry 0x40080738
I (59) boot: Chip Revision: 1
I (63) boot_comm: mismatch chip revision, expect 1, found 0
I (38) boot: ESP-IDF v3.2.3-35-g3bb4b4db4-dirty 2nd stage bootloader
I (38) boot: compile time 11:49:06
I (51) boot: Enabling RNG early entropy source...
I (51) boot: SPI Speed      : 80MHz
I (51) boot: SPI Mode       : DIO
I (55) boot: SPI Flash Size : 8MB
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 fctry            WiFi data        01 02 00010000 00006000
I (92) boot:  3 otadata          OTA data         01 00 00016000 00002000
I (99) boot:  4 ota_0            OTA app          00 10 00020000 00420000
I (107) boot: End of partition table
I (111) boot: No factory image, trying OTA 0
I (116) boot_comm: mismatch chip revision, expect 1, found 0
I (122) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x214684 (2180740) map
I (738) esp_image: segment 1: paddr=0x002346ac vaddr=0x3ffbdb60 size=0x051ec ( 20972) load
I (745) esp_image: segment 2: paddr=0x002398a0 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/ppark/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (746) esp_image: segment 3: paddr=0x00239ca8 vaddr=0x40080400 size=0x06368 ( 25448) load
I (762) esp_image: segment 4: paddr=0x00240018 vaddr=0x400d0018 size=0x15ff34 (1441588) map
0x400d0018: _stext at ??:?

I (1164) esp_image: segment 5: paddr=0x0039ff54 vaddr=0x40086768 size=0x17700 ( 96000) load
0x40086768: ram_set_txcap_reg at /home/aiqin/git_tree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_cal.c:2458 (discriminator 1)

I (1215) boot: Loaded app from partition at offset 0x20000
I (1215) boot: Disabling RNG early entropy source...
[conn_mgr_prov]: Found ssid: ESS-Sound-2.4G
[conn_mgr_prov]: Found password: phoenixplus
[app_main]: Connected with IP Address: 192.168.101.169
[alexa]: Waiting for time to be updated
[alexa]: Done getting current time: 1579044379
[alexa]: Authentication done
[dialog]: Entering VA_IDLE
E (2461) [bluetooth-internal]: Error reading paired device list from NVS
[speaker]: Volume changed to 40
[capabilities]: Capabilities unchanged
[endpoint_handler]: Cannot find endpoint URL in NVS. Setting default: https://avs-alexa-eu.amazon.com
[endpoint_handler]: AVS endpoint: https://avs-alexa-eu.amazon.com
W (3361) I2S: I2S driver already installed
W (3381) I2S: I2S driver already installed
/home/ppark/esp/esp-idf/components/freertos/tasks.c:684 (xTaskCreateStaticPinnedToCore)- assert failed!
abort() was called at PC 0x40091733 on core 0
0x40091733: xTaskCreateStaticPinnedToCore at /home/ppark/esp/esp-idf/components/freertos/tasks.c:4691

Backtrace: 0x40093937:0x3ffbc750 0x40093c69:0x3ffbc770 0x40091733:0x3ffbc790 0x4011bafe:0x3ffbc7d0 0x4010396a:0x3ffbc800 0x400d0e43:0x3ffbc920
0x40093937: invoke_abort at /home/ppark/esp/esp-idf/components/esp32/panic.c:707

0x40093c69: abort at /home/ppark/esp/esp-idf/components/esp32/panic.c:707

[... 60 lines cut ...]
[dialog]: Entering VA_IDLE
E (1932) [bluetooth-internal]: Error reading paired device list from NVS
[speaker]: Volume changed to 40
[capabilities]: Capabilities unchanged
[endpoint_handler]: Cannot find endpoint URL in NVS. Setting default: https://avs-alexa-eu.amazon.com
[endpoint_handler]: AVS endpoint: https://avs-alexa-eu.amazon.com
W (2802) I2S: I2S driver already installed
W (2822) I2S: I2S driver already installed
/home/ppark/esp/esp-idf/components/freertos/tasks.c:684 (xTaskCreateStaticPinnedToCore)- assert failed!
abort() was called at PC 0x40091733 on core 0
0x40091733: xTaskCreateStaticPinnedToCore at /home/ppark/esp/esp-idf/components/freertos/tasks.c:4691

Backtrace: 0x40093937:0x3ffbc750 0x40093c69:0x3ffbc770 0x40091733:0x3ffbc790 0x4011bafe:0x3ffbc7d0 0x4010396a:0x3ffbc800 0x400d0e43:0x3ffbc920
0x40093937: invoke_abort at /home/ppark/esp/esp-idf/components/esp32/panic.c:707

0x40093c69: abort at /home/ppark/esp/esp-idf/components/esp32/panic.c:707

0x40091733: xTaskCreateStaticPinnedToCore at /home/ppark/esp/esp-idf/components/freertos/tasks.c:4691

0x4011bafe: xTaskCreateStatic at /home/ppark/esp/esp-idf/components/freertos/include/freertos/task.h:608
 (inlined by) va_dsp_init at /home/ppark/esp/esp-va-sdk/board_support_pkgs/lyrat/dsp_driver/lyrat_driver/components/va_dsp/va_dsp.c:274

0x4010396a: app_main at /home/ppark/esp/esp-va-sdk/examples/amazon_alexa/main/app_main.c:212

0x400d0e43: main_task at /home/ppark/esp/esp-idf/components/esp32/cpu_start.c:506

avsheth, 2020-01-15:
Hi @jhpark555 
Can you try with below changes:
1. Replace MALLOC_CAP_INTERNAL with MALLOC_CAP_SPIRAM in board_support_pkgs/lyrat/dsp_driver/lyrat_driver/components/va_dsp/va_dsp.c (assuming this is your AUDIO_BOARD_PATH)
2. Reduce no. of WiFi Static Tx buffers from 32 to 26 in menuconfig->Component config->WiFi
3. Reduce no. of WiFi Static Rx buffers from 12 to 10 in menuconfig->Component config->WiFi

jhpark555, 2020-01-15:
Hi Amit and Vikram,

It worked as I followed your last instruction. Many Thanks.   I can demo now for my customer.
By the way I have one more question.
I want to use AAC decoder on Bluetooth a2dp sink. Does esp32 can support it?

vikramdattu, 2020-01-15:
Unfortunately, ESP32 bluetooth stack support only SBC for now.