espressif/esp-va-sdk #69: dialogflow - Failed to verify peer certificate!

Simo0082, 2019-11-07:
Hello,
I'm tryng to implement "as-is" example with DialogFlow. I successfully followed all steps reported in the readme.

Running the project I have this log:

E (722592) esp-tls: Failed to open new connection
E (722592) [sh2lib]: [sh2-connect] esp-tls connection failed
E (722602) [grpc_http_transport]: Failed to connect
I (724602) [grpc_http_transport]: Connecting to server: https://dialogflow.googleapis.com/
E (724822) esp-tls: mbedtls_ssl_handshake returned -0x2700
I (724822) esp-tls: Failed to verify peer certificate!
I (724822) esp-tls: verification info:   ! The certificate is not correctly signed by the trusted CA

NOTE:
I created all the authentication requirement as described in the readme and I configured the device running the nvs-set  commands properly configured.

Where is the problem?

Thanks

Regards

Simone

Simo0082, 2019-11-11:
Any suggestions?

ChiragAtal, 2019-11-11:
Hi @Simo0082 , the TLS certificates for Dialogflow seems to have changed and using the newer ones does not give this error. We will be pushing out a new release soon with the new certificates. Will update you once that is done.

Simo0082, 2019-11-11:
@ChiragAtal do you have any forecast about the release of the new version? I urgently need to deliver a prototype and I don't know how. Eventually, in the meantime, can I update the certificates on the current version? How can I do? Thanks a lot.

Simone

ChiragAtal, 2019-11-11:
@Simo0082 Can you apply this patch and try again.

[dialogflow_new_certificates.patch.zip](https://github.com/espressif/esp-va-sdk/files/3831299/dialogflow_new_certificates.patch.zip)

Simo0082, 2019-11-13:
@ChiragAtal  it works! Thank you very much for your effort!

closed 2019-11-13 by Simo0082

Joellneto, 2019-11-16:
Hi! I´m having the same problem with "google_voice_assistant" example from ESP-VA-SDK: successfully followed all steps reported in the readme, but I´m receiving the following messages, related to the certificate used:

E (8290) esp-tls: Failed to open new connection
E (8290) [sh2lib]: [sh2-connect] esp-tls connection failed
E (8300) [grpc_http_transport]: Failed to connect
I (10300) [grpc_http_transport]: Connecting to server: https://embeddedassistant.googleapis.com
E (10620) esp-tls: mbedtls_ssl_handshake returned -0x2700
I (10620) esp-tls: Failed to verify peer certificate!
I (10620) esp-tls: verification info:   ! The certificate is not correctly signed by the trusted CA

Is it possible to generate a patch to solve this?
Thanks in advance,
Joel