espressif/esp-va-sdk #71: E (57969) esp-tls: mbedtls_ssl_handshake returned -0x2700

sergireus, 2019-12-06:
Hello all, I'm triing to make the dialogflow example and once entered de wifi credentials i have this error, i'm triing some times with the same result.... please, anyone can help?¿?¿?...

Thanks

I (57469) audio_stream: Stream i2s_reader Event Started
I (57479) [lyrat_init]: Reader stream event 2
E (57969) esp-tls: mbedtls_ssl_handshake returned -0x2700
I (57969) esp-tls: Failed to verify peer certificate!
I (57969) esp-tls: verification info:   ! The certificate is not correctly signed by the trusted CA

E (57989) esp-tls: Failed to open new connection
E (57989) httpc: Failed to create a new TLS connection
E (57999) [auth-delegate]: Failed to connect to Auth URL "https://www.googleapis.com/oauth2/v3/token"
E (58009) [auth-delegate]: Error receiving refresh token response: -1
[auth-delegate]: Token will be refreshed after 60 seconds.

dhrishi, 2019-12-07:
See here: https://github.com/espressif/esp-va-sdk/issues/69 
You can test by applying the patch present here: https://github.com/espressif/esp-va-sdk/issues/69#issuecomment-552427865

sergireus, 2019-12-08:
Hi dhrishi...... thanks for the quick response .... 
I applied the patch and I have the same problem but the example is working now

Thanks very much

closed 2019-12-08 by sergireus