espressif/esp-va-sdk #74: Partition usage

PoojaG20, 2019-12-14:
Hi ,
I am new to it. I just wanted to know if on one device(e.g Lyrat 4.3) I have alexa sdk running can on same device i can have different program running. If yes then, can both work independently meaning when Alexa is invoked its program is used, otherwise different program works. How can we do it ?

avsheth, 2019-12-16:
Yes, in theory, you can have multiple things running Along with Alexa. But that is subjected to RAM availability, CPU availability etc.
Simplest implementation could be you can spawn a new thread from main, and run your application in that thread. This way both will run in parallel.

PoojaG20, 2019-12-16:
So the partition we see in this repository is not actually for fulfilling this ? Like in amazon-alexa example there is one partition.csv which has **nvs, phy_init, fctry, otadata, ota_0, ota_1**. Can you help on understanidn these, what are its purposes

kirkrudolph, 2019-12-16:
The partition.csv is simply dividing the storage into different parts. See https://docs.espressif.com/projects/esp-jumpstart/en/latest/firmwareupgrade.html for more information about the different types of storage and their purposes.

PoojaG20, 2019-12-17:
> Yes, in theory, you can have multiple things running Along with Alexa. But that is subjected to RAM availability, CPU availability etc.
> Simplest implementation could be you can spawn a new thread from main, and run your application in that thread. This way both will run in parallel.

If you have some example for the suggested implementation that would be very much helpful :)

avsheth, 2019-12-19:
Hi @PoojaG20 
You may refer to some of the IDF examples [here](https://github.com/espressif/esp-idf/tree/master/examples).
You can also checkout FreeRTOS APIs [here](https://www.freertos.org/FreeRTOS-quick-start-guide.html#page_top) to know about creating tasks and other OS synchronisation mechanisms.

Thanks