espressif/esp-va-sdk #75: Login with Amazon Error

kirkrudolph, 2019-12-17:
When I get to the 'esp_avs_open_security would like access to: Alexa Voice Services and Alexa Account Connection' screen and click 'Allow', the Login with Amazon page opens with 

"We're sorry! An error occurred when we tried to process your request. Rest assured, we're already working on the problem and expect to resolve it shortly."

Am I missing something? I am using the iOS app.

Thanks.

avsheth, 2019-12-17:
Hi @kirkrudolph 
Any chance you changed product ID in your main?
Can you try with the latest iOS app available on the app store?

kirkrudolph, 2019-12-18:
Hi @avsheth,
Yes, your are correct. I changed the product ID back and have gotten the demo working. Thanks for the help!

closed 2019-12-18 by kirkrudolph