espressif/esp-va-sdk #76: About amazon alexa.

dhruvchikhaliya12, 2019-12-18:
I had the same problem with esp32 alexa.
Plese help me with this issue.I think it only run on 8MB PSRAM but i see microwavemont video in which it works without external PSRAM.

ets Jun 8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6772
load:0x40078000,len:10016
load:0x40080400,len:7072
entry 0x40080750
I (61) boot: Chip Revision: 1
I (64) boot_comm: mismatch chip revision, expect 1, found 0
I (39) boot: ESP-IDF v3.2.3-33-g90af68bd3 2nd stage bootloader
I (39) boot: compile time 13:31:34
I (39) boot: Enabling RNG early entropy source...
I (45) qio_mode: Enabling default flash chip QIO
I (50) boot: SPI Speed : 80MHz
I (54) boot: SPI Mode : QIO
I (58) boot: SPI Flash Size : 4MB
I (62) boot: Partition Table:
I (66) boot: ## Label Usage Type ST Offset Length
I (73) boot: 0 nvs WiFi data 01 02 00009000 00006000
I (81) boot: 1 phy_init RF data 01 01 0000f000 00001000
I (88) boot: 2 factory factory app 00 00 00010000 00390000
I (96) boot: End of partition table
I (100) boot_comm: mismatch chip revision, expect 1, found 0
I (106) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1f2884 (2041988) map
I (629) esp_image: segment 1: paddr=0x002028ac vaddr=0x3ffbdb60 size=0x048e8 ( 18664) load
I (635) esp_image: segment 2: paddr=0x0020719c vaddr=0x40080000 size=0x00400 ( 1024) load
0x40080000: _WindowOverflow4 at C:/msys32/home/dhruv/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (636) esp_image: segment 3: paddr=0x002075a4 vaddr=0x40080400 size=0x08a6c ( 35436) load
I (656) esp_image: segment 4: paddr=0x00210018 vaddr=0x400d0018 size=0x1318f0 (1251568) map
0x400d0018: _stext at ??:?

I (971) esp_image: segment 5: paddr=0x00341910 vaddr=0x40088e6c size=0x0eca0 ( 60576) load
0x40088e6c: ld_inq_frm_isr_wrapper at ??:?

I (1004) boot: Loaded app from partition at offset 0x10000
I (1005) boot: Disabling RNG early entropy source...
E (123) [app_main]: Failed to alloc voice assistant config
abort() was called at PC 0x40102f43 on core 0
0x40102f43: app_main at C:/msys32/home/dhruv/esp/esp-va-sdk/examples/amazon_alexa/main/app_main.c:154 (discriminator 2)

Backtrace: 0x4009155b:0x3ffbc040 0x40091855:0x3ffbc060 0x40102f43:0x3ffbc080 0x400d0ccb:0x3ffbc1a0
0x4009155b: invoke_abort at C:/msys32/home/dhruv/esp/esp-idf/components/esp32/panic.c:707

0x40091855: abort at C:/msys32/home/dhruv/esp/esp-idf/components/esp32/panic.c:707

0x40102f43: app_main at C:/msys32/home/dhruv/esp/esp-va-sdk/examples/amazon_alexa/main/app_main.c:154 (discriminator 2)

0x400d0ccb: main_task at C:/msys32/home/dhruv/esp/esp-idf/components/esp32/cpu_start.c:506

avsheth, 2019-12-19:
Hi @dhruvchikhaliya12,
Can you please help me with a few details:
1. Are you running the default application or have you changed anything in the existing source? I see that it is failing to allocate the very first chunk of memory (which of just few bytes).
2. What is the IDF version you are compiling against?

dhruvchikhaliya12, 2019-12-19:
> Hi @dhruvchikhaliya12,
> Can you please help me with a few details:
>
> 1. Are you running the default application or have you changed anything in the existing source? I see that it is failing to [...]
1.No i doesn't change anything and i uses esp-wroom-32 module with es8388 audio codec ic.
2.Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0

dhruvchikhaliya12, 2019-12-19:
1. Are you running the default application or have you changed anything in the existing source? I see that it is failing to allocate the very first chunk of memory (which of just few bytes).
> 2. What is the IDF version you are compiling against?
1.No i doesn't change anything and i uses esp-wroom-32 module with es8388 audio codec ic.
2.Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0

dhruvchikhaliya12, 2019-12-19:
1.No i doesn't change anything and i uses esp-wroom-32 module with es8388 audio codec ic.
2.Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0

avsheth, 2019-12-19:
Hi @dhruvchikhaliya12 
ESP-WROOM-32 module doesn't have PSRAM. And the SDK requires PSRAM for supporting all features and functionalities of Alexa.
Since the app is allocating the structure on the PSRAM, it is failing.

You may want to get a board with ESP32-WROVER module.

Thanks

dhruvchikhaliya12, 2019-12-19:
There is anyway to disable some features of alexa or there is any way to support mr buddycasino code for es8388 audio codec.

avsheth, 2019-12-20:
Unfortunately, SPIRAM allocations are spread over the entire Alexa functionality (It is infact default allocation mechanism, to keep as much internal memory available for WiFi buffers, for better throughput).

As for adding support for es8388 audio codec, you can refer to code [here](https://github.com/espressif/esp-adf/tree/master/components/audio_hal/driver/es8388), and call its APIs in your code.

dhruvchikhaliya12, 2019-12-20:
Sorry for another question please will you tell me how to call api in code.
Thanks.

avsheth, 2019-12-23:
You need to call appropriate functions from es8388.h file in your code.
For example, if you want to initialize codec, write es8388_init() at the appropriate place in your code.

dhruvchikhaliya12, 2019-12-24:
ok Thanks avsheth i will try it.

dhruvchikhaliya12, 2019-12-27:
I soldered SRAM on my board but now it giving me this type of error
I (1032) boot: Loaded app from partition at offset 0x10000
I (1032) boot: Disabling RNG early entropy source...
[conn_mgr_prov]: Found ssid: Black Pearl
[conn_mgr_prov]: Found password: Electron
[va_button]: button pressed: 36
[app_main]: Disconnected. Event: 5. Connecting to the AP again
[app_main]: Disconnected. Event: 5. Connecting to the AP again
[app_main]: Disconnected. Event: 5. Connecting to the AP again
[app_main]: Disconnected. Event: 5. Connecting to the AP again
[app_main]: Disconnected. Event: 5. Connecting to the AP again
[app_main]: Disconnected. Event: 5. Connecting to the AP again
[app_main]: Disconnected. Event: 5. Connecting to the AP again
[app_main]: Disconnected. Event: 5. Connecting to the AP again
[app_main]: Disconnected. Event: 5. Connecting to the AP again

avsheth, 2019-12-27:
Is the configured AP up?
Looks like device is not able to connect to the AP/router.
If you want to configure a new network, you can do `make erase_flash flash`. And configure the device with Android/iOS app.

dhruvchikhaliya12, 2019-12-27:
Yes configured ap is up.
It stil not works.

dhruvchikhaliya12, 2019-12-27:
sorry now it works. thanks it works

dhruvchikhaliya12, 2019-12-27:
I complete all the setup process but now it not listen to me and show me this when i push the alexa button.
[http_stream]: [stream_post: 33]: /v20160207/events
[directive_proc]: Json data: {"directive":{"header":{"namespace":"SpeechRecognizer","name":"StopCapture","messageId":"e955086f-af0f-410d-bf03-35847069aedf","dialogRequestId":"132209a7-0673-40c4-b15d-11d51f37b26b"},"payload":{}}}
[dialog]: Listen end
[va_dsp]: Sending stop command
[va_dsp]: Event 3 unsupported in STOPPED state
[http_stream]: [sid: 33] Response code: 204
[dialog]: Response 204
[va_dsp]: Sending stop command
[va_dsp]: Event 5 unsupported in STOPPED state
[dialog]: Entering VA_IDLE
[dialog]: Stream finished
[dialog]: Event STREAM_CLOSED unsupported in IDLE state
[401 seconds]: [http_stream]: [stream_delete: 33] Internal: 48848, External: 1277628, min ever internal: 44348, largest free block: 45336

dhruvchikhaliya12, 2019-12-28:
Now it not playing song from any streaming services.

vikramdattu, 2019-12-29:
Could you please let us know which streaming service you have problem with?
You may try Saavn, Gaana, TuneIn etc, those should work.

For Amazon music, you need to get it activated from Amazon.

dhruvchikhaliya12, 2019-12-30:
Only tunein works i have saavn pro account but it didn't work.

avsheth, 2019-12-31:
Unfortunately, it won't work because it needs product ID to be whitelisted from Amazon. You need to create a new product and security profile on developer.amazon.com and request Amazon to whitelist that product ID.
Following which, you need to change product ID in the firmware app_main.c and the Android app.

dhruvchikhaliya12, 2019-12-31:
Thanks for your help i solve my issue as per your recomendation.

closed 2019-12-31 by dhruvchikhaliya12