espressif/esp-va-sdk #77: Not Working for amazon_alexa

PoojaG20, 2019-12-19:
**After flashing and provisioning(Wifi Configuration successful) also, its giving error as -**
[alexa]: Waiting for time to be updated
[alexa]: Done getting current time: 1576775963
[alexa]: Authentication done
[dialog]: Entering VA_IDLE
[speaker]: Volume changed to 40
[capabilities]: Capabilities unchanged
[endpoint_handler]: Cannot find endpoint URL in NVS. Setting default: https://avs-alexa-eu.amazon.com
[endpoint_handler]: AVS endpoint: https://avs-alexa-eu.amazon.com
[0;33mW (3179) I2S: I2S driver already installed[0m
[0;31mE (3189) esp-tls: mbedtls_ssl_set_hostname returned -0x7100[0m
[0;31mE (3189) esp-tls: Failed to open new connection[0m
[0;31mE (3189) httpc: Failed to create a new TLS connection[0m
[0;31mE (3189) [auth-delegate]: Failed to connect to Auth URL "https://api.amazon.com/auth/O2/token"[0m
[0;33mW (3199) I2S: I2S driver already installed[0m
[0;31mE (3199) [auth-delegate]: Error receiving refresh token response: -1[0m
[auth-delegate]: Token will be refreshed after 60 seconds.
[0;31mE (7089) [http_transport]: Failed to setup downchannel stream[0m
[endpoint_handler]: Cannot find endpoint URL in NVS. Setting default: https://avs-alexa-eu.amazon.com
[endpoint_handler]: AVS endpoint: https://avs-alexa-eu.amazon.com
[0;31mE (8759) [http_transport]: Failed to setup downchannel stream[0m
[endpoint_handler]: Cannot find endpoint URL in NVS. Setting default: https://avs-alexa-eu.amazon.com
[endpoint_handler]: AVS endpoint: https://avs-alexa-eu.amazon.com
**On flashing and restting device I get this log too -** 
E (3330) esp-tls: mbedtls_ssl_set_hostname returned -0x7100
E (3330) esp-tls: Failed to open new connection
E (3330) httpc: Failed to create a new TLS connection
E (3330) [auth-delegate]: Failed to connect to Auth URL "https://api.amazon.com/
auth/O2/token"
E (3340) [auth-delegate]: Error receiving refresh token response: -1

Please help am i missing something?

PoojaG20, 2019-12-21:
With esp-idf v 3.3.1 was getting the above issue, when tried with v3.3 its working.

avsheth, 2019-12-23:
As mentioned in the Getting started [guide](https://github.com/espressif/esp-va-sdk/blob/master/README-Getting-Started.md#clone-all-the-repositories) it should be compiled with v3.2 branch.

tcpipchip, 2020-07-07:
works nice on 3.2