espressif/esp-va-sdk #78: Questions about Alisa's single and dual microphones

DuHeLong, 2019-12-31:
Hello,
I'm using the latest SDK. During the test, I found that during the wake-up and dialogue of ALEXA, only the right microphone works, and the left microphone does not affect the function even if I short-circuit it. Why is this? Isn't it a dual microphone?
        I remember using the early SDK, shorting any microphone, ALEXA can work;

DuHeLong, 2019-12-31:
I tested it directly on the development board (esp32_lyrat).

vikramdattu, 2019-12-31:
For upload path, data is down-channeled and down sampled to 16K sampling rate.
Hence, another channel simply is irrelevant and data for that is ignored.

DuHeLong, 2020-01-06:
According to the specifications provided by your company, the function is a dual microphone function, and I remember that when testing in February last year, shielding any one microphone can work, can you tell us why? Did we test wrong before?

Refer to the following specifications：
      Key Supported Features:
•	Various mainstream, both lossy and lossless, compressed audio formats, including M4A, AAC, FLAC, OGG, OPUS, MP3, etc.
•	One-key configuration and wake-up from the standby mode.
•	SoftAP and Station mode.
•	Various wireless protocols：Wi-Fi 802.11b/g/n, Classic BT and Bluetooth LE.
•	A series of audio inputs, including Wi-Fi, BT-audio, DLNA, Line-in, etc.
•	Bluetooth LE network configuration, and smart network configuration with apps, such as WeChat.
•	Two microphones for the development of near-field and far-field voice recognition applications.
•	Peripherals for differentiated demands.