espressif/esp-va-sdk #80: complile failed in release/v3.2

skylli, 2020-01-04:
platform: windows
tool:https://dl.espressif.com/dl/esp32_win32_msys2_environment_and_toolchain-20181001.zip
esp-idf version: release/v3.2

`
CC build/esp_httpd_ota/src/esp_httpd_ota.o
In file included from E:/prj/esp32/rease_v3_2/msys32/home/b/esp/esp-va-sdk-master/components/esp_httpd_ota/include/esp_httpd_ota.h:7:0,
                 from E:/prj/esp32/rease_v3_2/msys32/home/b/esp/esp-va-sdk-master/components/esp_httpd_ota/src/esp_httpd_ota.c:6:
C:/Users/b/esp/esp-idf/components/esp_http_server/include/http_server.h:1:2: warning: #warning http_server.h has been renamed to esp_http_server.h, please update include directives [-Wcpp]
 #warning http_server.h has been renamed to esp_http_server.h, please update include directives
  ^
cc1.exe: warning: unrecognized command line option '-Wno-frame-address'
AR build/esp_httpd_ota/libesp_httpd_ota.a
CC build/esp_https_ota/src/esp_https_ota.o
AR build/esp_https_ota/libesp_https_ota.a
CC build/esp_local_ctrl/src/esp_local_ctrl_transport_ble.o
C:/Users/b/esp/esp-idf/components/esp_local_ctrl/src/esp_local_ctrl_transport_ble.c: In function 'start_ble_transport':
C:/Users/b/esp/esp-idf/components/esp_local_ctrl/src/esp_local_ctrl_transport_ble.c:35:36: warning: passing argument 2 of 'protocomm_ble_start' from incompatible pointer type [-Wincompatible-pointer-types]
     return protocomm_ble_start(pc, config->ble);
                                    ^
In file included from C:/Users/b/esp/esp-idf/components/esp_local_ctrl/src/esp_local_ctrl_transport_ble.c:20:0:
E:/prj/esp32/rease_v3_2/msys32/home/b/esp/esp-va-sdk-master/components/protocomm/include/transports/protocomm_ble.h:90:11: note: expected 'const protocomm_ble_config_t * {aka const struct <anonymous> *}' but argument is of type 'esp_local_ctrl_transport_config_ble_t * const {aka struct protocomm_ble_config * const}'
 esp_err_t protocomm_ble_start(protocomm_t *pc, const protocomm_ble_config_t *config);
           ^
C:/Users/b/esp/esp-idf/components/esp_local_ctrl/src/esp_local_ctrl_transport_ble.c: In function 'copy_ble_config':
C:/Users/b/esp/esp-idf/components/esp_local_ctrl/src/esp_local_ctrl_transport_ble.c:57:28: error: dereferencing pointer to incomplete type 'esp_local_ctrl_transport_config_ble_t {aka struct protocomm_ble_config}'
     memcpy(dest_config->ble->device_name,
                            ^
C:/Users/b/esp/esp-idf/components/esp_local_ctrl/src/esp_local_ctrl_transport_ble.c:58:27: error: dereferencing pointer to incomplete type 'esp_local_ctrl_transport_config_ble_t {aka struct protocomm_ble_config}'
            src_config->ble->device_name,
                           ^
C:/Users/b/esp/esp-idf/components/esp_local_ctrl/src/esp_local_ctrl_transport_ble.c: At top level:
cc1.exe: warning: unrecognized command line option '-Wno-frame-address'
make[1]: *** [/c/Users/b/esp/esp-idf/make/component_wrapper.mk:292：src/esp_local_ctrl_transport_ble.o] 错误 1
make: *** [C:\Users\b\esp\esp-idf/make/project.mk:584：component-esp_local_ctrl-build]
`

skylli, 2020-01-10:
Idf SDK  version v3.3.1  Compile  without problems。

vikramdattu, 2020-01-10:
Hello @skylli did you also applied the patches provided for IDF release/v3.2?

Those are under `idf_patches` directory.

skylli, 2020-01-16:
@vikramdattu  I have  try, but not luck.

skylli, 2020-01-16:
I decided to use v3.3.1, are there any risk hidden ?

avsheth, 2020-01-16:
Hi @skylli 
Did you try clean build after applying those patches?
As for v3.3.1, it shouldn't be a problem too, it's just we are yet to do a thorough testing with it.

skylli, 2020-01-16:
@avsheth  This time wrok, this time i download another idf sdk  with release/v3.2 and applied the patches.

closed 2021-01-14 by avsheth