espressif/esp-va-sdk #81: Provision with ESP Alexa failed

lxdemon, 2020-01-10:
I'm testing example amazon_alexa with ESP32 Audio Kit
I downloaded ESP Alexa APP ang try to provision, but i got these ERROR from esp log:
[va_button]: button pressed: 36
[conn_mgr_prov]: Provisioning started with: 
	service name: ESP-Alexa-6294 
	service key: 
	proof of possession (pop): abcd1234
[avs_config]: APP Got: ANZNCRCBXSZsoiHMGTow, amzn1.application-oa2-client.320d0c28204541e4a061b7aff8e7d1b9, amzn://com.espressif.provbleavs, abcd1234
[auth-delegate]: Auth delegate with comp app: ANZNCRCBXSZsoiHMGTow, amzn1.application-oa2-client.320d0c28204541e4a061b7aff8e7d1b9, amzn://com.espressif.provbleavs, abcd1234E (578129) esp-tls: couldn't get hostname for :api.amazon.com:
E (578139) esp-tls: Failed to open new connection
E (578149) httpc: Failed to create a new TLS connection
E (578149) [auth-delegate]: Failed to connect to Auth URL "https://api.amazon.com/auth/O2/token"
E (578159) [auth-delegate]: Authentication attempt failed. Retrying. Please check connectivity and/or credentials. In case of authentication failure try resetting the device to factory mode and restart provisioning
E (578679) esp-tls: couldn't get hostname for :api.amazon.com:
E (578679) esp-tls: Failed to open new connection
E (578679) httpc: Failed to create a new TLS connection
E (578679) [auth-delegate]: Failed to connect to Auth URL "https://api.amazon.com/auth/O2/token"
E (578689) [auth-delegate]: Authentication attempt failed. Retrying. Please check connectivity and/or credentials. In case of authentication failure try resetting the device to factory mode and restart provisioning
E (579209) esp-tls: couldn't get hostname for :api.amazon.com:
E (579209) esp-tls: Failed to open new connection
E (579209) httpc: Failed to create a new TLS connection
E (579209) [auth-delegate]: Failed to connect to Auth URL "https://api.amazon.com/auth/O2/token"
E (579219) [auth-delegate]: Authentication attempt failed. Retrying. Please check connectivity and/or credentials. In case of authentication failure try resetting the device to factory mode and restart provisioning
E (579739) esp-tls: couldn't get hostname for :api.amazon.com:
E (579739) esp-tls: Failed to open new connection
E (579739) httpc: Failed to create a new TLS connection
E (579739) [auth-delegate]: Failed to connect to Auth URL "https://api.amazon.com/auth/O2/token"
E (579749) [auth-delegate]: Authentication attempt failed. Retrying. Please check connectivity and/or credentials. In case of authentication failure try resetting the device to factory mode and restart provisioning
E (580269) esp-tls: couldn't get hostname for :api.amazon.com:
E (580269) esp-tls: Failed to open new connection
E (580269) httpc: Failed to create a new TLS connection
E (580269) [auth-delegate]: Failed to connect to Auth URL "https://api.amazon.com/auth/O2/token"
W (580279) wifi: Error! Should use default active scan time parameter for WiFi scan when Bluetooth is enabled!!!!!!

E (580279) [auth-delegate]: Authentication attempt failed. Retrying. Please check connectivity and/or credentials. In case of authentication failure try resetting the device to factory mode and restart provisioning
E (580809) esp-tls: couldn't get hostname for :api.amazon.com:
E (580809) esp-tls: Failed to open new connection
E (580809) httpc: Failed to create a new TLS connection
E (580809) [auth-delegate]: Failed to connect to Auth URL "https://api.amazon.com/auth/O2/token"
E (580819) [auth-delegate]: Authentication attempt failed. Retrying. Please check connectivity and/or credentials. In case of authentication failure try resetting the device to factory mode and restart provisioning
E (581339) esp-tls: couldn't get hostname for :api.amazon.com:
E (581339) esp-tls: Failed to open new connection
E (581339) httpc: Failed to create a new TLS connection
E (581339) [auth-delegate]: Failed to connect to Auth URL "https://api.amazon.com/auth/O2/token"
E (581349) [auth-delegate]: Authentication attempt failed. Retrying. Please check connectivity and/or credentials. In case of authentication failure try resetting the device to factory mode and restart provisioning
E (581869) esp-tls: couldn't get hostname for :api.amazon.com:
E (581869) esp-tls: Failed to open new connection
E (581869) httpc: Failed to create a new TLS connection
E (581869) [auth-delegate]: Failed to connect to Auth URL "https://api.amazon.com/auth/O2/token"
E (581879) [auth-delegate]: Authentication attempt failed. Retrying. Please check connectivity and/or credentials. In case of authentication failure try resetting the device to factory mode and restart provisioning
[conn_mgr_prov]: Scan results:
    S.N. SSID                             RSSI
    [ 0] HeyGoogle                         -40
    [ 1] Remotec_ASUS_2.4g                 -60
    [ 2] ChinaNet-7Xss                     -64
    [ 3] Keylinks_1900                     -68
    [ 4] KK                                -71
    [ 5] DIRECT-56-HP M227f LaserJet       -72
    [ 6] SZREMOTEC2                        -73
    [ 7] ChinaNet-cOwM                     -74
    [ 8] Headway                           -76
    [ 9] ChinaNet-KEVP                     -78
    [10] yixiang                           -78
    [11] GENZT-WIFI                        -82
    [12] ChinaNet-9nJr                     -83
    [13] GENZT-Guest                       -83
    [14] 火箭代运营                   -84
    [15] tianbao100                        -87
    [16] SZREMOTEC                         -87
    [17] ChinaNet-C2Ra                     -88
[... 160 lines cut ...]
A10     : 0x00000028  A11     : 0x00000000  A12     : 0x3ffbd4d0  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000020  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x40099030  LEND    : 0x4009903b  LCOUNT  : 0x00000000  
0x40099030: memset at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memset.S:142

0x4009903b: memset at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memset.S:152

Backtrace: 0x00000000:0x3ffbbe90 0x4011ba9f:0x3ffbbeb0 0x4011bbc9:0x3ffbbed0 0x4012355d:0x3ffbbf00 0x4011ccfc:0x3ffbbf20 0x4010103d:0x3ffbbf70 0x400d0fc3:0x3ffbc090
0x4011ba9f: va_reset_init at ??:?

0x4011bbc9: speaker_init at ??:?

0x4012355d: alexa_speaker_init at ??:?

0x4011ccfc: alexa_init at ??:?

0x4010103d: app_main at /home/lixin/workspace/esp-va/esp-va-sdk/examples/amazon_alexa/main/app_main.c:206

0x400d0fc3: main_task at /home/lixin/workspace/esp-va/esp-idf/components/esp32/cpu_start.c:519

Rebooting...

And the ESP Alexa APP show information: read from BLE failed

I don't know what's the problem

avsheth, 2020-01-15:
Does this happen everytime?

lxdemon, 2020-01-16:
Yes, everytime.
I also tried google_voice_assistant example, got similar error:

I (39434) [gva]: Done getting current time: 1579139377
I (39434) [tls_certification]: Done setting global CA store
I (39434) auth-delegate-config: Client ID or Refresh token found in NVS
I (39444) auth-delegate-config: Returning auth delegate subsequent auth configuration
I (39444) audio_stream: Starting playback_writer stream
I (39454) audio_stream: Starting audio stream playback_writer
I (39454) audio_stream: Stream playback_writer Event Started
[va_button]: button pressed: 39
I (39464) [app_va_cb]: Dialog state is: 8
[dialog]: Entering VA_IDLE
I (39474) [va_nvs_utils]: No value set for: speaker_vol
I (39484) [speaker]: Using default volume
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x00000000  PS      : 0x00060730  A0      : 0x8011dbba  A1      : 0x3ffb2830  
A2      : 0x00000000  A3      : 0x0000003c  A4      : 0x3ffbd83c  A5      : 0x0000000c  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x80100fc9  A9      : 0x3ffb27f0  
A10     : 0x0000003c  A11     : 0x00000000  A12     : 0x3ffbd8d4  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x40093bc1  LEND    : 0x40093bd1  LCOUNT  : 0xfffffffe  
0x40093bc1: strlen at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/strlen.S:84

0x40093bd1: strlen at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/strlen.S:96

Backtrace: 0x00000000:0x3ffb2830 0x4011dbb7:0x3ffb2850 0x4011dc78:0x3ffb2870 0x4011bf75:0x3ffb28a0 0x4011ab3a:0x3ffb28c0 0x40100ea7:0x3ffb2900 0x400d117b:0x3ffb29c0
0x4011dbb7: auth_delegate_config_get_subsequent_auth at ??:?

0x4011dc78: speaker_init at ??:?

0x4011bf75: google_speaker_init at ??:?

0x4011ab3a: gva_init at ??:?

0x40100ea7: app_main at /home/lixin/workspace/esp-va/esp-va-sdk/examples/google_voice_assistant/main/app_main.c:154

0x400d117b: main_task at /home/lixin/workspace/esp-va/esp-idf/components/esp32/cpu_start.c:519
Entering gdb stub now.

Because i'm using different hardware(ESP32 Audio Kit), so i wonder if this is the problem, i just removed the codec driver and replaced with my own where i get from ai-thinker.

Now we intend to use LyraT to test this, hope it is work .

avsheth, 2020-01-16:
Going by the logs, in both cases, it failed in speaker_init and could be related to the codec.
What is the IDF and toolchain version you are using?

lxdemon, 2020-01-16:
I'm just followed the Getting Started Guide: 
https://github.com/espressif/esp-va-sdk/blob/master/README-Getting-Started.md

IDF: v3.2.3-141-g6916052e6-dirty
OS: ubuntu18.04

avsheth, 2020-01-16:
toolchain version?
You can check that using command xtensa-esp32-elf-gcc -v

lxdemon, 2020-01-16:
Using built-in specs.
COLLECT_GCC=xtensa-esp32-elf-gcc
COLLECT_LTO_WRAPPER=/home/lixin/workspace/esp32/xtensa-esp32-elf/bin/../libexec/gcc/xtensa-esp32-elf/5.2.0/lto-wrapper
Target: xtensa-esp32-elf
Configured with: /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/configure --build=x86_64-build_pc-linux-gnu --host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf --prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf --with-local-prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf/sysroot --with-sysroot=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf/sysroot --with-newlib --enable-threads=no --disable-shared --with-pkgversion='crosstool-NG crosstool-ng-1.22.0-80-g6c4433a' --disable-__cxa_atexit --enable-cxx-flags='-fno-rtti -ffunction-sections' --with-gmp=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpfr=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpc=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-isl=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-cloog=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-libelf=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --enable-lto --enable-target-optspace --without-long-double-128 --disable-libgomp --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libquadmath-support --disable-nls --enable-languages=c,c++ --disable-libstdcxx-verbose --enable-threads=posix --enable-gcov-custom-rtio
Thread model: posix
gcc version 5.2.0 (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a)

vikramdattu, 2020-01-16:
Do you, by any chance, set wrong `AUDIO_BOARD_PATH`?
What's the IDF version?

lxdemon, 2020-01-16:
Because i have no board_support_pkgs for this hardware, so i just keep it default
export AUDIO_BOARD_PATH=/home/lixin/workspace/esp-va/esp-va-sdk/board_support_pkgs/lyrat/audio_board/audio_board_lyrat/

And this is my project structure
[image]

IDF version: v3.2.3-141-g6916052e6-dirty

vikramdattu, 2020-01-16:
@lxdemon you might need to create a board package similar to other packages. `lyrat` uses `es8388` codec.
Similarly, your board might be using some different codec. Can you please confirm this? Which dev board are you using exactly?
In your screen shot, I can see `ac101` driver there in components directory. Can you please create a `board_support_pkgs` with the same?

lxdemon, 2020-01-16:
I will try this later, Thank you!

webtop, 2020-01-19:
@vikramdattu 
> Do you, by any chance, set wrong `AUDIO_BOARD_PATH`?
> What's the IDF version?

I had a very similar problem and realized after reading your comment that I had my board path set wrong.

Thank you

closed 2021-01-14 by avsheth