espressif/esp-va-sdk #82: mbedTLS failing connection

webtop, 2020-01-19:
I've just flashed the Alexa project to my Lyra-TD MSC board, which went fine, but I cannot get a connection to the AVS endpoint due to a failing mbedTLS hostname set.

This is what I get upon boot of my device after flashing:

> I (62) boot: Chip Revision: 1
> I (68) boot_comm: chip revision: 1, min. bootloader chip revision: 0
> I (40) boot: ESP-IDF v3.3.1 2nd stage bootloader
> I (40) boot: compile time 21:17:16
> I (41) boot: Enabling RNG early entropy source...
> I (45) qio_mode: Enabling default flash chip QIO
> I (50) boot: SPI Speed      : 80MHz
> I (54) boot: SPI Mode       : QIO
> I (58) boot: SPI Flash Size : 4MB
> I (63) boot: Partition Table:
> I (66) boot: ## Label            Usage          Type ST Offset   Length
> I (73) boot:  0 nvs              WiFi data        01 02 00009000 00006000
> I (81) boot:  1 phy_init         RF data          01 01 0000f000 00001000
> I (88) boot:  2 factory          factory app      00 00 00010000 00390000
> I (96) boot: End of partition table
> I (100) boot_comm: chip revision: 1, min. application chip revision: 0
> I (107) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x215608 (2184712) map
> I (666) esp_image: segment 1: paddr=0x00225630 vaddr=0x3ffbdb60 size=0x05178 ( 20856) load
> I (672) esp_image: segment 2: paddr=0x0022a7b0 vaddr=0x40080000 size=0x00400 (  1024) load
> 0x40080000: _WindowOverflow4 at E:/Libraries/esp/esp-idf/components/freertos/xtensa_vectors.S:1779
> 
> I (673) esp_image: segment 3: paddr=0x0022abb8 vaddr=0x40080400 size=0x05458 ( 21592) load
> I (688) esp_image: segment 4: paddr=0x00230018 vaddr=0x400d0018 size=0x149134 (1347892) map
> 0x400d0018: _stext at ??:?
> 
> I (1030) esp_image: segment 5: paddr=0x00379154 vaddr=0x40085858 size=0x19338 (103224) load
> 0x40085858: bt_track_pll_cap at /home/aiqin/git_tree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_ana.c:756
> 
> I (1081) boot: Loaded app from partition at offset 0x10000
> I (1081) boot: Disabling RNG early entropy source...
> [conn_mgr_prov]: Found ssid: - removed -
> [conn_mgr_prov]: Found password: - removed -
> [app_main]: Connected with IP Address: 192.168.0.13
> [alexa]: Waiting for time to be updated
> [alexa]: Done getting current time: 1579413588
> [alexa]: Authentication done
> [dialog]: Entering VA_IDLE
> E (4974) esp-tls: mbedtls_ssl_set_hostname returned -0x7100
> E (4974) esp-tls: Failed to open new connection
> E (4974) httpc: Failed to create a new TLS connection
> E (4974) [auth-delegate]: Failed to connect to Auth URL "https://api.amazon.com/auth/O2/token"
> E (4984) [auth-delegate]: Error receiving refresh token response: -1
> [auth-delegate]: Token will be refreshed after 60 seconds.
> [speaker]: Volume changed to 40
> [capabilities]: Capabilities unchanged
> [endpoint_handler]: AVS endpoint: https://avs-alexa-na.amazon.com
> W (5164) I2S: I2S driver already installed
> E (6864) [http_transport]: Failed to setup downchannel stream
> [endpoint_handler]: AVS endpoint: https://avs-alexa-na.amazon.com

This seems to be the offending line:

> E (4974) esp-tls: mbedtls_ssl_set_hostname returned -0x7100

ESP-IDF version: 3.3.1
Board path set correctly to Lyra-TD MSC (I thought this was initially the issue as I had the wrong path, but same issue after change).

Any thoughts on what might be causing this to fail to set hostname?
I considered adding the WolfSSL component to the project to see if that solved it, but don't want to head down that route to get the same result, especially if the issue is just something board specific.

Thanks,
Paul

avsheth, 2020-01-19:
Hi @webtop,
Could you confirm a few things:
1. Have you enabled Bluetooth?
2. Can you try once with IDF v3.2? (and applying couple of additional patches present in idf_patches directory)
3. You are trying with Vanilla amazon_alexa app right? (sans any of your custom changes, if any)?

Thanks

webtop, 2020-01-19:
Hi @avsheth 

Yes, moving down to v3.2 seems to have fixed the issue. I was using esp-adf for a while at v3.3.1.

Thanks for your help.

closed 2020-01-19 by avsheth