espressif/esp-va-sdk #85: Couldn't find refreshToken.

yeonjeseo, 2020-02-11:
Sorry to ask stupid question.
I bought ESP32LyraTD-MSC, and i'm trying make use of google VA.
I followed as the document commented, but I couldn't find the refreshToken in my credential.json.
I'm new to this, and I have no idea what to do.
Any advice?

ChiragAtal, 2020-02-19:
Hi @yeonjeseo , I think Google has changed their documentation. Can you follow the steps 1, 2 and 3 from here: https://developers.google.com/assistant/sdk/reference/device-registration/register-device-manual#get-access-token and try again? You should get the JSON file with 'client_id', 'client_secret' and 'refresh_token'.

avsheth, 2021-01-14:
Closing this due to no activity. Please feel free to reopen if it isn't fixed.

closed 2021-01-14 by avsheth