espressif/esp-va-sdk #87: Building the branch feature/aia-beta fails to link.

alessiolei, 2020-03-26:
I followed the guide and installed esp-idf release/4.0 in WSL.
When trying to build the project using CMake the build itself is successful (not in the following log but i can give that log as well if needed) and when it tries to link i got this:

Checking Python dependencies...
Python requirements from /home/alessio/esp/esp-idf/requirements.txt are satisfied.
Executing action: all (aliases: build)
Running ninja in directory /home/alessio/esp/esp-va-sdk/examples/amazon_aia/build
Executing "ninja all"...
[1/5] Performing build step for 'bootloader'
ninja: no work to do.
[2/3] Linking CXX executable aia.elf
FAILED: aia.elf
: && /home/alessio/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++  -mlongcalls -Wno-frame-address  -nostdlib CMakeFiles/a
ia.elf.dir/project_elf_src.c.obj  -o aia.elf  esp-idf/esp_ringbuf/libesp_ringbuf.a esp-idf/driver/libdriver.a esp-idf/wpa_supplicant/libwpa_supplicant.a esp-idf/efuse/
libefuse.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/app_update/libapp_update.a esp-idf/spi_flash/libspi_flash.a esp-idf/nvs_flash/libnvs_flash.a esp-
idf/esp_wifi/libesp_wifi.a esp-idf/esp_eth/libesp_eth.a esp-idf/lwip/liblwip.a esp-idf/tcpip_adapter/libtcpip_adapter.a esp-idf/esp_event/libesp_event.a esp-idf/pthrea
d/libpthread.a esp-idf/espcoredump/libespcoredump.a esp-idf/esp32/libesp32.a esp-idf/xtensa/libxtensa.a esp-idf/esp_common/libesp_common.a esp-idf/esp_rom/libesp_rom.a
 esp-idf/soc/libsoc.a esp-idf/log/liblog.a esp-idf/heap/libheap.a esp-idf/freertos/libfreertos.a esp-idf/vfs/libvfs.a esp-idf/newlib/libnewlib.a esp-idf/cxx/libcxx.a e
sp-idf/app_trace/libapp_trace.a esp-idf/asio/libasio.a esp-idf/bt/libbt.a esp-idf/coap/libcoap.a esp-idf/console/libconsole.a esp-idf/nghttp/libnghttp.a esp-idf/esp-tl
s/libesp-tls.a esp-idf/esp_adc_cal/libesp_adc_cal.a esp-idf/esp_gdbstub/libesp_gdbstub.a esp-idf/tcp_transport/libtcp_transport.a esp-idf/esp_http_client/libesp_http_c
lient.a esp-idf/esp_http_server/libesp_http_server.a esp-idf/esp_https_ota/libesp_https_ota.a esp-idf/protobuf-c/libprotobuf-c.a esp-idf/protocomm/libprotocomm.a esp-i
df/mdns/libmdns.a esp-idf/esp_local_ctrl/libesp_local_ctrl.a esp-idf/esp_websocket_client/libesp_websocket_client.a esp-idf/expat/libexpat.a esp-idf/wear_levelling/lib
wear_levelling.a esp-idf/sdmmc/libsdmmc.a esp-idf/fatfs/libfatfs.a esp-idf/freemodbus/libfreemodbus.a esp-idf/jsmn/libjsmn.a esp-idf/json/libjson.a esp-idf/libsodium/l
iblibsodium.a esp-idf/mqtt/libmqtt.a esp-idf/openssl/libopenssl.a esp-idf/spiffs/libspiffs.a esp-idf/ulp/libulp.a esp-idf/unity/libunity.a esp-idf/wifi_provisioning/li
bwifi_provisioning.a esp-idf/main/libmain.a esp-idf/json_parser/libjson_parser.a esp-idf/httpc/libhttpc.a esp-idf/streams/libstreams.a esp-idf/utils/libutils.a esp-idf
/media_hal/libmedia_hal.a esp-idf/misc/libmisc.a esp-idf/audio_hal/libaudio_hal.a esp-idf/audio_pipeline/libaudio_pipeline.a esp-idf/sys_playback/libsys_playback.a esp
-idf/basic_player/libbasic_player.a esp-idf/esp_httpd_ota/libesp_httpd_ota.a esp-idf/multipart_parser/libmultipart_parser.a esp-idf/sh2lib/libsh2lib.a esp-idf/uuid-gen
/libuuid-gen.a esp-idf/esp-aws-iot/libesp-aws-iot.a esp-idf/is31fl3236/libis31fl3236.a esp-idf/audio_board_dspg/libaudio_board_dspg.a esp-idf/codec_es8388/libcodec_es8
388.a -Wl,--cref -Wl,--Map=/home/alessio/esp/esp-va-sdk/examples/amazon_aia/build/aia.map -fno-rtti -fno-lto /home/alessio/esp/esp-va-sdk/components/voice_assistant/li
b/libaia.a /home/alessio/esp/esp-va-sdk/components/codecs/lib/libcodecs.a /home/alessio/esp/esp-va-sdk/components/ssdp/lib/libssdp.a /home/alessio/esp/esp-va-sdk/board
_support_pkgs/esp32_lyratd_dspg/dsp_driver/dspg_driver/components/dbmd5-ipc/lib/libdspg-ipc.a /home/alessio/esp/esp-va-sdk/board_support_pkgs/esp32_lyratd_dspg/dsp_dri
ver/dspg_driver/components/va_dsp/lib/libva_dsp.a esp-idf/main/libmain.a esp-idf/streams/libstreams.a esp-idf/utils/libutils.a esp-idf/media_hal/libmedia_hal.a esp-idf
/misc/libmisc.a esp-idf/audio_hal/libaudio_hal.a esp-idf/audio_pipeline/libaudio_pipeline.a esp-idf/sys_playback/libsys_playback.a esp-idf/basic_player/libbasic_player
.a esp-idf/uuid-gen/libuuid-gen.a esp-idf/is31fl3236/libis31fl3236.a esp-idf/audio_board_dspg/libaudio_board_dspg.a esp-idf/codec_es8388/libcodec_es8388.a /home/alessi
o/esp/esp-va-sdk/components/voice_assistant/lib/libaia.a /home/alessio/esp/esp-va-sdk/components/codecs/lib/libcodecs.a /home/alessio/esp/esp-va-sdk/components/ssdp/li
b/libssdp.a /home/alessio/esp/esp-va-sdk/board_support_pkgs/esp32_lyratd_dspg/dsp_driver/dspg_driver/components/dbmd5-ipc/lib/libdspg-ipc.a /home/alessio/esp/esp-va-sd
k/board_support_pkgs/esp32_lyratd_dspg/dsp_driver/dspg_driver/components/va_dsp/lib/libva_dsp.a esp-idf/asio/libasio.a esp-idf/coap/libcoap.a esp-idf/esp_gdbstub/libes
p_gdbstub.a esp-idf/esp_https_ota/libesp_https_ota.a esp-idf/esp_http_client/libesp_http_client.a esp-idf/esp_local_ctrl/libesp_local_ctrl.a esp-idf/esp_websocket_clie
nt/libesp_websocket_client.a esp-idf/expat/libexpat.a esp-idf/freemodbus/libfreemodbus.a esp-idf/mqtt/libmqtt.a esp-idf/tcp_transport/libtcp_transport.a esp-idf/openss
l/libopenssl.a esp-idf/spiffs/libspiffs.a esp-idf/ulp/libulp.a esp-idf/unity/libunity.a esp-idf/esp_httpd_ota/libesp_httpd_ota.a esp-idf/esp-aws-iot/libesp-aws-iot.a e
sp-idf/jsmn/libjsmn.a esp-idf/httpc/libhttpc.a esp-idf/esp_adc_cal/libesp_adc_cal.a esp-idf/json_parser/libjson_parser.a /home/alessio/esp/esp-va-sdk/components/esp-do
wnmix/lib/libesp-downmix.a esp-idf/libsodium/liblibsodium.a esp-idf/wifi_provisioning/libwifi_provisioning.a esp-idf/protocomm/libprotocomm.a esp-idf/bt/libbt.a -L/hom
e/alessio/esp/esp-idf/components/bt/controller/lib -lbtdm_app esp-idf/protobuf-c/libprotobuf-c.a esp-idf/mdns/libmdns.a esp-idf/console/libconsole.a esp-idf/json/libjs
on.a esp-idf/multipart_parser/libmultipart_parser.a esp-idf/sh2lib/libsh2lib.a esp-idf/esp-tls/libesp-tls.a esp-idf/fatfs/libfatfs.a esp-idf/wear_levelling/libwear_lev
elling.a esp-idf/sdmmc/libsdmmc.a esp-idf/esp_http_server/libesp_http_server.a esp-idf/nghttp/libnghttp.a esp-idf/esp_ringbuf/libesp_ringbuf.a esp-idf/driver/libdriver
.a esp-idf/wpa_supplicant/libwpa_supplicant.a esp-idf/efuse/libefuse.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/app_update/libapp_update.a esp-idf/sp
i_flash/libspi_flash.a esp-idf/nvs_flash/libnvs_flash.a esp-idf/esp_wifi/libesp_wifi.a esp-idf/esp_eth/libesp_eth.a esp-idf/lwip/liblwip.a esp-idf/tcpip_adapter/libtcp
ip_adapter.a esp-idf/esp_event/libesp_event.a esp-idf/pthread/libpthread.a esp-idf/espcoredump/libespcoredump.a esp-idf/esp32/libesp32.a esp-idf/xtensa/libxtensa.a esp
-idf/esp_common/libesp_common.a esp-idf/esp_rom/libesp_rom.a esp-idf/soc/libsoc.a esp-idf/log/liblog.a esp-idf/heap/libheap.a esp-idf/freertos/libfreertos.a esp-idf/vf
s/libvfs.a esp-idf/newlib/libnewlib.a esp-idf/cxx/libcxx.a esp-idf/app_trace/libapp_trace.a esp-idf/mbedtls/mbedtls/library/libmbedtls.a esp-idf/mbedtls/mbedtls/librar
y/libmbedcrypto.a esp-idf/mbedtls/mbedtls/library/libmbedx509.a /home/alessio/esp/esp-idf/components/esp_wifi/lib_esp32/libcoexist.a /home/alessio/esp/esp-idf/componen
ts/esp_wifi/lib_esp32/libcore.a /home/alessio/esp/esp-idf/components/esp_wifi/lib_esp32/libespnow.a /home/alessio/esp/esp-idf/components/esp_wifi/lib_esp32/libmesh.a /
home/alessio/esp/esp-idf/components/esp_wifi/lib_esp32/libnet80211.a /home/alessio/esp/esp-idf/components/esp_wifi/lib_esp32/libphy.a /home/alessio/esp/esp-idf/compone
nts/esp_wifi/lib_esp32/libpp.a /home/alessio/esp/esp-idf/components/esp_wifi/lib_esp32/librtc.a /home/alessio/esp/esp-idf/components/esp_wifi/lib_esp32/libsmartconfig.
a esp-idf/esp_ringbuf/libesp_ringbuf.a esp-idf/driver/libdriver.a esp-idf/wpa_supplicant/libwpa_supplicant.a esp-idf/efuse/libefuse.a esp-idf/bootloader_support/libboo
tloader_support.a esp-idf/app_update/libapp_update.a esp-idf/spi_flash/libspi_flash.a esp-idf/nvs_flash/libnvs_flash.a esp-idf/esp_wifi/libesp_wifi.a esp-idf/esp_eth/l
ibesp_eth.a esp-idf/lwip/liblwip.a esp-idf/tcpip_adapter/libtcpip_adapter.a esp-idf/esp_event/libesp_event.a esp-idf/pthread/libpthread.a esp-idf/espcoredump/libespcor
edump.a esp-idf/esp32/libesp32.a esp-idf/xtensa/libxtensa.a esp-idf/esp_common/libesp_common.a esp-idf/esp_rom/libesp_rom.a esp-idf/soc/libsoc.a esp-idf/log/liblog.a e
[... cut ...]
/home/alessio/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: (.text.dl_
basic_conv_layer_quantised_weight+0x190): undefined reference to `ceil`
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

for reference
~/esp/esp-va-sdk/examples/amazon_aia$ echo $AUDIO_BOARD_PATH
/home/alessio/esp/esp-va-sdk/board_support_pkgs/esp32_vaquita_dspg/audio_board/audio_board_vaquita_dspg

ChiragAtal, 2020-03-26:
@alessiolei , Can you tell what OS and its version are you using? Also can you check the cmake version?

Also, can add this line and try again?
esp-idf/tools/cmake/project.cmake:394:
`target_link_libraries(${project_elf} "-Wl,--start-group")`

alessiolei, 2020-03-26:
Hi, i'm using Debian 10 and using CMAke 3.13.4

Adding that line fixed the issue. Thanks a lot

closed 2020-05-27 by kedars

SamMcCormick61, 2021-02-06:
I had to add     target_link_libraries(${project_elf} "-Wl,--start-group")
on line 462 in order to get it to work.

jalakamkiranadspl, 2021-03-30:
@SamMcCormick61 solution worked for me. I am using Ubuntu 20.04 and esp idf 4.4 version.

Nalininalu, 2021-08-29:
> @SamMcCormick61 solution worked for me. I am using Ubuntu 20.04 and esp idf 4.4 version.

Hai.. I'm facing similar issue.. bt this one is not working for me  20.04.2 and used all version (esp-idf) ...working fine till menuconfigration bt I'm getting this kind of error while build/flash

Nalininalu, 2021-08-29:
> @alessiolei , Can you tell what OS and its version are you using? Also can you check the cmake version?
> 
> Also, can add this line and try again?
> esp-idf/tools/cmake/project.cmake:394:
> `target_link_libraries(${project_elf} "-Wl,--start-group")`

Haii... I'm facing similar issue.. bt this one is not working for me  20.04.2 and used all version (esp-idf) working fine till menuconfigration bt I'm getting this kind of error while build/flash.. I'm really stuck

Nalininalu, 2021-08-29:
> I had to add target_link_libraries(${project_elf} "-Wl,--start-group")
> on line 462 in order to get it to work.

Hai.. I'm facing similar issue.. bt this one is not working for me  20.04.2 and used all version (esp-idf) ...working fine till menuconfigration bt I'm getting this kind of error while build/flash.. Im really stuck

Nalininalu, 2021-08-29:
> @alessiolei , Can you tell what OS and its version are you using? Also can you check the cmake version?
> 
> Also, can add this line and try again?
> esp-idf/tools/cmake/project.cmake:394:
> `target_link_libraries(${project_elf} "-Wl,--start-group")`

> I had to add target_link_libraries(${project_elf} "-Wl,--start-group")
> on line 462 in order to get it to work.

Hai.. I'm facing similar issue.. bt this one is not working for me  ... I'm using Ubuntu 20.04.2 and used all version (esp-idf).. working fine till menuconfigration bt I'm getting this kind of error while build/flash.. Im really stuck