espressif/esp-va-sdk #90: branch feature/aia-beta，  Cannot specify link libraries for target "voice_assistant" which is not built by this project.

enric-lee, 2020-04-16:
OS:                     ubuntu 16.04 x86_64
cmake:               3.6.3
python:              2.7.12
compile chain:  xtensa-esp32-elf-gcc8_2_0-esp-2020r1-linux-amd64

log is as follow:
> idf.py flash monitor
Checking Python dependencies...
Python requirements from /home/rak/esp32-voice/esp-idf/requirements.txt are satisfied.
Adding flash's dependency "all" to list of actions
Executing action: all (aliases: build)
Running cmake in directory /home/rak/esp32-voice/esp-va-sdk/examples/amazon_aia/build
Executing "cmake -G 'Unix Makefiles' -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DWARN_UNINITIALIZED=1 -DCCACHE_ENABLE=0 /home/rak/esp32-voice/esp-va-sdk/examples/amazon_aia"...
Warn about uninitialized values.
-- Found Git: /usr/bin/git (found version "2.7.4") 
-- Component directory /home/rak/esp32-voice/esp-idf/components/cbor does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/rak/esp32-voice/esp-idf/components/tinyusb does not contain a CMakeLists.txt file. No component will be added
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/rak/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/rak/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/rak/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/rak/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /home/rak/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project version: 5ed0869
-- Building ESP-IDF components for target esp32
Loading defaults file /home/rak/esp32-voice/esp-va-sdk/examples/amazon_aia/sdkconfig.defaults...
CMake Warning at /home/rak/esp32-voice/esp-idf/tools/cmake/crosstool_version_check.cmake:28 (message):
  Xtensa toolchain /home/rak/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
  crosstool-ng version esp-2020r1 doesn't match supported version esp-2019r2.
  Check Getting Started documentation or proceed at own risk.
Call Stack (most recent call first):
  /home/rak/esp32-voice/esp-idf/components/esp32/project_include.cmake:21 (crosstool_version_check)
  /home/rak/esp32-voice/esp-idf/tools/cmake/build.cmake:305 (include)
  /home/rak/esp32-voice/esp-idf/tools/cmake/build.cmake:458 (__build_process_project_includes)
  /home/rak/esp32-voice/esp-idf/tools/cmake/project.cmake:348 (idf_build_process)
  CMakeLists.txt:75 (project)

-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Found Perl: /usr/bin/perl (found version "5.22.1") 
-- Adding linker script /home/rak/esp32-voice/esp-va-sdk/examples/amazon_aia/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script /home/rak/esp32-voice/esp-idf/components/esp32/ld/esp32.extram.bss.ld
-- Adding linker script /home/rak/esp32-voice/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script /home/rak/esp32-voice/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script /home/rak/esp32-voice/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/rak/esp32-voice/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/rak/esp32-voice/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/rak/esp32-voice/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
CMake Error at /home/rak/esp32-voice/esp-va-sdk/components/voice_assistant/CMakeLists.txt:18 (target_link_libraries):
  Cannot specify link libraries for target "voice_assistant" which is not
  built by this project.

-- Configuring incomplete, errors occurred!
See also "/home/rak/esp32-voice/esp-va-sdk/examples/amazon_aia/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
rak@ubuntu:~/esp32-voice/esp-va-sdk/examples/amazon_aia$

ChiragAtal, 2020-04-16:
@enric-lee , the toolchain you are using is esp-2020r1. Can you try again with esp-2019r2?
Also what is your esp-idf branch?

enric-lee, 2020-04-16:
The branch is feature/aia-beta. I want to compile ESP32-Vaquita-DSPG.

enric-lee, 2020-04-16:
esp-idf  4.0

enric-lee, 2020-04-16:
When I change the toolchain to esp-2019r2, the error still exists.

_rak@ubuntu:~/esp32-voice/esp-va-sdk/examples/amazon_aia$ idf.py flash monitor
Checking Python dependencies...
Python requirements from /home/rak/esp32-voice/esp-idf/requirements.txt are satisfied.
Adding flash's dependency "all" to list of actions
Executing action: all (aliases: build)
Running cmake in directory /home/rak/esp32-voice/esp-va-sdk/examples/amazon_aia/build
Executing "cmake -G 'Unix Makefiles' -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DWARN_UNINITIALIZED=1 -DCCACHE_ENABLE=0 /home/rak/esp32-voice/esp-va-sdk/examples/amazon_aia"...
Warn about uninitialized values.
-- Found Git: /usr/bin/git (found version "2.7.4") 
-- IDF_TARGET not set, using default target: esp32
-- Project version: 5ed0869
-- Building ESP-IDF components for target esp32
Loading defaults file /home/rak/esp32-voice/esp-va-sdk/examples/amazon_aia/sdkconfig.defaults...
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Found Perl: /usr/bin/perl (found version "5.22.1") 
-- Adding linker script /home/rak/esp32-voice/esp-va-sdk/examples/amazon_aia/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script /home/rak/esp32-voice/esp-idf/components/esp32/ld/esp32.extram.bss.ld
-- Adding linker script /home/rak/esp32-voice/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script /home/rak/esp32-voice/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script /home/rak/esp32-voice/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/rak/esp32-voice/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/rak/esp32-voice/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/rak/esp32-voice/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
CMake Error at /home/rak/esp32-voice/esp-va-sdk/components/voice_assistant/CMakeLists.txt:18 (target_link_libraries):
  Cannot specify link libraries for target "voice_assistant" which is not
  built by this project.

-- Configuring incomplete, errors occurred!
See also "/home/rak/esp32-voice/esp-va-sdk/examples/amazon_aia/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1_

enric-lee, 2020-04-16:
CMakeOutput.log is as follow:

rak@ubuntu:~/esp32-voice/esp-va-sdk/examples/amazon_aia$ cat /home/rak/esp32-voice/esp-va-sdk/examples/amazon_aia/build/CMakeFiles/CMakeOutput.log
The target system is: Generic -  - 
The host system is: Linux - 4.15.0-74-generic - x86_64
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: /home/rak/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc 
Build flags: -mlongcalls;-Wno-frame-address
Id flags: 

The output was:
0

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.out"

The C compiler identification is GNU, found in "/home/rak/esp32-voice/esp-va-sdk/examples/amazon_aia/build/CMakeFiles/3.6.3/CompilerIdC/a.out"

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: /home/rak/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ 
Build flags: -mlongcalls;-Wno-frame-address
Id flags: 

The output was:
0

Compilation of the CXX compiler identification source "CMakeCXXCompilerId.cpp" produced "a.out"

The CXX compiler identification is GNU, found in "/home/rak/esp32-voice/esp-va-sdk/examples/amazon_aia/build/CMakeFiles/3.6.3/CompilerIdCXX/a.out"

Checking whether the ASM compiler is GNU using "--version" matched "(GNU assembler)|(GCC)|(Free Software Foundation)":
xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Determining if the C compiler works passed with the following output:
Change Dir: /home/rak/esp32-voice/esp-va-sdk/examples/amazon_aia/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_274be/fast"
/usr/bin/make -f CMakeFiles/cmTC_274be.dir/build.make CMakeFiles/cmTC_274be.dir/build
make[1]: Entering directory '/home/rak/esp32-voice/esp-va-sdk/examples/amazon_aia/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_274be.dir/testCCompiler.c.obj
/home/rak/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc    -mlongcalls -Wno-frame-address    -o CMakeFiles/cmTC_274be.dir/testCCompiler.c.obj   -c /home/rak/esp32-voice/esp-va-sdk/examples/amazon_aia/build/CMakeFiles/CMakeTmp/testCCompiler.c
Linking C executable cmTC_274be
/usr/local/bin/cmake -E cmake_link_script CMakeFiles/cmTC_274be.dir/link.txt --verbose=1
/home/rak/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc  -mlongcalls -Wno-frame-address   -nostdlib  CMakeFiles/cmTC_274be.dir/testCCompiler.c.obj  -o cmTC_274be 
/home/rak/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: warning: cannot find entry symbol _start; defaulting to 0000000000400054
make[1]: Leaving directory '/home/rak/esp32-voice/esp-va-sdk/examples/amazon_aia/build/CMakeFiles/CMakeTmp'

Detecting C compiler ABI info compiled with the following output:
Change Dir: /home/rak/esp32-voice/esp-va-sdk/examples/amazon_aia/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_1d8c0/fast"
/usr/bin/make -f CMakeFiles/cmTC_1d8c0.dir/build.make CMakeFiles/cmTC_1d8c0.dir/build
make[1]: Entering directory '/home/rak/esp32-voice/esp-va-sdk/examples/amazon_aia/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_1d8c0.dir/CMakeCCompilerABI.c.obj
/home/rak/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc    -mlongcalls -Wno-frame-address    -o CMakeFiles/cmTC_1d8c0.dir/CMakeCCompilerABI.c.obj   -c /usr/local/share/cmake-3.6/Modules/CMakeCCompilerABI.c
Linking C executable cmTC_1d8c0
/usr/local/bin/cmake -E cmake_link_script CMakeFiles/cmTC_1d8c0.dir/link.txt --verbose=1
/home/rak/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc  -mlongcalls -Wno-frame-address   -nostdlib -v CMakeFiles/cmTC_1d8c0.dir/CMakeCCompilerABI.c.obj  -o cmTC_1d8c0  
Using built-in specs.
COLLECT_GCC=/home/rak/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
COLLECT_LTO_WRAPPER=/home/rak/xtensa-esp32-elf/bin/../libexec/gcc/xtensa-esp32-elf/8.2.0/lto-wrapper
Target: xtensa-esp32-elf
Configured with: /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/configure --build=x86_64-build_pc-linux-gnu --host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf --prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf --with-local-prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf --with-headers=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf/include --with-newlib --enable-threads=no --disable-shared --with-pkgversion='crosstool-NG esp-2019r2' --disable-__cxa_atexit --enable-cxx-flags=-ffunction-sections --disable-libgomp --disable-libmudflap --disable-libmpx --disable-libssp --disable-libquadmath --disable-libquadmath-support --with-gmp=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpfr=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-mpc=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --with-isl=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools --enable-lto --enable-target-optspace --without-long-double-128 --disable-nls --enable-multiarch --enable-languages=c,c++ --disable-libstdcxx-verbose --enable-threads=posix --enable-gcov-custom-rtio --enable-libstdcxx-time=yes
Thread model: posix
gcc version 8.2.0 (crosstool-NG esp-2019r2) 
COMPILER_PATH=/home/rak/xtensa-esp32-elf/bin/../libexec/gcc/xtensa-esp32-elf/8.2.0/:/home/rak/xtensa-esp32-elf/bin/../libexec/gcc/:/home/rak/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../x [... cut ...]
    Feature record: CXX_FEATURE:1cxx_template_template_parameters
    Feature record: CXX_FEATURE:0cxx_thread_local
    Feature record: CXX_FEATURE:0cxx_trailing_return_types
    Feature record: CXX_FEATURE:0cxx_unicode_literals
    Feature record: CXX_FEATURE:0cxx_uniform_initialization
    Feature record: CXX_FEATURE:0cxx_unrestricted_unions
    Feature record: CXX_FEATURE:0cxx_user_literals
    Feature record: CXX_FEATURE:0cxx_variable_templates
    Feature record: CXX_FEATURE:0cxx_variadic_macros
    Feature record: CXX_FEATURE:0cxx_variadic_templates

enric-lee, 2020-04-17:
> @enric-lee , the toolchain you are using is esp-2020r1. Can you try again with esp-2019r2?
> Also what is your esp-idf branch?
esp-idf 4.0

albanoandrea, 2020-04-24:
I have the same problem, did the test clean 3 times on Ubuntu 16.04 and one time on Ubuntu 18.04 on Windows WSL.
Always same error.

Here exaclty what I did (everything is contained in MAIN_FOLDER, you need to export that path correctly):
```
git clone --recursive https://github.com/espressif/esp-idf.git
cd esp-idf; git checkout release/v4.0; git submodule init; git submodule update --init --recursive; cd ..
git clone https://github.com/espressif/esp-aws-iot.git
cd esp-aws-iot; git submodule update --init --recursive; cd ..
git clone https://github.com/espressif/esp-va-sdk.git
cd esp-va-sdk; git checkout feature/aia-beta; cd ..
cd esp-idf/
./install.sh 
. ./export.sh
cd ..

export MAIN_FOLDER=/path/to/main/folder
export ESPPORT=/dev/ttyUSB1
export IDF_PATH=$MAIN_FOLDER/esp-idf
export AWS_IOT_PATH=$MAIN_FOLDER/esp-aws-iot
export AUDIO_BOARD_PATH=$MAIN_FOLDER/esp-va-sdk/board_support_pkgs/esp32_vaquita_dspg/audio_board/audio_board_vaquita_dspg/

cd esp-va-sdk/examples/amazon_aia/
idf.py flash monitor
```
And here the error:
```
user@Ubuntu-M2-Andrea:~/man_folder/esp-va-sdk/examples/amazon_aia$ idf.py flash monitor
Checking Python dependencies...
Python requirements from /home/user/man_folder//esp-idf/requirements.txt are satisfied.
Adding flash's dependency "all" to list of actions
Executing action: all (aliases: build)
Running cmake in directory /home/user/man_folder/esp-va-sdk/examples/amazon_aia/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DWARN_UNINITIALIZED=1 -DCCACHE_ENABLE=0 /home/user/man_folder/esp-va-sdk/examples/amazon_aia"...
Warn about uninitialized values.
-- Found Git: /usr/bin/git (found version "2.7.4") 
-- Component directory /home/user/man_folder/esp-idf/components/cbor does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/user/man_folder/esp-idf/components/tinyusb does not contain a CMakeLists.txt file. No component will be added
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/user/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/user/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/user/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/user/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /home/user/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project version: 5ed0869
-- Building ESP-IDF components for target esp32
Loading defaults file /home/user/man_folder/esp-va-sdk/examples/amazon_aia/sdkconfig.defaults...
-- Found PythonInterp: /home/user/.espressif/python_env/idf4.0_py2.7_env/bin/python (found version "2.7.12") 
-- Found Perl: /usr/bin/perl (found version "5.22.1") 
-- Adding linker script /home/user/man_folder/esp-va-sdk/examples/amazon_aia/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script /home/user/man_folder/esp-idf/components/esp32/ld/esp32.extram.bss.ld
-- Adding linker script /home/user/man_folder/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script /home/user/man_folder/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script /home/user/man_folder/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/user/man_folder/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/user/man_folder/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/user/man_folder/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
CMake Error at /home/user/man_folder/esp-va-sdk/components/voice_assistant/CMakeLists.txt:18 (target_link_libraries):
  Cannot specify link libraries for target "voice_assistant" which is not
  built by this project.

-- Configuring incomplete, errors occurred!
See also "/home/user/man_folder/esp-va-sdk/examples/amazon_aia/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
user@Ubuntu-M2-Andrea:~/man_folder/esp-va-sdk/examples/amazon_aia$ 

```

ChiragAtal, 2020-04-24:
@albanoandrea , Can you check your cmake version? and also update it and try again?

albanoandrea, 2020-04-24:
Actually I was using and out of the tree cmake version 3.10.2
I'v removed that and now I'm using cmake version 3.5.1 (the up-to-date retreived by apt-get)

With this version the error is similar, but different:
```
What version of cmake is the one I should use?

user@Ubuntu-M2-Andrea:~/man_folderesp-va-sdk/examples/amazon_aia$ idf.py flash monitor
cmake: /usr/local/lib/libcurl.so.4: no version information available (required by cmake)
Checking Python dependencies...
Python requirements from /home/user/man_folder/esp-idf/requirements.txt are satisfied.
Adding flash's dependency "all" to list of actions
Executing action: all (aliases: build)
Running cmake in directory /home/user/man_folderesp-va-sdk/examples/amazon_aia/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DWARN_UNINITIALIZED=1 -DCCACHE_ENABLE=0 /home/user/man_folderesp-va-sdk/examples/amazon_aia"...
cmake: /usr/local/lib/libcurl.so.4: no version information available (required by cmake)
Warn about uninitialized values.
-- Found Git: /usr/bin/git (found version "2.7.4") 
-- Component directory /home/user/man_folderesp-idf/components/tinyusb does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/user/man_folderesp-idf/components/cbor does not contain a CMakeLists.txt file. No component will be added
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/user/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project version: 5ed0869
-- Building ESP-IDF components for target esp32
Loading defaults file /home/user/man_folderesp-va-sdk/examples/amazon_aia/sdkconfig.defaults...
-- Found PythonInterp: /home/user/.espressif/python_env/idf4.0_py2.7_env/bin/python (found version "2.7.12") 
-- Found Perl: /usr/bin/perl (found version "5.22.1") 
-- Adding linker script /home/user/man_folderesp-va-sdk/examples/amazon_aia/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script /home/user/man_folderesp-idf/components/esp32/ld/esp32.extram.bss.ld
-- Adding linker script /home/user/man_folderesp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script /home/user/man_folderesp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script /home/user/man_folderesp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/user/man_folderesp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/user/man_folderesp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/user/man_folderesp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
CMake Error at /home/user/man_folderesp-va-sdk/components/codecs/CMakeLists.txt:10 (target_link_libraries):
  Cannot specify link libraries for target "codec" which is not built by this
  project.

-- Configuring incomplete, errors occurred!
See also "/home/user/man_folderesp-va-sdk/examples/amazon_aia/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
user@Ubuntu-M2-Andrea:~/man_folderesp-va-sdk/examples/amazon_aia$
```
Which version do you suggest to try?

albanoandrea, 2020-04-24:
Ok, installed manually version 3.17.1 and now it build.
Can this be added to requirements?

ChiragAtal, 2020-04-24:
@albanoandrea , Okay, thanks. Yes we will add this.

closed 2020-05-27 by kedars