espressif/esp-va-sdk #91: [0;31mE (4421)[http_transport]: aws_iot_mqtt_init returned error : -2 [0m abort() was called at PC 0x400dc926 on core 1

enric-lee, 2020-04-17:
branch: feature/aia-beta
borad: esp32_vaquita_dspg
esp_idf: v4.0
AWS Account: Using Espressif's AWS Account

**The serial port log is as follow:**
[0;32mI (772) esp_image: segment 1: paddr=0x001dbdb0 vaddr=0x3ffbdb60 size=0x04260 ( 16992) load[0m
[0;32mI (778) esp_image: segment 2: paddr=0x001e0018 vaddr=0x400d0018 size=0x1214f0 (1185008) map[0m

[0;32mI (1192) esp_image: segment 3: paddr=0x00301510 vaddr=0x3ffc1dc0 size=0x0032c (   812) load[0m
[0;32mI (1193) esp_image: segment 4: paddr=0x00301844 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (1200) esp_image: segment 5: paddr=0x00301c4c vaddr=0x40080400 size=0x1bf7c (114556) load[0m

[0;32mI (1273) boot: Loaded app from partition at offset 0x20000[0m
[0;32mI (1273) boot: Disabling RNG early entropy source...[0m

[0;33mW (1778) spi_flash: Detected size(16384k) larger than the size in the binary image header(8192k). Using the size in the binary image header.[0m

I (1861) wifi:wifi driver task: 3ffd5a1c, prio:23, stack:3584, core=0
I (1871) wifi:wifi firmware version: 8ac0d06
I (1871) wifi:config NVS flash: enabled
I (1871) wifi:config nano formating: disabled
I (1871) wifi:Init dynamic tx buffer num: 32
I (1871) wifi:Init data frame dynamic rx buffer num: 32
I (1881) wifi:Init management frame dynamic rx buffer num: 32
I (1881) wifi:Init management short buffer num: 32
I (1891) wifi:Init static tx buffer num: 32
I (1891) wifi:Init static rx buffer size: 1600
I (1901) wifi:Init static rx buffer num: 10
I (1901) wifi:Init dynamic rx buffer num: 32
[0;31mE (1931) ES8311: ES8311 Write Reg error[0m

>> 
I (2041) wifi:mode : sta (c4:4f:33:40:e7:75)
I (2051) wifi:Set ps type: 0

I (2171) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (2171) wifi:state: init -> auth (b0)
I (2171) wifi:state: auth -> assoc (0)
I (2191) wifi:state: assoc -> run (10)

I (2291) wifi:connected with AccessMe, aid = 11, channel 1, 40U, bssid = 9a:bb:99:27:01:33
I (2291) wifi:security type: 3, phy: bgn, rssi: -14
I (2301) wifi:pm start, type: 0

I (2301) wifi:AP's beacon interval = 102400 us, DTIM period = 1

[app_main]: Connected with IP Address: 192.168.6.169
[alexa]: Waiting for time to be updated

[alexa]: Done getting current time: 1587094334
[alexa]: Authentication done

[dialog]: Entering VA_IDLE
[speaker]: Volume changed to 40

[attachment]: New with content_id: ais_attachment
[http_transport]: AWS Account ID: 28218307
[http_transport]: Client ID: 7884f7a2-03a4-4155-9378-19
[http_transport]: Current time in London: Fri Apr 17 03:32:14 2020
[0;31mE (4421) [http_transport]: aws_iot_mqtt_init returned error : -2[0m
abort() was called at PC 0x400dc926 on core 1

ELF file SHA256: 579715a9e42a760f4e887c25e2e0cf5020199f3750e9917ad9f0aa4675caefb8

Backtrace: 0x40094748:0x3ffe8220 0x40094b35:0x3ffe8240 0x400dc926:0x3ffe8260

Entering gdb stub now.
$T0b#e6

ChiragAtal, 2020-04-17:
@enric-lee , `-2` error means `NULL_VALUE_ERROR`. Also from the log, the `AWS Account ID` and the `Client ID` look incomplete. Could you create and flash the mfg.bin file, and try again?

enric-lee, 2020-04-20:
> @enric-lee , `-2` error means `NULL_VALUE_ERROR`. Also from the log, the `AWS Account ID` and the `Client ID` look incomplete. Could you create and flash the mfg.bin file, and try again?

I use espressif's AWS Account and create mfg.bin file according the mannual:
https://github.com/espressif/esp-va-sdk/tree/feature/aia-beta#31-using-espressifs-aws-account

enric-lee, 2020-04-20:
Thank you, I can connect AWS today.

closed 2020-04-20 by enric-lee