espressif/esp-va-sdk #92: [alexa]: Waiting for authentication tokens.

Kasifa, 2020-04-19:
Add device by ESP Alexa on iOS can get the final page "Things to try", HOWEVER, can not see the device in the "Manage Devices" . Also, the mic LES flash blue, and in the APP Amazon Alexa these is a device named "xxx's ESP Alexa"[offline]

PLEASE HELP!

case: example/amazon_aia
board: esp32_vaquita_dspg
esp_idf: v4.2
AWS Account: Using own AWS Account

following the monitor information:
----------------------------------------------------------------------------------
Winchester:amazon_aia $ make -j8 monitor ESPPORT=/dev/cu.SLAB_USBtoUART
Toolchain path: $HOME/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: esp-2020r1
Compiler version: 8.2.0
Python requirements from $HOME//esp/esp-idf/requirements.txt are satisfied.
MONITOR
--- idf_monitor on /dev/cu.SLAB_USBtoUART 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4
load:0x3fff0034,len:6952
load:0x40078000,len:14360
load:0x40080400,len:3980
entry 0x40080680
I (29) boot: ESP-IDF v4.2-dev-1097-g2e14149bf 2nd stage bootloader
I (29) boot: compile time 18:47:17
I (29) boot: chip revision: 3
I (34) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 80MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 8MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 fctry            WiFi data        01 02 00010000 00006000
I (93) boot:  3 otadata          OTA data         01 00 00016000 00002000
I (101) boot:  4 ota_0            OTA app          00 10 00020000 003f0000
I (108) boot:  5 ota_1            OTA app          00 11 00410000 003f0000
I (116) boot: End of partition table
I (120) boot_comm: chip revision: 3, min. application chip revision: 0
I (127) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x1be500 (1828096) map
I (717) esp_image: segment 1: paddr=0x001de528 vaddr=0x3ffbdb60 size=0x01af0 (  6896) load
I (720) esp_image: segment 2: paddr=0x001e0020 vaddr=0x400d0020 size=0x1259a4 (1202596) map
0x400d0020: _stext at ??:?

I (1106) esp_image: segment 3: paddr=0x003059cc vaddr=0x3ffbf650 size=0x028b4 ( 10420) load
I (1110) esp_image: segment 4: paddr=0x00308288 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /Users/czeng/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (1114) esp_image: segment 5: paddr=0x00308690 vaddr=0x40080400 size=0x19b14 (105236) load
I (1175) boot: Loaded app from partition at offset 0x20000
I (1175) boot: Disabling RNG early entropy source...
I (1771) wifi:wifi driver task: 3ffd59f0, prio:23, stack:3584, core=0
I (1781) wifi:wifi firmware version: 4234f65
I (1781) wifi:wifi certification version: v7.0
I (1781) wifi:config NVS flash: enabled
I (1781) wifi:config nano formating: disabled
I (1791) wifi:Init dynamic tx buffer num: 32
I (1791) wifi:Init data frame dynamic rx buffer num: 32
I (1791) wifi:Init management frame dynamic rx buffer num: 32
I (1801) wifi:Init management short buffer num: 32
I (1801) wifi:Init static tx buffer num: 32
I (1811) wifi:Init static rx buffer size: 1600
I (1811) wifi:Init static rx buffer num: 10
I (1821) wifi:Init dynamic rx buffer num: 32
[app_main]: Using the non-NVS aws_endpoint: a1eh5q7r1qopia-ats.iot.us-east-2.amazonaws.com
[app_main]: Using the non-NVS aws_account_id: 285349601562
[app_main]: Using the non-NVS client_id: c44f3340ed91
I (2741) wifi:mode : sta (c4:4f:33:40:ed:91)
I (2741) wifi:Set ps type: 0

I (2861) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (2861) wifi:state: init -> auth (b0)
I (2871) wifi:state: auth -> assoc (0)
I (3021) wifi:state: assoc -> run (10)
I (3041) wifi:connected with DL, aid = 1, channel 1, BW20, bssid = 54:a7:03:16:6e:fb
I (3041) wifi:security type: 4, phy: bgn, rssi: -52
I (3051) wifi:pm start, type: 0

I (3161) wifi:AP's beacon interval = 102400 us, DTIM period = 1
[app_main]: Connected with IP Address: 192.168.100.108
[alexa]: Waiting for time to be updated
[alexa]: Done getting current time: 1587293747
[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens.
[alexa]: Waiting for authentication tokens.

avsheth, 2020-04-20:
Hi @Kasifa can you please help with a few things:
1. what is the version of your iOS?
2. what is the app version you are using (ESP Alexa)? Can you check if there's an update available and check with the latest app?
3. Can you check if both the device and the phone are connected to the same network?
4. Can you try rebooting both device and the app and check if device is visible under "Manage Devices" now?
5. Can you try once by erasing the flash and re-flashing everything and start with provisioning process again?

Kasifa, 2020-04-21:
THANK YOU FOR YOUR REPLY

> Hi @Kasifa can you please help with a few things:
> 
> 1. what is the version of your iOS? 
**13.4**
> 2. what is the app version you are using (ESP Alexa)? Can you check if there's an update available and check with the latest app? 
 **version 2.1** 
> 3. Can you check if both the device and the phone are connected to the same network? 
**identical network and I tried to switch various networks**
> 4. Can you try rebooting both device and the app and check if device is visible under "Manage Devices" now?
**tried few times to rebooting and not working**
> 5. Can you try once by erasing the flash and re-flashing everything and start with provisioning process again?
**also tried few times to easing and re-flashing and not work**

Additionally, I use "make -j8 flash ESPPORT=/dev/cu.SLAB_USBtoUART" to flash, OTHERWISE, there was an error shown as below. Does this matter?
-------------------------------------------------------------

[1155/1156] Linking CXX executable aia.elf
FAILED: aia.elf 
: && /Users/czeng/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++  -mlongcalls -Wno-frame-address  -nostdlib CMakeFiles/aia.elf.dir/project_elf_src.c.obj  -o aia.elf  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/driver/libdriver.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/efuse/libefuse.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/app_update/libapp_update.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/lwip/liblwip.a  esp-idf/tcpip_adapter/libtcpip_adapter.a  esp-idf/esp_event/libesp_event.a  esp-idf/pthread/libpthread.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/esp32/libesp32.a  esp-idf/xtensa/libxtensa.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/soc/libsoc.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/freertos/libfreertos.a  esp-idf/vfs/libvfs.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/app_trace/libapp_trace.a  esp-idf/asio/libasio.a  esp-idf/bt/libbt.a  esp-idf/coap/libcoap.a  esp-idf/console/libconsole.a  esp-idf/nghttp/libnghttp.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc_cal/libesp_adc_cal.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/mdns/libmdns.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/esp_websocket_client/libesp_websocket_client.a  esp-idf/expat/libexpat.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/fatfs/libfatfs.a  esp-idf/freemodbus/libfreemodbus.a  esp-idf/jsmn/libjsmn.a  esp-idf/json/libjson.a  esp-idf/libsodium/liblibsodium.a  esp-idf/mqtt/libmqtt.a  esp-idf/openssl/libopenssl.a  esp-idf/spiffs/libspiffs.a  esp-idf/ulp/libulp.a  esp-idf/unity/libunity.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/main/libmain.a  esp-idf/json_parser/libjson_parser.a  esp-idf/httpc/libhttpc.a  esp-idf/streams/libstreams.a  esp-idf/utils/libutils.a  esp-idf/media_hal/libmedia_hal.a  esp-idf/misc/libmisc.a  esp-idf/audio_hal/libaudio_hal.a  esp-idf/audio_pipeline/libaudio_pipeline.a  esp-idf/sys_playback/libsys_playback.a  esp-idf/basic_player/libbasic_player.a  esp-idf/esp_httpd_ota/libesp_httpd_ota.a  esp-idf/multipart_parser/libmultipart_parser.a  esp-idf/sh2lib/libsh2lib.a  esp-idf/uuid-gen/libuuid-gen.a  esp-idf/esp-aws-iot/libesp-aws-iot.a  esp-idf/audio_board_vaquita_dspg/libaudio_board_vaquita_dspg.a  esp-idf/neo_pixel_led/libneo_pixel_led.a  esp-idf/codec_es8311/libcodec_es8311.a  -Wl,--cref -Wl,--Map=/Users/czeng/esp/esp-va-sdk/examples/amazon_aia/build/aia.map  -fno-rtti  -fno-lto  /Users/czeng/esp/esp-va-sdk/components/voice_assistant/lib/libaia.a  /Users/czeng/esp/esp-va-sdk/components/codecs/lib/libcodecs.a  /Users/czeng/esp/esp-va-sdk/components/ssdp/lib/libssdp.a  /Users/czeng/esp/esp-va-sdk/board_support_pkgs/esp32_vaquita_dspg/dsp_driver/dspg_driver/components/dbmd5-ipc/lib/libdspg-ipc.a  /Users/czeng/esp/esp-va-sdk/board_support_pkgs/esp32_vaquita_dspg/dsp_driver/dspg_driver/components/va_dsp/lib/libva_dsp.a  esp-idf/main/libmain.a  esp-idf/streams/libstreams.a  esp-idf/utils/libutils.a  esp-idf/media_hal/libmedia_hal.a  esp-idf/misc/libmisc.a  esp-idf/audio_hal/libaudio_hal.a  esp-idf/audio_pipeline/libaudio_pipeline.a  esp-idf/sys_playback/libsys_playback.a  esp-idf/basic_player/libbasic_player.a  esp-idf/uuid-gen/libuuid-gen.a  esp-idf/audio_board_vaquita_dspg/libaudio_board_vaquita_dspg.a  esp-idf/neo_pixel_led/libneo_pixel_led.a  esp-idf/codec_es8311/libcodec_es8311.a  /Users/czeng/esp/esp-va-sdk/components/voice_assistant/lib/libaia.a  /Users/czeng/esp/esp-va-sdk/components/codecs/lib/libcodecs.a  /Users/czeng/esp/esp-va-sdk/components/ssdp/lib/libssdp.a  /Users/czeng/esp/esp-va-sdk/board_support_pkgs/esp32_vaquita_dspg/dsp_driver/dspg_driver/components/dbmd5-ipc/lib/libdspg-ipc.a  /Users/czeng/esp/esp-va-sdk/board_support_pkgs/esp32_vaquita_dspg/dsp_driver/dspg_driver/components/va_dsp/lib/libva_dsp.a  esp-idf/asio/libasio.a  esp-idf/coap/libcoap.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/ [... cut ...] b.a(dl_lib.o): in function `dl_basic_conv_layer_quantised_weight':
(.text.dl_basic_conv_layer_quantised_weight+0x157): undefined reference to `ceil'
/Users/czeng/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: (.text.dl_basic_conv_layer_quantised_weight+0x190): undefined reference to `ceil'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

ChiragAtal, 2020-04-21:
@Kasifa , 
1. Using `make` directly is fine. Although, for the linking issue when using `idf.py build`, can you try https://github.com/espressif/esp-va-sdk/issues/87#issuecomment-604481636 ?
2. Can you factory reset the device and send a complete log of the provisioning flow too?
3. Do you have the Amazon shopping app installed on your iOS device?
4. Also, if possible, could you try an Android phone?

Kasifa, 2020-04-21:
Hi avsheth，

I start over again and it DOES work this time. have no idea....

Kasifa, 2020-04-21:
> @Kasifa ,
> 
> 1. Using `make` directly is fine. Although, for the linking issue when using `idf.py build`, can you try [#87 (comment)](https://github.com/espressif/esp-va-sdk/issues/87#issuecomment-604481636) ?
> 2. Can you factory reset the device and send a complete log of the provisioning flow too?
> 3. Do you have the Amazon shopping app installed on your iOS device?
> 4. Also, if possible, could you try an Android phone?

thank you.  I will check the issue with idf.py later

closed 2020-05-27 by ChiragAtal