espressif/esp-va-sdk #93: Can Google Dialogflow and Alexa work at the same time？

skylli, 2020-05-09:
I'm building a robot which can do someting with voice and answer some question like Alexa.
I know i can build a alexa custom skill to do that, but i have to start with 'Alexa ask robot to do somthing ' which experience is very poor.
Can Google Dialogflow and Alexa work at the same time? If it can, how i can make it?

avsheth, 2020-05-11:
Hi @skylli 
Unfortunately it's not possible to have multiple built-in voice assistants as of yet.
Besides from the usecase perspective, how would you identify which query to be sent to the Alexa and which one to be sent to the DialogFlow?
It would either require multiple WW support (e.g 'OK google' and 'Alexa') and some way for DSPs to notify the host of which WW has been detected. Or it needs some sort of query/speech processing locally. Both of which would require significant time and effort. :)

skylli, 2020-05-15:
@avsheth 
Thanks for your answer.
So the multiple WW  is wake Aws-Alexa with 'Alexa', and wake DialogFlow with 'Ok Google' or 'Hey, rootbot'. How can we  to achieve that.
Can you provide an example of the implementatio. 
I have connect Espressif, and they have send me an sdk of gitlab, but it oes not help.

avsheth, 2020-05-15:
Hi @skylli 
Like I said, it requires significant time and efforts. Multiple WW support (in fact we only have `Alexa` WW support right now), and support for multiple voice assistant clouds is yet to be implemented, which is quite some work.

skylli, 2020-05-16:
Are there any plans to achieve this function？

skylli, 2020-05-16:
@avsheth 
I just read Google Assistant Doc, and i find we can [ custom device actions ](https://developers.google.com/assistant/sdk/device-actions-overview#custom)
Does our esp-va-sdk can do that ?