espressif/esp-va-sdk #94: issue wile compiling

manmohan777, 2020-05-26:
I am getting the following error while executing  "idf.py flash monitor" command

xtensa-esp32-elf-gcc.exe: error: unrecognized command line option '-mfix-esp32-psram-cache-strategy=memw' ;

my xtensa-esp32-elf-gcc version is 8.2.0.

vikramdattu, 2020-05-26:
Only 5.2 based toolchain + IDF 3.2 is supported currently.

Please follow this: https://github.com/espressif/esp-va-sdk/blob/master/README-Getting-Started.md

manmohan777, 2020-05-27:
thank you for your reply.
by upgrading the toolchain I solved that issue but now I am getting a new error.

undefined reference to 'audio_stream_init'
collect2.exe: error: Id returned 1 exit status

vikramdattu, 2020-05-27:
Could you please try clean build? Remove `build/` folder and try again.
Old files might be messing around.

manmohan777, 2020-05-27:
already done that still getting the same error.

manmohan777, 2020-05-27:
I also cloned the project again but still getting the same error

ChiragAtal, 2020-05-27:
@manmohan777 Which example are you trying to build?

manmohan777, 2020-05-27:
esp-va-sdk\examples\amazon_aia

manmohan777, 2020-05-27:
I don't know how now the error has changed.
now its saying 
undefined reference to 'exit'
undefined reference to 'logf'
undefined reference to 'round'
undefined reference to 'ceilf'
undefined reference to 'log10f'
undefined reference to 'sinf'

ChiragAtal, 2020-05-27:
Regarding the toolchain and IDF version to be used, please use the one mentioned in the Readme instead of the one @vikramdattu mentioned above. For AIA, toolchain version esp-2019r2 or esp-2020r1 (both of which is based on 8.2.0) can be used and the IDF branch release/v4.0.

Regarding your earlier issue which you were getting with this toolchain, can you send the full logs? Also can you send you cmake version and try with the latest version again?

manmohan777, 2020-05-27:
my toolchain version is  esp-2020r1
and my cmake version is 3.17.2

manmohan777, 2020-05-27:
C:\Users\himraw\Documents\alxaDev\esp-va-sdk\examples\amazon_aia>idf.py flash monitor
Checking Python dependencies...
Python requirements from C:\Users\himraw\Documents\alxaDev\esp-idf\requirements.txt are satisfied.
Adding flash's dependency "all" to list of actions
Executing action: all (aliases: build)
Running cmake in directory c:\users\himraw\documents\alxadev\esp-va-sdk\examples\amazon_aia\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DWARN_UNINITIALIZED=1 -DCCACHE_ENABLE=0 c:\users\himraw\documents\alxadev\esp-va-sdk\examples\amazon_aia"...
Warn about uninitialized values.
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.26.2.windows.1")
-- Component directory C:/Users/himraw/Documents/alxaDev/esp-idf/components/aws_iot does not contain a CMakeLists.txt file. No component will be added
-- Component directory C:/Users/himraw/Documents/alxaDev/esp-idf/components/cbor does not contain a CMakeLists.txt file. No component will be added
-- Component directory C:/Users/himraw/Documents/alxaDev/esp-idf/components/micro-ecc does not contain a CMakeLists.txt file. No component will be added
-- Component directory C:/Users/himraw/Documents/alxaDev/esp-idf/components/tinyusb does not contain a CMakeLists.txt file. No component will be added
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program Files/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Program Files/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Program Files/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Program Files/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/Program Files/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project version: 212ac08
-- Building ESP-IDF components for target esp32
Loading defaults file C:/Users/himraw/Documents/alxaDev/esp-va-sdk/examples/amazon_aia/sdkconfig.defaults...
CMake Warning at C:/Users/himraw/Documents/alxaDev/esp-idf/tools/cmake/crosstool_version_check.cmake:28 (message):
  Xtensa toolchain C:/Program
  Files/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe crosstool-ng version
  esp-2020r1 doesn't match supported version esp-2019r2.  Check Getting
  Started documentation or proceed at own risk.
Call Stack (most recent call first):
  C:/Users/himraw/Documents/alxaDev/esp-idf/components/esp32/project_include.cmake:21 (crosstool_version_check)
  C:/Users/himraw/Documents/alxaDev/esp-idf/tools/cmake/build.cmake:305 (include)
  C:/Users/himraw/Documents/alxaDev/esp-idf/tools/cmake/build.cmake:458 (__build_process_project_includes)
  C:/Users/himraw/Documents/alxaDev/esp-idf/tools/cmake/project.cmake:348 (idf_build_process)
  CMakeLists.txt:75 (project)

-- Found PythonInterp: C:/Users/himraw/AppData/Local/Programs/Python/Python38/python.exe (found version "3.8.3")
-- Could NOT find Perl (missing: PERL_EXECUTABLE)
-- Adding linker script C:/Users/himraw/Documents/alxaDev/esp-va-sdk/examples/amazon_aia/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script C:/Users/himraw/Documents/alxaDev/esp-idf/components/esp32/ld/esp32.extram.bss.ld
-- Adding linker script C:/Users/himraw/Documents/alxaDev/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script C:/Users/himraw/Documents/alxaDev/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script C:/Users/himraw/Documents/alxaDev/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Users/himraw/Documents/alxaDev/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Users/himraw/Documents/alxaDev/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/Users/himraw/Documents/alxaDev/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Components: app_trace app_update asio audio_board_vaquita_dspg audio_hal audio_pipeline basic_player bootloader bootloader_support bt coap codec_es8311 codecs console cxx dbmd5-ipc driver efuse esp-aws-iot esp-downmix esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_http_client esp_http_server esp_httpd_ota esp_https_ota esp_https_server esp_local_ctrl esp_ringbuf esp_rom esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap httpc idf_test jsmn json json_parser libsodium log lwip main mbedtls mdns media_hal misc mqtt multipart_parser neo_pixel_led newlib nghttp nvs_flash openssl partition_table protobuf-c protocomm pthread sdmmc sh2lib soc speech_recog spi_flash spiffs ssdp streams sys_playback tcp_transport tcpip_adapter ulp ulp_adc unity utils uuid-gen va_dsp vfs voice_assistant wear_levelling wifi_provisioning wpa_supplicant xtensa
[... cut ...]
c:/progra~1/xtensa~1/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/himraw/Documents/alxaDev/esp-va-sdk/components/speech_recog/lib\libdl_lib.a(dl_lib.o): in function `dl_basic_conv_layer_quantised_weight':
(.text.dl_basic_conv_layer_quantised_weight+0x157): undefined reference to `ceil'
c:/progra~1/xtensa~1/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: (.text.dl_basic_conv_layer_quantised_weight+0x190): undefined reference to `ceil'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

manmohan777, 2020-05-27:
Adding ESP-IDF tools to PATH...
Not using an unsupported version of tool xtensa-esp32-elf found in PATH: esp-2020r1-8.2.0.
Not using an unsupported version of tool cmake found in PATH: 3.17.2.
Not using an unsupported version of tool ninja found in PATH: 1.10.0.

ChiragAtal, 2020-05-27:
Can you add this suggestion and try again?
https://github.com/espressif/esp-va-sdk/issues/87#issuecomment-604481636

manmohan777, 2020-05-27:
Thank you so much for your help. This worked for me.

closed 2020-05-27 by manmohan777